zephyrproject-rtos/zephyr #78553: JLink Runner should prefer ELF over HEX over BIN (currently it ignores ELF)

nelarsen, 2024-09-17:
I observed significantly long execution times for the `west flash` command in a specific Zephyr project where the output .elf file has scattered sections. In this project:
 - The program code is placed at the beginning of the flash.
 - A custom section containing configuration data is placed near the end of the flash.

As a result, the large unused flash area in between is unnecessarily erased, flashed, and verified by JLink Commander, which takes over a minute—very long.

**Why is this happening?**
By default, the Zephyr JLink runner flashes a BIN file, which does not recognize "sections" and includes several megabytes of unused space (filled with 0xFF), leading to the unnecessary flashing of empty regions.

Even enabling HEX file generation (CONFIG_BUILD_OUTPUT_HEX=y) doesn’t help, as (for unknown reason) the HEX file also includes the empty areas, just like the BIN file. However, the always-generated ELF file has the correct sections and flashes in seconds, leaving unrelated flash areas untouched.

**Proposal:**
The JLink Runner should always prioritize flashing the ELF over HEX over BIN files. I will submit a PR for discussion.

nordicjm, 2024-09-17:
> JLink Runner should prefer ELF over HEX over BIN

No it should not, hex file can be signed, elf file has no signature, so if using MCUboot this change would completely break it. Use `west debug` if you want to load the elf file

nelarsen, 2024-09-17:
Thank you, I did not consider that. Actually I would prefer that my custom section does not end up in the elf/hex/bin at all. Which I indeed can archieve with NOLOAD in the custom linker snippet, I addded with zephyr_linker_sources(). But, alas, Zephyr still places .last_section after the custom section, such than my image (now not containing the custom section) anyway turns out huge.

nelarsen, 2024-09-17:
> Have you tried setting `CONFIG_BUILD_NO_GAP_FILL`?
Great hint! With this option, the generated hex file is small and flashes fast (=solves my issue) because the gaps are not filled. I'm wondering why gaps are filled by default, though.

closed 2024-09-17 by nelarsen