zephyrproject-rtos/zephyr #78591: drivers: i3c: common: ENTDAA after request is not possible after RSTDAA

albertofloyd, 2024-09-17:
**Describe the bug**
If Reset command and reset DAA are issued, cannot request dynamic assignment (DAA) again which per spec shall be possible
[image]

After I3C bus initialization occurs and DAA takes place, if application sends a reset CCC command and reset DAA CCC, following request dynamic assignment (DAA) is aborted with error indicating the original dynamic address was already assigned.

0:01:17.113,000] <err> i3c: PID 0x0208006c100b already has dynamic address (0x09) assigned
[00:01:17.113,000] <err> npcx_i3c: npcx_i3c_do_daa: Assign new DA error

**To Reproduce**

Simple app performing the following sequence in main:
i3c_ccc_do_rstact_all(i3c_dev, I3C_CCC_RSTACT_RESET_WHOLE_TARGET)
i3c_ccc_do_rstdaa_all(i3c_dev)
LOG_WRN("Perform DAA");
i3c_do_daa(i3c_dev);

west build -c -p always -b npcx4m8f_evb 

**Expected behavior**

After RSTDAA is issue a ENTDAA command shall allow to performed.

**Impact**

This is against I3C spec expectations. (Section 5.1.9.3.3)

**Logs and console output**

[00:01:17.112,000] <dbg> npcx_i3c: npcx_i3c_do_ccc: Enter npcx_i3c_do_ccc
[00:01:17.112,000] <dbg> npcx_i3c: npcx_i3c_do_ccc: CCC[0x2a]
[00:01:17.112,000] <dbg> npcx_i3c: npcx_i3c_send_request: npcx_i3c_send_request fc41
[00:01:17.112,000] <dbg> npcx_i3c: npcx_i3c_send_request: npcx_i3c_send_request 2
[00:01:17.112,000] <dbg> npcx_i3c: npcx_i3c_do_ccc: Enter npcx_i3c_do_ccc
[00:01:17.112,000] <dbg> npcx_i3c: npcx_i3c_do_ccc: CCC[0x06]
[00:01:17.112,000] <dbg> npcx_i3c: npcx_i3c_send_request: npcx_i3c_send_request fc41
[00:01:17.112,000] <dbg> npcx_i3c: npcx_i3c_send_request: npcx_i3c_send_request 2
[00:01:17.112,000] <dbg> npcx_i3c: npcx_i3c_do_ccc: Enter npcx_i3c_do_ccc
[00:01:17.112,000] <dbg> npcx_i3c: npcx_i3c_do_ccc: CCC[0x01]
[00:01:17.112,000] <dbg> npcx_i3c: npcx_i3c_send_request: npcx_i3c_send_request fc41
[00:01:17.113,000] <dbg> npcx_i3c: npcx_i3c_send_request: npcx_i3c_send_request 2
[00:01:17.113,000] <wrn> i3c_m: Perform DAA
[00:01:17.113,000] <dbg> npcx_i3c: npcx_i3c_do_daa: npcx_i3c_do_daa
[00:01:17.113,000] <dbg> npcx_i3c: npcx_i3c_do_daa: DAA: ENTDAA
[00:01:17.113,000] <dbg> npcx_i3c: npcx_i3c_request_daa: npcx_i3c_request_daa
[00:01:17.113,000] <dbg> npcx_i3c: npcx_i3c_send_request: npcx_i3c_send_request 44
[00:01:17.113,000] <dbg> npcx_i3c: npcx_i3c_do_daa: DAA: Rcvd PID 0x0208006c100b
[00:01:17.113,000] <err> i3c: PID 0x0208006c100b already has dynamic address (0x09) assigned
[00:01:17.113,000] <err> npcx_i3c: npcx_i3c_do_daa: Assign new DA error
[00:01:17.113,000] <dbg> npcx_i3c: npcx_i3c_send_request: npcx_i3c_send_request 2
[00:01:17.113,000] <err> i3c_m: DAA -22

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**

dcpleung, 2024-09-17:
@XenuIsWatching Is it something that can be reproduced on your side? I don't have a I3C setup right now so I cannot even test any changes.

XenuIsWatching, 2024-09-18:
I do not have this platform...

but i think I know what is going on. You need to be manually setting all the i3c devices through the `i3c_device_desc` `dynamic_addr` to `0` after you do the RSTDAA.

albertofloyd, 2024-09-20:
@XenuIsWatching  I believe this is not platform specific since the check that throws the error is i3c_common code.
Also, is your suggestion that the client's driver code manipulates the i3c controller driver data directly? 

Wouldn't be better that this is handled in I3C bus driver?  I created this draft based on 2nd option, this works, but seems this particular scenario (RSTDAA, ENTDAA) needs more changes e.g. with said PR change the ENTDAA succeeds.

https://github.com/zephyrproject-rtos/zephyr/pull/78784

```
00:00:12.045,000] <dbg> npcx_i3c: npcx_i3c_do_ccc: Enter npcx_i3c_do_ccc
[00:00:12.045,000] <dbg> npcx_i3c: npcx_i3c_do_ccc: CCC[0x2a]
[00:00:12.045,000] <dbg> npcx_i3c: npcx_i3c_send_request: npcx_i3c_send_request fc41
[00:00:12.045,000] <dbg> npcx_i3c: npcx_i3c_send_request: npcx_i3c_send_request 2
[00:00:12.045,000] <wrn> i3c: Reset dynamic address for device i3c-dev2@208006c100b
[00:00:12.045,000] <dbg> npcx_i3c: npcx_i3c_do_ccc: Enter npcx_i3c_do_ccc
[00:00:12.045,000] <dbg> npcx_i3c: npcx_i3c_do_ccc: CCC[0x06]
[00:00:12.045,000] <dbg> npcx_i3c: npcx_i3c_send_request: npcx_i3c_send_request fc41
[00:00:12.045,000] <dbg> npcx_i3c: npcx_i3c_send_request: npcx_i3c_send_request 2
[00:00:12.046,000] <dbg> npcx_i3c: npcx_i3c_do_ccc: Enter npcx_i3c_do_ccc
[00:00:12.046,000] <dbg> npcx_i3c: npcx_i3c_do_ccc: CCC[0x01]
[00:00:12.046,000] <dbg> npcx_i3c: npcx_i3c_send_request: npcx_i3c_send_request fc41
[00:00:12.046,000] <dbg> npcx_i3c: npcx_i3c_send_request: npcx_i3c_send_request 2
[00:00:12.046,000] <wrn> i3c_m: Perform DAA
[00:00:12.046,000] <dbg> npcx_i3c: npcx_i3c_do_daa: DAA: ENTDAA
[00:00:12.046,000] <dbg> npcx_i3c: npcx_i3c_request_daa: npcx_i3c_request_daa
[00:00:12.046,000] <dbg> npcx_i3c: npcx_i3c_send_request: npcx_i3c_send_request 44
[00:00:12.046,000] <dbg> npcx_i3c: npcx_i3c_do_daa: DAA: Rcvd PID 0x0208006c100b
[00:00:12.046,000] <dbg> npcx_i3c: npcx_i3c_request_daa: npcx_i3c_request_daa
[00:00:12.046,000] <dbg> npcx_i3c: npcx_i3c_send_request: npcx_i3c_send_request 44
[00:00:12.046,000] <dbg> npcx_i3c: npcx_i3c_do_daa: PID 0x0208006c100b assigned dynamic address 0x0a
[00:00:12.046,000] <wrn> i3c_m: Find I3C device using PID
[00:00:12.046,000] <wrn> i3c_m: Dynamic addr a

```
But when using the I3C descriptor it looks like some descriptor data is stale.
```

[00:00:12.048,000] <inf> i3c_m: Max Read Length: 11450
[00:00:12.048,000] <dbg> npcx_i3c: npcx_i3c_ibi_enable: npcx_i3c_ibi_enable bcr0 dcr0
[00:00:12.048,000] <err> npcx_i3c: npcx_i3c_ibi_enable: device is not ibi capable
```
Even though I3C device is IBI capable

closed 2024-12-16 by github-actions[bot]