zephyrproject-rtos/zephyr #78619: Remove CONFIG_PINCTRL from all defconfig files

nordicjm, 2024-09-18:
**Describe the bug**
This Kconfig has wrongly been added to defconfig files, it is not the right place for it, it has never been the right place for it, drivers that need it should select the symbol in their Kconfig entries. The following boards do this wrongly and need to be fixed:

[ ] boards/96boards/meerkat96/96b_meerkat96_mcimx7d_m4_defconfig

[ ] boards/arduino/nicla_sense_me/arduino_nicla_sense_me_defconfig

@soburi
[ ] boards/arduino/uno_r4/arduino_uno_r4_minima_defconfig
[ ] boards/arduino/uno_r4/arduino_uno_r4_wifi_defconfig

@jadonk
[ ] boards/beagle/beagleconnect_freedom/beagleconnect_freedom_defconfig

[ ] boards/digilent/zybo/zybo_defconfig

[ ] boards/element14/warp7/warp7_mcimx7d_m4_defconfig

[ ] boards/enclustra/mercury_xu/mercury_xu_defconfig

[ ] boards/ezurio/rm1xx_dvk/rm1xx_dvk_defconfig

@nandojve
[ ] boards/gd/gd32a503v_eval/doc/index.rst
[ ] boards/gd/gd32f350r_eval/doc/index.rst
[ ] boards/gd/gd32l233r_eval/doc/index.rst

@ifyall
[ ] boards/infineon/cy8ckit_062s4/cy8ckit_062s4_defconfig
[ ] boards/infineon/cy8cproto_062_4343w/cy8cproto_062_4343w_defconfig
[ ] boards/infineon/cy8cproto_063_ble/cy8cproto_063_ble_defconfig
[ ] boards/infineon/cyw920829m2evk_02/cyw920829m2evk_02_defconfig

[ ] boards/madmachine/mm_feather/mm_feather_defconfig
[ ] boards/madmachine/mm_swiftio/mm_swiftio_defconfig

@jvasanth1
[ ] boards/microchip/mec1501modular_assy6885/mec1501modular_assy6885_defconfig
[ ] boards/microchip/mec1501modular_assy6885/mec1501modular_assy6885_defconfig
[ ] boards/microchip/mec15xxevb_assy6853/mec15xxevb_assy6853_defconfig
[ ] boards/microchip/mec15xxevb_assy6853/mec15xxevb_assy6853_defconfig
[ ] boards/microchip/mec172xevb_assy6906/mec172xevb_assy6906_defconfig
[ ] boards/microchip/mec172xmodular_assy6930/mec172xmodular_assy6930_defconfig

[ ] boards/mikroe/clicker_ra4m1/mikroe_clicker_ra4m1_defconfig

@nordicjm
[ ] boards/nordic/nrf7002dk/nrf7002dk_nrf5340_cpuapp_defconfig
[ ] boards/nordic/nrf7002dk/nrf7002dk_nrf5340_cpuapp_nrf7001_defconfig
[ ] boards/nordic/nrf7002dk/nrf7002dk_nrf5340_cpuapp_nrf7001_ns_defconfig
[ ] boards/nordic/nrf7002dk/nrf7002dk_nrf5340_cpuapp_ns_defconfig
[ ] boards/nordic/nrf7002dk/nrf7002dk_nrf5340_cpunet_defconfig
[ ] boards/nordic/nrf9131ek/nrf9131ek_nrf9131_defconfig
[ ] boards/nordic/nrf9131ek/nrf9131ek_nrf9131_ns_defconfig

@MulinChao @ChiHuaL
[ ] boards/nuvoton/npcx4m8f_evb/npcx4m8f_evb_defconfig
[ ] boards/nuvoton/npcx7m6fb_evb/npcx7m6fb_evb_defconfig
[ ] boards/nuvoton/npcx9m6f_evb/npcx9m6f_evb_defconfig

@cyliangtw
[ ] boards/nuvoton/numaker_m2l31ki/numaker_m2l31ki_defconfig
[ ] boards/nuvoton/numaker_pfm_m467/numaker_pfm_m467_defconfig
[ ] boards/nuvoton/numaker_pfm_m487/numaker_pfm_m487_defconfig

@decsny 
[ ] boards/nxp/frdm_k22f/frdm_k22f_defconfig
[ ] boards/nxp/frdm_k64f/frdm_k64f_defconfig
[ ] boards/nxp/frdm_k82f/frdm_k82f_defconfig
[ ] boards/nxp/frdm_ke15z/frdm_ke15z_defconfig
[... 113 lines cut ...]

@fkokosinski @kgugala @tgorochowik
[ ] boards/sparkfun/red_v_things_plus/sparkfun_red_v_things_plus_defconfig

[ ] boards/sparkfun/thing_plus_matter_mgm240p/sparkfun_thing_plus_matter_mgm240p_defconfig

[ ] boards/technexion/pico_pi/pico_pi_mcimx7d_m4_defconfig

@vaishnavachath
[ ] boards/ti/cc1352p1_launchxl/cc1352p1_launchxl_defconfig
[ ] boards/ti/cc1352p7_launchpad/cc1352p7_lp_defconfig
[ ] boards/ti/cc1352r1_launchxl/cc1352r1_launchxl_defconfig
[ ] boards/ti/cc1352r_sensortag/cc1352r_sensortag_defconfig
[ ] boards/ti/cc26x2r1_launchxl/cc26x2r1_launchxl_defconfig
[ ] boards/ti/sk_am62/sk_am62_am6234_m4_defconfig

[ ] boards/toradex/colibri_imx7d/colibri_imx7d_mcimx7d_m4_defconfig
[ ] boards/toradex/verdin_imx8mp/verdin_imx8mp_mimx8ml8_m7_ddr_defconfig
[ ] boards/toradex/verdin_imx8mp/verdin_imx8mp_mimx8ml8_m7_defconfig

[ ] boards/udoo/udoo_neo_full/udoo_neo_full_mcimx6x_m4_defconfig

[ ] boards/vcc-gnd/yd_stm32h750vb/yd_stm32h750vb_defconfig

Please submit fixes then tick once they have been merged.

billwatersiii, 2024-09-18:
These boards are not owned by Infineon (ifyall):

boards/madmachine/mm_feather/mm_feather_defconfig
boards/madmachine/mm_swiftio/mm_swiftio_defconfig

nordicjm, 2024-09-19:
> These boards are not owned by Infineon (ifyall):
> 
> boards/madmachine/mm_feather/mm_feather_defconfig boards/madmachine/mm_swiftio/mm_swiftio_defconfig

There is actually a gap there but for some reason github is added a gap between the infineon ones too, but if you edit the post it will show no line gaps for the infineon boards but one after before the madmachine ones

nordicjm, 2024-10-10:
A whole lot more:

@nandojve
- [ ] soc/atmel/sam/Kconfig.defconfig
- [ ] soc/atmel/sam0/Kconfig.defconfig
___
@nandojve
- [ ] soc/gd/gd32/Kconfig.defconfig
___
@ifyall
- [ ] soc/infineon/cat3/Kconfig.defconfig
___
@Dino-Li @GTLin08 @RuibinChang
- [ ] soc/ite/ec/it8xxx2/Kconfig.defconfig.series
___
@jvasanth1 @VenkatKotakonda @albertofloyd
- [ ] soc/microchip/mec/mec174x/Kconfig.defconfig.mec1743qlj
- [ ] soc/microchip/mec/mec174x/Kconfig.defconfig.mec1743qsz
- [ ] soc/microchip/mec/mec175x/Kconfig.defconfig.mec1753qlj
- [ ] soc/microchip/mec/mec175x/Kconfig.defconfig.mec1753qsz
- [ ] soc/microchip/mec/mech172x/Kconfig.defconfig.mech1723nlj
- [ ] soc/microchip/mec/mech172x/Kconfig.defconfig.mech1723nsz
___
@yonsch
- [ ] soc/raspberrypi/Kconfig.defconfig
___
@soburi @KhiemNguyenT
- [ ] soc/renesas/ra/ra2a1/Kconfig.defconfig
- [ ] soc/renesas/ra/ra4e2/Kconfig.defconfig
- [ ] soc/renesas/ra/ra4m2/Kconfig.defconfig
- [ ] soc/renesas/ra/ra4m3/Kconfig.defconfig
- [ ] soc/renesas/ra/ra4w1/Kconfig.defconfig
- [ ] soc/renesas/ra/ra6e1/Kconfig.defconfig
- [ ] soc/renesas/ra/ra6e2/Kconfig.defconfig
- [ ] soc/renesas/ra/ra6m1/Kconfig.defconfig
- [ ] soc/renesas/ra/ra6m2/Kconfig.defconfig
- [ ] soc/renesas/ra/ra6m3/Kconfig.defconfig
- [ ] soc/renesas/ra/ra6m4/Kconfig.defconfig
- [ ] soc/renesas/ra/ra6m5/Kconfig.defconfig
- [ ] soc/renesas/ra/ra8d1/Kconfig.defconfig
- [ ] soc/renesas/ra/ra8m1/Kconfig.defconfig
- [ ] soc/renesas/ra/ra8t1/Kconfig.defconfig
___
@aaillet @lorc
- [ ] soc/renesas/rcar/rcar_gen3/Kconfig.defconfig
- [ ] soc/renesas/rcar/rcar_gen4/Kconfig.defconfig.r8a779f0
___
@andy-liu-telink
- [ ] soc/telink/tlsr/tlsr951x/Kconfig.defconfig
___
@vaishnavachath
- [ ] soc/ti/k3/am6x/Kconfig.defconfig

nordicjm, 2024-10-31:
@MulinChao @ChiHuaL @cyliangtw @jhedberg @ruuddw @evgeniy-paltsev @nandojve @ifyall @jvasanth1 @VenkatKotakonda @albertofloyd @soburi @KhiemNguyenT @aaillet @lorc @andy-liu-telink @vaishnavachath you all have boards/socs here with this issue, can you submit fixes for them to select PINCTRL properly?

soburi, 2024-10-31:
> @MulinChao @ChiHuaL @cyliangtw @jhedberg @ruuddw @evgeniy-paltsev @nandojve @ifyall @jvasanth1 @VenkatKotakonda @albertofloyd @soburi @KhiemNguyenT @aaillet @lorc @andy-liu-telink @vaishnavachath [...]

My part, 

> @soburi https://github.com/zephyrproject-rtos/zephyr/pull/76794
> 
> boards/arduino/uno_r4/arduino_uno_r4_minima_defconfig
> boards/arduino/uno_r4/arduino_uno_r4_wifi_defconfig

This is now waiting to merge.

> @soburi @KhiemNguyenT
> 
> soc/renesas/ra/ra2a1/Kconfig.defconfig
> ...

@KhiemNguyenT

First, Could the Renesas team please handle it?

ifyall, 2024-10-31:
@soburi, I have marked the Infineon boards as done. They were handled in https://github.com/zephyrproject-rtos/zephyr/pull/78734

soburi, 2024-10-31:
@ifyall 
> @soburi, I have marked the Infineon boards as done. They were handled in #78734

Thank you for the update.

@nordicjm 
Please confirm it.

nordicjm, 2024-11-01:
@ifyall seems `soc/infineon/cat3/Kconfig.defconfig` has not been done

rruuaanng, 2024-11-19:
I've commit a PR about openisa. Maybe we can add me to task list ;)

nordicjm, 2024-11-19:
Pinging on boards/socs that have had no response or fix posted:

@jvasanth1

boards/microchip/mec1501modular_assy6885/mec1501modular_assy6885_defconfig
boards/microchip/mec1501modular_assy6885/mec1501modular_assy6885_defconfig
boards/microchip/mec15xxevb_assy6853/mec15xxevb_assy6853_defconfig
boards/microchip/mec15xxevb_assy6853/mec15xxevb_assy6853_defconfig
boards/microchip/mec172xevb_assy6906/mec172xevb_assy6906_defconfig
boards/microchip/mec172xmodular_assy6930/mec172xmodular_assy6930_defconfig

@jhedberg

boards/silabs/dev_kits/sltb010a/sltb010a_defconfig
boards/silabs/dev_kits/xg24_dk2601b/xg24_dk2601b_defconfig
boards/silabs/dev_kits/xg27_dk2602a/xg27_dk2602a_defconfig
boards/silabs/radio_boards/slwrb4180a/slwrb4180a_defconfig
boards/silabs/radio_boards/xg24_rb4187c/xg24_rb4187c_defconfig

@ruuddw @evgeniy-paltsev

boards/snps/emsdp/emsdp_emsdp_em11d_defconfig

@vaishnavachath

boards/ti/cc1352p1_launchxl/cc1352p1_launchxl_defconfig
boards/ti/cc1352r1_launchxl/cc1352r1_launchxl_defconfig
boards/ti/cc1352r_sensortag/cc1352r_sensortag_defconfig
boards/ti/cc26x2r1_launchxl/cc26x2r1_launchxl_defconfig
boards/ti/sk_am62/sk_am62_am6234_m4_defconfig
soc/ti/k3/am6x/Kconfig.defconfig

@ifyall

soc/infineon/cat3/Kconfig.defconfig

@jvasanth1 @VenkatKotakonda @albertofloyd

soc/microchip/mec/mec174x/Kconfig.defconfig.mec1743qlj
soc/microchip/mec/mec174x/Kconfig.defconfig.mec1743qsz
soc/microchip/mec/mec175x/Kconfig.defconfig.mec1753qlj
soc/microchip/mec/mec175x/Kconfig.defconfig.mec1753qsz
soc/microchip/mec/mech172x/Kconfig.defconfig.mech1723nlj
soc/microchip/mec/mech172x/Kconfig.defconfig.mech1723nsz

@aaillet @lorc

soc/renesas/rcar/rcar_gen3/Kconfig.defconfig
soc/renesas/rcar/rcar_gen4/Kconfig.defconfig.r8a779f0

@andy-liu-telink

soc/telink/tlsr/tlsr951x/Kconfig.defconfig

The target for this is zephyr 4.1, please submit fixes in the short term future for this.

fabiobaltieri, 2024-11-19:
Arch WG:
- summarized the issue, asking for response from the board maintainers
- @erwango asked if this is going to have to done for CONFIG_GPIO as well
  - the pinctrl driver for stm32 depends on gpio, so every board needs to enable gpio to use pinctrl

tagunil, 2024-11-22:
> > > boards/snps/emsdp/emsdp_emsdp_em11d_defconfig
> > 
> > 
> > Hi @nordicjm, What should we do if our SoCs use generic drivers like spi_dw or uart_ns16550, which might require PINCTRL depending on specific hardware configuration of these IPs? We'd really like to avoid making PINCTRL a hard requirement for the drivers, keeping the dependency somewhere between the SoC level and the board level.
> 
> `select PINCTRL if ...`

Do you mean listing the SoCs in the Kconfig of the driver? Wouldn't such a list grow uncontrollably for drivers used by many different SoCs from different vendors, like ns16550?

gmarull, 2024-11-22:
> Do you mean listing the SoCs in the Kconfig of the driver? Wouldn't such a list grow uncontrollably for drivers used by many different SoCs from different vendors, like ns16550?

you can always do `select PINCTRL if !...` I guess they'll be the exception and not the rule. After all, these generic APIs (like pinctrl) exist to facilitate generic IP drivers.

jhedberg, 2024-11-22:
Considering that we have a bunch of Kconfig macros for DT, I wonder if it'd be possible to use those to detect if a DT node with the ns16550 compat has a `pinctrl-N` property, and if so `select PINCTRL`?

jhedberg, 2024-11-22:
> Considering that we have a bunch of Kconfig macros for DT, I wonder if it'd be possible to use those to detect if a DT node with the ns16550 compat has a `pinctrl-N` property, and if so `select PINCTRL`?

What I mean is, something like this:
```
select PINCTRL if $(dt_compat_any_has_prop,$(DT_COMPAT_NS16550),pinctrl-0)
```

PR #79223 (2024-10-05): boards: mikroe: remove CONFIG_PINCTRL from defconfig of RA4M1 clicker
This PR fixes #78619 for the Mikroe RA4M1 Clicker board.
commit 627c3329 (2024-10-05): boards: mikroe: remove CONFIG_PINCTRL from defconfig of RA4M1 clicker

This PR fixes #78619 for the Mikroe RA4M1 Clicker board.
PR #79424 (2024-10-08): da1469x: Remove CONFIG_PINCTRL from all defconfig files
Remove CONFIG_PINCTRL from all defconfig files.

Fixes #78619
commit 6c6a1e55 (2024-10-08): da1469x: Remove CONFIG_PINCTRL from all defconfig files

Remove CONFIG_PINCTRL from all defconfig files.

Fixes #78619
closed 2024-12-06 by kartben