zephyrproject-rtos/zephyr #78638: "CONFIG_NET_CONFIG_CLOCK_SNTP_INIT" leads to POSIX library error

Jowllel, 2024-09-18:
**Describe the bug**
When trying to check the "CONFIG_NET_CONFIG_CLOCK_SNTP_INIT" option, the option "POSIX_CLOCK" has to be selected. The POSIX Clock option is deprecated. (POSIX_TIMERS should be used?) Additionally the compilation gives me the following error:

``` 
In file included from C:/Users/jschaller/zephyrproject/zephyr/lib/posix/options/clock.c:8:
C:/Users/jschaller/zephyrproject/zephyr/lib/posix/options/posix_clock.h:17:10: fatal error: zephyr/syscalls/posix_clock.h: No such file or directory
   17 | #include <zephyr/syscalls/posix_clock.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

**To Reproduce**

1. prj.conf:

```
CONFIG_GPIO=y

CONFIG_SPI=y
CONFIG_NET_L2_ETHERNET=y
CONFIG_ETH_W5500=y
CONFIG_ETH_W5500_TIMEOUT=1000
CONFIG_NET_TCP=y
CONFIG_ETH_DRIVER=y
CONFIG_NET_DHCPV4=y
CONFIG_NET_MGMT=y
CONFIG_NETWORKING=y
CONFIG_NET_IPV4=y
CONFIG_NET_DHCPV4_OPTION_CALLBACKS=y

CONFIG_TEST_RANDOM_GENERATOR=y

CONFIG_LOG=y
CONFIG_RTC=y
CONFIG_RTC_LOG_LEVEL_DBG=y

CONFIG_THREAD_ANALYZER=y
CONFIG_THREAD_ANALYZER_USE_LOG=y
CONFIG_THREAD_ANALYZER_AUTO=n
CONFIG_THREAD_ANALYZER_AUTO_INTERVAL=60
CONFIG_THREAD_ANALYZER_AUTO_STACK_SIZE= 2048
CONFIG_THREAD_NAME=y

CONFIG_NET_CONFIG_CLOCK_SNTP_INIT=y
CONFIG_NET_SOCKETS=y
CONFIG_SNTP=y
CONFIG_POSIX_CLOCK=y
```

2. Board: nucleo_f401re

**Environment:**

 - OS: Windows
 - Dev: CLion / West
 - Zephyr Version: 3.7.99

rlubos, 2024-09-19:
@Jowllel It seems this has already been adressed? See https://github.com/zephyrproject-rtos/zephyr/pull/78003

Jowllel, 2024-09-19:
Hi, Thanks for the reply. But this only fixes the POSIX_CLOCK deprecated problem.
I still got the error with the file that can't be found: #include <zephyr/syscalls/posix_clock.h>
But it worked after deleting the Build folder and rebuilding all the files.

rlubos, 2024-09-19:
> I still got the error with the file that can't be found: #include <zephyr/syscalls/posix_clock.h>
But it worked after deleting the Build folder and rebuilding all the files.

So are you confirming the issue is fixed? Not an expert on those posix configs, but it could be the case that you simply needed pristine build after changing configurations.

closed 2024-09-19 by Jowllel

valcarcexyz, 2024-10-02:
Error appeared and as @Jowllel said, a pristine build solved the issue

tonyarkles, 2024-11-10:
I ran into the same issue on a brand new clean checkout of v3.7.0 (`west init --mr v3.7.0`). Started my project, did a quick build to make sure my minimal `main.c` would build and flash, and then added a bunch of options to `prj.conf` including `CONFIG_POSIX_API=y` and `CONFIG_POSIX_TIMERS=y`. Trying to build again was broken until I did a pristine build with `west build -b mimxrt1024_evk -p always`.