zephyrproject-rtos/zephyr #78668: Missing header in `subsys/shell/modules/kernel_service/heap.c`

gminn, 2024-09-18:
**Describe the bug**

The new file `subsys/shell/modules/kernel_service/heap.c` added in https://github.com/zephyrproject-rtos/zephyr/pull/77668 is missing `#include "kernel_shell.h"`, resulting in a build failure when this file is included:

```
/Users/gminnehan/projects/zephyr-rtos-workspace/zephyr/subsys/shell/modules/kernel_service/heap.c:13:41: warning: 'struct shell' declared inside parameter list will not be visible outside of this definition or declaration
   13 | static int cmd_kernel_heap(const struct shell *sh, size_t argc, char **argv)
      |                                         ^~~~~
/Users/gminnehan/projects/zephyr-rtos-workspace/zephyr/subsys/shell/modules/kernel_service/heap.c: In function 'cmd_kernel_heap':
/Users/gminnehan/projects/zephyr-rtos-workspace/zephyr/subsys/shell/modules/kernel_service/heap.c:23:17: warning: implicit declaration of function 'shell_error' [-Wimplicit-function-declaration]
   23 |                 shell_error(sh, "Failed to read kernel system heap statistics (err %d)", err);
      |                 ^~~~~~~~~~~
/Users/gminnehan/projects/zephyr-rtos-workspace/zephyr/subsys/shell/modules/kernel_service/heap.c:24:25: error: 'ENOEXEC' undeclared (first use in this function)
   24 |                 return -ENOEXEC;
      |                         ^~~~~~~
/Users/gminnehan/projects/zephyr-rtos-workspace/zephyr/subsys/shell/modules/kernel_service/heap.c:24:25: note: each undeclared identifier is reported only once for each function it appears in
/Users/gminnehan/projects/zephyr-rtos-workspace/zephyr/subsys/shell/modules/kernel_service/heap.c:27:9: warning: implicit declaration of function 'shell_print' [-Wimplicit-function-declaration]
   27 |         shell_print(sh, "free:           %zu", stats.free_bytes);
      |         ^~~~~~~~~~~
/Users/gminnehan/projects/zephyr-rtos-workspace/zephyr/subsys/shell/modules/kernel_service/heap.c: At top level:
/Users/gminnehan/projects/zephyr-rtos-workspace/zephyr/subsys/shell/modules/kernel_service/heap.c:34:21: error: expected ')' before '(' token
   34 | KERNEL_CMD_ADD(heap, NULL, "System heap usage statistics.", cmd_kernel_heap);
      |                     ^
      |                     )
/Users/gminnehan/projects/zephyr-rtos-workspace/zephyr/subsys/shell/modules/kernel_service/heap.c:13:12: warning: 'cmd_kernel_heap' defined but not used [-Wunused-function]
   13 | static int cmd_kernel_heap(const struct shell *sh, size_t argc, char **argv)
      |            ^~~~~~~~~~~~~~~
```

**To Reproduce**

Build a sample with shell enabled, heap > 0, and heap runtime stats enabled:

```
west build -b qemu_cortex_m3 zephyr/samples/hello_world -- -DCONFIG_SYS_HEAP_RUNTIME_STATS=y -DCONFIG_SHELL=y -DCONFIG_HEAP_MEM_POOL_SIZE=30000
```

**Expected behavior**

Successful build with the above settings. Additionally, this looks to be missing coverage in CI. 

**Impact**

Annoyance, currently a similar sample app used in CI tests is failing to build.

**Logs and console output**

N/A

**Environment (please complete the following information):**

- OS: MacOS
- Toolchain: Zephyr SDK
- Testing on Zephyr latest. At the time of writing, tests were run on 9d9089edd09

**Additional context**

N/A

gminn, 2024-09-18:
cc @ycsin for your awareness!

ycsin, 2024-09-18:
> cc @ycsin for your awareness!

Thanks for the clear and concise report, fix is in #78670

PR #78670 (2024-09-19): shell: kernel_service: heap: add required header
Add the "kernel_shell.h" header which is required but missing, causing build error.

Fixes #78668
closed 2024-09-19 by nashif