zephyrproject-rtos/zephyr #78676: NXP MIMXRT1160-EVK regression issue, apps do not boot properly

DerekSnell, 2024-09-18:
Commit d1a6b453455d6964ea1cfc200e241f4a3f5c4abd updated the bus clock for RT1160, and now `hello_world` and other apps do not boot properly.  I am using RevA3 of the MIMXRT1160-EVK board.  The MIMXRT1170-EVK apps work fine for me.  

To replicate, `hello_world` is built with the command below.  After flashing, nothing prints out the debug console.
```
west build -b mimxrt1160_evk//cm7 samples/hello_world/ --pristine
```
Also, this clock change interferes with the debug connection, and JLink cannot connect to the MCU.  To resolve it, I force the boot ROM into Serial Download mode by setting the DIP switches SW1 to Off-Off-Off-On.  Then press the reset button SW3 to boot in Serial Download mode.  Program the flash with a working app, or erase the flash.  Revert SW5 back to Off-Off-On-Off to boot from flash, and press SW3.  Now the debugger can connect.

@PetervdPerk-NXP, FYI

dleach02, 2024-09-19:
ping @hakehuang

hakehuang, 2024-09-20:
> ping @hakehuang

@dleach02 sorry it is a miss check at my side, I initially disable it for Jlink FCB issue, but I forget to enable it in after the a46fb055bc1e1f9a19560b56155c0ceeb16f840e get merged. the issue fixed now, CI will include this board

PR #78735 (2024-09-23): soc: nxp: rt116x: Fix bus clocking
Fixes #78676, Reverts bus clock settings.  Follows MCUXpresso SDK clock settings, and sets to output of SysPLL2 PFD3 at 198 MHz.
closed 2024-09-23 by nashif