zephyrproject-rtos/zephyr #78724: net: checksum offloading not working for virtual network interfaces used for Virtual LAN (VLAN)

stomschik-ask, 2024-09-19:
**Describe the bug**
The checksum offloading capabilities of an Ethernet interface are not respected for virtual network interfaces as the function `need_calc_checksum()` in "net_if.c" always returns true for the virtual interfaces. This increases CPU utilization as checksums are calculated in software even though they would be automatically calculated by the Ethernet MAC. Even worse, some Ethernet MACs require the checksum to be cleared to zero for TX checksum offloading as otherwise an incorrect checksum will be generated by the Ethernet MAC (for example [NXP MIMXRT1060-EVK](https://docs.zephyrproject.org/latest/boards/nxp/mimxrt1060_evk/doc/index.html)).

**To Reproduce**
Use the [Virtual LAN](https://docs.zephyrproject.org/latest/samples/net/vlan/README.html) sample
1. disable IPv6 as IPv6 will disable TX/RX checksum offloading on i.MX RT1060 (`CONFIG_NET_IPV6=n`)
2. west build --pristine -b mimxrt1060_evk samples/net/vlan
3. west flash
4. From PC, ping 192.0.2.1 (untagged, working)
5. From PC, ping 198.51.100.1 (VLAN 100, request timed out)
6. Incorrect checksum (0xffff) for IP header and ICMP message will be shown in Wireshark for the Echo reply

**Expected behavior**
TX/RX checksum offloading should also work on virtual network interfaces used for Virtual LAN (VLAN).

**Impact**
TX/RX checksum offloading can be disabled for i.MX RT1060 by setting `CONFIG_ETH_NXP_ENET_HW_ACCELERATION=n`. Therefore, impact is mainly the increased CPU utilization.

**Environment:**
 - OS: Windows
 - Zephyr SDK 0.16.8
 - Zephyr 3.7.0

jukkar, 2024-09-20:
@stomschik-ask I have a fix proposal at # #78745. Would you be able to check if it works for you?

stomschik-ask, 2024-09-23:
Thanks, the fix is working fine for me.

PR #78745 (2024-09-24): net: if: Check chksum offloading properly for VLAN interfaces
Make sure we check the checksum offloading capabilities correctly for VLAN interfaces. Use the real Ethernet interface when doing the check.

Fixes #78724
commit 00bb90a4 (2024-09-24): net: if: Check chksum offloading properly for VLAN interfaces

Make sure we check the checksum offloading capabilities correctly
for VLAN interfaces. Use the real Ethernet interface when doing the
check.

Fixes #78724
commit a62c7417 (2024-09-26): net: if: Check chksum offloading properly for VLAN interfaces

Make sure we check the checksum offloading capabilities correctly
for VLAN interfaces. Use the real Ethernet interface when doing the
check.

Fixes #78724

(cherry picked from commit 00bb90a42e7f583704090c5e38e1b3fd933cf409)
closed 2024-09-24 by aescolar