zephyrproject-rtos/zephyr #78729: LittleFS fails samples silently fails for NXP MIMXRT1020-EVK board

luispimo, 2024-09-19:
UART samples work correctly, but all the LittleFS samples  fail on this board (eg `samples/subsys/shell/fs`).

However, a few months ago, everything worked correctly.

**To Reproduce**

```
west build -p always -b mimxrt1020_evk samples/subsys/shell/fs
...
[182/182] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       97992 B         8 MB      1.17%
             RAM:       31200 B        32 MB      0.09%
            ITCM:        6232 B        64 KB      9.51%
            DTCM:          0 GB        64 KB      0.00%
           OCRAM:          0 GB       128 KB      0.00%
        IDT_LIST:          0 GB        32 KB      0.00%
```
```
west flash -r jlink
-- west flash: rebuilding
ninja: no work to do.
-- west flash: using runner jlink
-- runners.jlink: reset after flashing requested
-- runners.jlink: JLink version: 7.96f
-- runners.jlink: Flashing file: /Users/luis/Dev/zephyr-workspace/zephyr/build/zephyr/zephyr.bin
```

Nothing appears on the serial console, not even the boot message. 

The debugger exists due to a signal.

nixward, 2024-09-22:
You could try git bisecting this @luispimo to find the feature breaking commit.

luispimo, 2024-09-24:
Apparently, the problem is not in Zephyr itself, as I have tried versions 3.5, 3.6, and 3.7, obtaining the same result. It must be in the NXP hal module or in the littlefs module.

nakato, 2024-10-08:
I wonder if I've been hitting the same bug.  I had created a discussion about it at https://github.com/zephyrproject-rtos/zephyr/discussions/79033

I trigger it a bit differently, using fstab automount, and I have managed to trigger the issue on both a RP2040 and NRF52840 based board.  But, it does present the same, as the lockup during mkfs happens so quickly there's no output unless you start putting delays in.

I've found I can get logs out of the console on boot by inserting a healthy amount of `k_sleep(K_MSEC(1000))` into the littlefs format functions, but I've yet to work out what is going on.  I can put a log message and sleep on line 1991, and get it, but a log message and sleep at the very top of the `lfs_dir_traverse` never shows up, which is called at [1993](https://github.com/zephyrproject-rtos/littlefs/blob/zephyr/lfs.c#L1993) during mkfs, which I don't get how that happens.

luispimo, 2024-10-08:
Downgrading the LittleFS module to 652f2c5 does not work for the mimxrt1020_evk board when running LittleFS examples.

luispimo, 2024-10-09:
Update: Downgrading hal_nxp to d45b14c1 solves the problem and littles work again.

danieldegrasse, 2024-10-17:
I believe this is the HAL module, as you identified- the flash configuration block for the RT1020 is now programming the `P[4]` bit of the read parameter bits in the attached flash device, which changes the number of dummy cycles needed for read operations, and breaks Zephyr's flash driver. Can you try to reproduce this with #80016?

luispimo, 2024-10-18:
I applied PR 80017, but the issue still remains.

luispimo, 2024-10-18:
For the Teensy 41 (flash=W25Q16JVUXIM) board, the main branch works correctly without applying the PR. However, for the MIMXRT1020-EVK (flash=IS25WP064), neither of the branches work.

danieldegrasse, 2024-10-28:
Closing as #80016 has merged, @luispimo please feel free to ping if this should be reopened

closed 2024-10-28 by danieldegrasse