zephyrproject-rtos/zephyr #78762: NPCX4M8F: tests/kernel/sched/schedule_api/kernel.scheduler fail

KamilxPaszkiet, 2024-09-20:
**Describe the bug**
Test from:
```
tests/kernel/sched/schedule_api/kernel.scheduler
tests/kernel/sched/schedule_api/kernel.scheduler.dumb_timeslicing
tests/kernel/sched/schedule_api/kernel.scheduler.multiq 
tests/kernel/sched/schedule_api/kernel.scheduler.slice_perthread
```
 failing with timeout

**To Reproduce**
twister --device-testing --hardware-map npcx4m8f.map -T tests/kernel/sched/schedule_api/ -vv

**Expected behavior**
Test pass

**Logs and console output**
kernel.scheduler.multiq log:
```
*** Booting Zephyr OS build v3.7.0-3117-gbd03883744d4 (delayed boot 500ms) ***
Running TESTSUITE threads_scheduling
===================================================================
START - test_bad_priorities
 PASS - test_bad_priorities in 0.002 seconds
===================================================================
START - test_busy_wait_cooperative
 PASS - test_busy_wait_cooperative in 0.105 seconds
===================================================================
START - test_k_thread_priority_get_init_null
E: 0 is not a valid k_thread
E: address is not a known kernel object
E: syscall z_vrfy_k_thread_priority_get failed check: access denied
E: r0/a1:  0x00000000  r1/a2:  0x00000000  r2/a3:  0x00000000
E: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x00000000
E:  xpsr:  0x00000000
E: Faulting instruction address (r15/pc): 0xaaaaaaaa
E: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
E: Current thread: 0x200c1e30 (unknown)
Caught system error -- reason 3 1
Fatal error expected as part of test case.
 PASS - test_k_thread_priority_get_init_null in 0.047 seconds
===================================================================
START - test_k_thread_priority_set_init_null
E: 0 is not a valid k_thread
E: address is not a known kernel object
E: syscall z_vrfy_k_thread_priority_set failed check: access denied
E: r0/a1:  0x00000000  r1/a2:  0x00000000  r2/a3:  0x00000000
E: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x00000000
E:  xpsr:  0x00000000
E: Faulting instruction address (r15/pc): 0xaaaaaaaa
E: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
E: Current thread: 0x200c1e30 (unknown)
Caught system error -- reason 3 1
Fatal error expected as part of test case.
 PASS - test_k_thread_priority_set_init_null in 0.047 seconds
===================================================================
START - test_k_thread_priority_set_overmax
E: syscall z_vrfy_k_thread_priority_set failed check: invalid thread priority 40
E: r0/a1:  0x00000000  r1/a2:  0x00000000  r2/a3:  0x00000000
E: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x00000000
E:  xpsr:  0x00000000
E: Faulting instruction address (r15/pc): 0xaaaaaaaa
E: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
E: Current thread: 0x200c1e30 (unknown)
Caught system error -- reason 3 1
Fatal error expected as part of test case.
 PASS - test_k_thread_priority_set_overmax in 0.042 seconds
===================================================================
START - test_k_thread_priority_set_upgrade
E: syscall z_vrfy_k_thread_priority_set failed check: thread priority may only be downgraded (-1 < 0)
[... 147 lines cut ...]
 - PASS - [threads_scheduling.test_sleep_wakeup_preemptible] duration = 0.005 seconds
 - SKIP - [threads_scheduling.test_slice_perthread] duration = 0.002 seconds
 - FAIL - [threads_scheduling.test_slice_reset] duration = 0.525 seconds
 - PASS - [threads_scheduling.test_slice_scheduling] duration = 13.056 seconds
 - PASS - [threads_scheduling.test_time_slicing_disable_preemptible] duration = 0.505 seconds
 - PASS - [threads_scheduling.test_time_slicing_preemptible] duration = 0.504 seconds
 - PASS - [threads_scheduling.test_unlock_nested_sched_lock] duration = 0.105 seconds
 - PASS - [threads_scheduling.test_unlock_preemptible] duration = 0.105 seconds
 - PASS - [threads_scheduling.test_user_k_is_preempt] duration = 0.007 seconds
 - PASS - [threads_scheduling.test_user_k_wakeup] duration = 0.005 seconds
 - PASS - [threads_scheduling.test_wakeup_expired_timer_thread] duration = 0.003 seconds
 - PASS - [threads_scheduling.test_yield_cooperative] duration = 0.005 seconds
SUITE PASS - 100.00% [threads_scheduling_1cpu]: pass = 1, fail = 0, skip = 0, total = 1 duration = 0.039 seconds
 - PASS - [threads_scheduling_1cpu.test_priority_preemptible_wait_prio] duration = 0.039 seconds
------ TESTSUITE SUMMARY END ------
===================================================================
RunID: d917088e144c3ad1ae132f626f818d70
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**

 - OS: Linux
 - Zephyr SDK 0.16.8
 - Zephyr 3.7.0

KamilxPaszkiet, 2025-01-22:
@MulinChao kindly reminder

closed 2025-04-07 by github-actions[bot]

SzymonRichert, 2025-04-09:
@ChiHuaL you recently helped to close very old problem https://github.com/zephyrproject-rtos/zephyr/issues/66185 . Maybe you could also help with this bug ?