zephyrproject-rtos/zephyr #78816: stm32f4 sdmmc sdio error clock is not 48MHz

fariouche, 2024-09-22:
**Describe the bug**
I'm trying to use the sdcard port of my stm32f412g-disco board.
For that, I enabled the sdmmc via a dts overlay and relevant CONFIGs and when mounting I'm getting an error:
<err> stm32_sdmmc: SDMMC Clock is not 48MHz (100000000)

When I add CONFIG_SDMMC_STM32_CLOCK_CHECK=n, everything works as expected.

I tried to enable the clk_hsi48 node, but it does not exist in the stm32f4 device tree.
Is it an error I should worry? Is my workaround the correct way to fix this error?

**To Reproduce**
My overlay:
&sdmmc1 {
    status = "okay";

    compatible = "st,stm32-sdmmc";
        pinctrl-0 = <&sdio_d0_pc8 &sdio_d1_pc9
                                 &sdio_d2_pc10 &sdio_d3_pc11
                                 &sdio_ck_pc12 &sdio_cmd_pd2>;
        cd-gpios = <&gpiod 3 GPIO_ACTIVE_LOW>;
        pinctrl-names = "default";
};

My prj.conf
CONFIG_DISK_ACCESS=y
CONFIG_DISK_DRIVER_SDMMC=y
CONFIG_SDMMC_STM32=y
CONFIG_FAT_FILESYSTEM_ELM=y
CONFIG_FS_FATFS_LFN=y
CONFIG_FS_FATFS_MOUNT_MKFS=n
CONFIG_SDIO_STACK=y

I'm using zephyr 3.7 tag

**Expected behavior**
The fs_mount function should work with CONFIG_SDMMC_STM32_CLOCK_CHECK

gautierg-st, 2024-09-24:
The clock check is a copy of what is done for USB. Both USB and SDMMC are often clocked by a dedicated 48 MHz clock. It is mandatory for USB, though it doesn't seem to be the case with SDMMC so it may be superfluous.

For F412, the clock condition is this one according to the Reference Manual: 
`FrequencyPCLK2() > (3 x Width() / 32) ⁄ FrequencySDIO_CK()`

In your case, PCLK2 is 100MHz, Width is 1 (default value), and SDIO_CK is 50 MHz (100 MHz SYSCLK divided by SDMMC prescaler of 2), so the calculation matches and I think you can ignore this check.

As for the 48 MHz clock, the device tree misses two elements to make it work. There is a clock mux CK48MSEL that would be necessary, and also the PLLQ of PLLI2S (which means adding the Q channel in the dts bindings).

fariouche, 2024-09-24:
ok, thank you.
So if i understand it correctly, the sdmmc is working as is, but things are mising in the bindings to handle 48MHz clock.

Do we keep this issue open until this is solved?
From my point of view it works as is, even if it's maybe not as fast as possible maybe

FRASTM, 2024-11-28:
I have  added few lines of code  in the PR https://github.com/zephyrproject-rtos/zephyr/pull/82259
 to enable the sdmmc clock on the stm32f412/f413 
It requires clk48 to deliver a 48MHz clock, the PLLI2S to be configured if source for the clk48.
I tested on a stm32f413h_disco 
and I could retrieve the correct setting in the RCC PLLI2S and DCKCFGR2 registers.

closed 2024-12-13 by kartben