zephyrproject-rtos/zephyr #78821: why i2s3 is missing in stm32f412 dtsi?

fariouche, 2024-09-22:
**Is your enhancement proposal related to a problem? Please describe.**
I'm using the stm32f412g-disco and woulike lkke to use the i2s.
I know that the i2s is not written a supported in zephyr's doc of the board, but i was hopping that enabl8ng i2s3 would be enough.
But i2s3 is not defined in the dtsi. i2s1, i2s2, i2s4 and i2s5, but no i2s3... strange.

Any reason?
adding i2s3 (by copying it from a similar soc) will be enough?
I'm not yet sure about the value to put inside (interrupts, dma etc)

**Describe the solution you'd like**
enabling i2s3 and that's all

marwaiehm-st, 2024-09-23:
HI @fariouche, 

You can add this I2S3 to stm32f412.dtsi. It's similar to I2S3 stm32f401, but I added some configurations to ensure efficient and timely data transfer for the I2S peripheral:
```assembly
		i2s3: i2s@40003c00 {
			compatible = "st,stm32-i2s";
			#address-cells = <1>;
			#size-cells = <0>;
			reg = <0x40003c00 0x400>;
			clocks = <&rcc STM32_CLOCK_BUS_APB1 0x00008000>;
			interrupts = <51 5>;
			dmas = <&dma1 5 0 STM32_DMA_PERIPH_TX  | STM32_DMA_16BITS | \
					STM32_DMA_PRIORITY_HIGH)
		                &dma1 0 0 (STM32_DMA_PERIPH_RX | STM32_DMA_16BITS | \
					STM32_DMA_PRIORITY_HIGH)>;
			dma-names = "tx", "rx";
			status = "disabled";
		};
``` 
**Note**: To use I2S3, you must disable SPI3 to free up the shared resources for I2S3 to function correctly.

fariouche, 2024-09-23:
great, thank!
yes, it's true for all i2s/spi on stm32 right?

fariouche, 2024-09-24:
Is it normal that the dmas here have 4 fields, while the one I used in f401 has 5 fields (&dma1 5 0 0x400 0x3)?

fariouche, 2024-09-24:
I had to add STM32_DMA_FIFO_FULL at the end of the dmas and now it compiles

marwaiehm-st, 2024-09-24:
> Is it normal that the dmas here have 4 fields, while the one I used in f401 has 5 fields (&dma1 5 0 0x400 0x3)?

Yes, you are right. I forgot to add 0x3 at the end.

marwaiehm-st, 2024-09-24:
> great, thank! yes, it's true for all i2s/spi on stm32 right?

Yes, it is true for all I2S/SPI peripherals on STM32 microcontrollers that you need to disable the SPI functionality to use the I2S functionality. This is due to the shared hardware resources between these peripherals.

fariouche, 2024-09-24:
I will be able to test it soon, after i figure out why mcuboot is returning an error now all of a sudden after a fresh write 9f rhe image

marwaiehm-st, 2024-09-24:
> I will be able to test it soon, after i figure out why mcuboot is returning an error now all of a sudden after a fresh write 9f rhe image

Maybe you are flashing a corrupted image. Have you tried reflashing your board with an image generated from a basic sample like HelloWorld ?

fariouche, 2024-09-24:
ok, I'm missing a else in i2s_stm32_set_clock to handle case when pclk_len <= 1 commited last month.
With that patch, I have no more errors in config

fariouche, 2024-09-24:
Same....
in fact the only function I'm calling are configure, write, trigger_start, write write etc
And already right after trigger_start, I do not see any call to dma_tx_callback()
So something is probably not set properly I think (clocks? I've see an old issue talking about enabling PLLI2S clock, but this seems to be irrelevant now are this config was removed)

fariouche, 2024-09-24:
I'm moving forward. I added this:
&plli2s {
	div-m = <8>;
	mul-n = <192>;
	div-r = <3>;
	status = "okay";
};

(not sure about the value, taken from some other board)
Now dma_tx_callback() is called. Getting new errors now, but I need to check what they mean (queue_get returned an error from inside the callback and my loop didn't even had the time to push a new buffer (maybe my plli2s settings are wrong)

marwaiehm-st, 2024-09-24:
Yes the plli2s looks wrong, test wit these settings
```c
&plli2s{
        #clock-cells = <0>;
	clocks = <&clk_hse>;
	div-m = <4>;
	mul-n = <72>;
	div-r = <2>;
	status = "okay";
};
```
 On the CLOCKS of node i2s3 add this line to inicate that the I2S peripheral will use the clock generated by the PLLI2S_R output
```c
clocks = <&rcc STM32_CLOCK_BUS_APB1 0x00008000>,
	 <&rcc STM32_SRC_PLLI2S_R I2S_SEL(0)>;
```

fariouche, 2024-09-25:
Thank you; I have no more errors now, and the audio loop seems to work.
Still have no sound, but this is most probably because the stm32f412g-disco board uses a codec (WM8994ECS) that may need some i2c commands to set the volume (first time I deal with such codec)

Thanks for you help.
To sumup the needed modifications:

```
&plli2s{
        #clock-cells = <0>;
	clocks = <&clk_hse>;
	div-m = <4>;
	mul-n = <72>;
	div-r = <2>;
	status = "okay";
};
```

```
soc {
        i2s3: i2s@40003c00 {
	        compatible = "st,stm32-i2s";
	        #address-cells = <1>;
	        #size-cells = <0>;
	        reg = <0x40003c00 0x400>;
	        clocks = <&rcc STM32_CLOCK_BUS_APB1 0x00008000
	                  &rcc STM32_SRC_PLLI2S_R I2S_SEL(0)>;
	        interrupts = <51 5>;
	        dmas = <&dma1 5 0 (STM32_DMA_PERIPH_TX | STM32_DMA_16BITS | STM32_DMA_PRIORITY_HIGH) STM32_DMA_FIFO_FULL
		            &dma1 0 0 (STM32_DMA_PERIPH_RX | STM32_DMA_16BITS | STM32_DMA_PRIORITY_HIGH) STM32_DMA_FIFO_FULL>;
	        dma-names = "tx", "rx";
	        status = "disabled";
	        label = "i2s3";
        };
			
    };

```

And of course, for the application:
```
&i2s3 {
    status = "okay";
    pinctrl-0 = <&i2s3_ck_pb12           &i2s3_sd_pb5         &i2s3_ws_pa4&i2s3_mck_pc7   >;
    pinctrl-names = "default";
   mck-enabled;
};

```

No other modifications for now are needed for me. I will post here again as soon as I've fixed the codec issue

fariouche, 2024-09-27:
I got some news. So far, I still have no sound. Cannot say for sure if the audio codec is incorrectly configured or if the i2S output is incorrect.
I used my logic analyzer, and checked the signals of mck (too high for my cheap logic analyzer, frequency seems to be around 2MHz) but a slower signal like ws shows a period of 32us (31250Hz, this is strange)
I'm using the following config:
i2s_cfg.word_size = 16;
i2s_cfg.channels = 2;
i2s_cfg.format = I2S_FMT_DATA_FORMAT_I2S;
i2s_cfg.frame_clk_freq = 44100;

This may explain why I do not have sound. I'm still searching the root cause

fariouche, 2024-09-28:
I confirm the clocks: mck is 8MHz, ck is 1MHz and ws is 31250
ws is 29410 Hz if mck is disabled, ck around 961500

All values are correct in ratio, but the values are not correct (the plli2s is 72MHz)
I will check the datasheet to better understand what is happening

fariouche, 2024-09-30:
Some progress: For some reason, the clock_control_get_rate function returns 72MHz while in fact it should be 48Mhz. I'm not sure why yet, but forcing freq_in value to 48MHz gives me correct clocks

marwaiehm-st, 2024-09-30:
You are rigth because the 48 MHz clock frequency is often used because it is a common multiple of standard audio sample rates (e.g., 8 kHz, 16 kHz, 32 kHz, 48 kHz, 96 kHz). Using a 48 MHz clock ensures that the I2S peripheral can generate accurate and stable audio clocks, which is critical for high-quality audio applications.

So the PLLI2S should be modified to return 48 MHz
```shell
&plli2s {
	mul-n = <192>;
	div-r = <4>;
	status = "okay";
};
```

fariouche, 2024-09-30:
Thanks, it's better indeed. Just for my understanding, why the previous settings haven't changed the output clock? (48MHz) I was expecting the plli2s to configure the clock correctly by using the dts settings but it seems to only use the dts to "read" the clock instead (ie, even if it's actually 48MHz, it will return 72MHz in my case)

marwaiehm-st, 2024-10-01:
The previous settings, do you mean this ?
```shell
&plli2s{
        #clock-cells = <0>;
	clocks = <&clk_hse>;
	div-m = <4>;
	mul-n = <72>;
	div-r = <2>;
	status = "okay";
};
``` 
The clock settings for peripherals like PLLI2S is primarily driven by DTS file. It is possible to configure the clock Kconfig or direct firmware configuration are less common and not recommended for standard peripheral clock settings.

fariouche, 2024-10-04:
Just to let you know that I sorted out my codec issue, and now I have sound from my stm32f412 discovery board.
It would help to have sane defaults in the dts I think.

Thanks for your help!

fariouche, 2025-08-24:
Hello,
I migrated to zephyr 4.2, and now the i2s3 is not working again. I'm getting and error from the dts parser.
I updated a bit the definition of the i2s3 node in my overlay like so:
```
soc {
        i2s3: i2s@40003c00 {
	        compatible = "st,stm32-i2s";
	        #address-cells = <1>;
	        #size-cells = <0>;
	        reg = <0x40003c00 0x400>;
	        clocks = <&rcc STM32_CLOCK(APB1, 15U)>,
	                 <&rcc STM32_SRC_PLLI2S_R I2S_SEL(0)>;
	        interrupts = <51 5>;
	        dmas = <&dma1 5 0 (STM32_DMA_PERIPH_TX | STM32_DMA_16BITS | STM32_DMA_PRIORITY_HIGH) STM32_DMA_FIFO_FULL
		            &dma1 0 0 (STM32_DMA_PERIPH_RX | STM32_DMA_16BITS | STM32_DMA_PRIORITY_HIGH) STM32_DMA_FIFO_FULL>;
	        dma-names = "tx", "rx";
	        status = "disabled";
        };
			
    };
```

But I still get an error I do not understand:
stm32.overlay:13 (column 73): parse error: expected number or parenthesized expression
CMake Error at /home/fariouche/.platformio/packages/framework-zephyr3/cmake/modules/dts.cmake:306 (execute_process):

Line 13 is `<&rcc STM32_SRC_PLLI2S_R I2S_SEL(0)>;` but it looks normal to me, and seeing other that defined the second clock, it is the same...