zephyrproject-rtos/zephyr #78830: sample: samples/net/sockets/echo_server, board: native_sim Build failure

Freey0, 2024-09-23:
**Describe the bug**

Follow the [Networking with native_sim board tutorial](https://docs.zephyrproject.org/latest/connectivity/networking/native_sim_setup.html#networking-with-native-sim), Failed to build echo_server on the native_sim board

**To Reproduce**

west build -p always -b native_sim zephyr/samples/net/sockets/echo_server

**Expected behavior**

build success
**Impact**

**Logs and console output**

-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/xxx/zephyr/samples/net/sockets/echo_server
-- CMake version: 3.29.6
-- Found Python3: /home/xxx/.venv/bin/python3 (found suitable version "3.8.10", minimum required is "3.8") found components: Interpreter
-- Cache files will be written to: /home/xxx/.cache/zephyr
-- Zephyr version: 3.7.99 (/home/xxx/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: native_sim, qualifiers: native
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.8 (/opt/zephyr-sdk-0.16.8)
-- Found toolchain: host (gcc/ld)
-- Found Dtc: /opt/zephyr-sdk-0.16.8/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/xxx/zephyr/boards/native/native_sim/native_sim.dts
-- Generated zephyr.dts: /home/xxx/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/xxx/build/zephyr/include/generated/zephyr/devicetree_generated.h
-- Including generated dts.cmake file: /home/xxx/build/zephyr/dts.cmake
Parsing /home/xxx/zephyr/samples/net/sockets/echo_server/Kconfig
Loaded configuration '/home/xxx/zephyr/boards/native/native_sim/native_sim_defconfig'
Merged configuration '/home/xxx/zephyr/samples/net/sockets/echo_server/prj.conf'
Configuration saved to '/home/xxx/build/zephyr/.config'
Kconfig header saved to '/home/xxx/build/zephyr/include/generated/zephyr/autoconf.h'
-- Found GnuLd: /usr/bin/ld.bfd (found version "2.34")
-- The C compiler identification is GNU 9.4.0
-- The CXX compiler identification is GNU 9.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/gcc
-- Setting build type to 'MinSizeRel' as none was specified.
-- Using ccache: /usr/bin/ccache
-- Looking for xtensa/config/core-isa.h
-- Looking for xtensa/config/core-isa.h - not found
-- Configuring done (25.2s)
-- Generating done (0.4s)
-- Build files have been written to: /home/xxx/build
-- west build: building application
[9/1202] Preparing syscall dependency handling

[125/1202] Generating include/generated/zephyr/version.h
-- Zephyr version: 3.7.99 (/home/xxx/zephyr), build: v3.7.0-3233-g4cc3134aa1c1
[595/1202] Building C object modules/picolibc/CMakeFiles/c.dir/newlib/libc/tinystdio/vfscanf.c.obj
FAILED: modules/picolibc/CMakeFiles/c.dir/newlib/libc/tinystdio/vfscanf.c.obj 
ccache /usr/bin/ccache /usr/bin/gcc  -I/home/xxx/modules/lib/picolibc/newlib/libm/common -isystem /home/xxx/build/modules/picolibc/picolibc/include -Os -DNDEBUG -nostdlib -D_LIBC -fno-strict-aliasing -Os -imacros /home/xxx/build/zephyr/include/generated/zephyr/autoconf.h -fno-common -g -gdwarf-4 -fdiagnostics-color=always -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wdouble-promotion -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -fno-asynchronous-unwind-tables -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/home/xxx/zephyr/samples/net/sockets/echo_server=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/xxx/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/xxx=WEST_TOPDIR -ffunction-sections -fdata-sections -m32 -msse2 -mfpmath=sse -fvisibility=hidden -nostdinc -isystem /usr/lib/gcc/x86_64-linux-gnu/9/include -include /home/xxx/zephyr/arch/posix/include/undef_system_defines.h -ffreestanding -fno-builtin -D_POSIX_THREADS -fno-stack-protector -Wextra -Werror=implicit-function-declaration -Werror=vla -Warray-bounds -Wold-style-definition -frounding-math -fsignaling-nans -MD -MT modules/picolibc/CMakeFiles/c.dir/newlib/libc/tinystdio/vfscanf.c.obj -MF modules/picolibc/CMakeFiles/c.dir/newlib/libc/tinystdio/vfscanf.c.obj.d -o modules/picolibc/CMakeFiles/c.dir/newlib/libc/tinystdio/vfscanf.c.obj -c /home/xxx/modules/lib/picolibc/newlib/libc/tinystdio/vfscanf.c
/home/xxx/modules/lib/picolibc/newlib/libc/tinystdio/vfscanf.c:578:17: error: expected ‘;’, ‘,’ or ‘)’ before ‘.’ token
  578 | #define ap my_ap.ap
      |                 ^
/home/xxx/modules/lib/picolibc/newlib/libc/tinystdio/vfscanf.c:860:58: note: in expansion of macro ‘ap’
  860 | int __d_vfscanf (FILE * stream, const char *fmt, va_list ap) { return vfscanf(stream, fmt, ap); }
      |                                                          ^~
[618/1202] Building C object modules/picolibc/CMakeFiles/c.dir/newlib/libm/common/s_infinity.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/xxx/build

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used
 
 Ubuntu20.04
 zephyr-sdk-0.16.8
 main

**Additional context**

pdgendt, 2024-09-23:
Builds on `main` (4cc3134aa1c), have you updated using `west update`?

Freey0, 2024-09-23:
> Builds on ([4cc3134](https://github.com/zephyrproject-rtos/zephyr/commit/4cc3134aa1c13caa2a065b9c3f6bfb255b45c711)), have you updated using ?`main``west update`

Build on -- Zephyr version: 3.7.99 (/home/xxx/zephyr), build: v3.7.0-3233-g4cc3134aa1c1, 
Yes, Has been updated using west update

keith-packard, 2024-09-23:
cmake is mis-detecting compiler features in this configuration, which is done by compiling some short code snippets.
That's leading it down a backwards-compatibility code path which the bug represented in the error message.

It looks like the issue boils down to not using `-fno-common` while compiling that particular test. I've fixed the test (so that you'll get aliases working) and also fixed the backwards compatibility code path in case someone builds with a compiler that doesn't have the alias attribute. 

See https://github.com/picolibc/picolibc/pull/841 -- once that is merged upstream, I'll work on getting it into the zephyr module.

closed 2024-10-01 by aescolar