zephyrproject-rtos/zephyr #78864: Issue with the zephyr application signing using secure boot V2 for ESP32C3 chip.

rana-roy, 2024-09-23:
**Describe the bug**
I am trying to build the Zephyr sample application, https://github.com/zephyrproject-rtos/zephyr/tree/v3.7.0/samples/drivers/adc/adc_dt, and sign the application image using espressif's secure boot V2. When I flash the code into the esp32c3_devkitm, the board resets continuously. I have built the sample code for "Hello world" from esp-idf along with the secure boot v2 enabled to flash the bootloader into the board. I flashed the ESP-IDF built hello world signed binaries into the board.
Now, I am trying to build the Zephyr app signed binary using both the ways:
- Using the "zephyr.bin" and then sign using: 
     espsecure.py sign_data -v 2 -k secure_boot_signing_key.pem -o ./build/signed_zephyr.bin ./build/zephyr.bin
- Using the "zephyr.elf" and then convert it into binary and then sign it using: 
     esptool.py --chip esp32c3 -p /dev/ttyACM0 elf2image --pad-to-size 4KB --secure-pad-v2 ./build/zephyr.elf 
     espsecure.py sign_data -v 2 -k secure_boot_signing_key.pem -o ./build/signed_zephyr.bin ./build/zephyr.bin

When I flash the binary into the app sector, the application is not running as expected.

- Target Platform: ESP32C3-DevkitM
- Target Board: esp32c3_devkitm

**To Reproduce**
Go to ESP-IDF directory, build the hello_world sample app along with secure boot v2, and flash the code.
Go to zephyr sample app: https://github.com/zephyrproject-rtos/zephyr/tree/v3.7.0/samples/drivers/adc/adc_dt
Build command: west build -b esp32c3_devkitm .
Sign the zephyr.bin using: espsecure.py sign_data -v 2 -k secure_boot_signing_key.pem -o ./build/signed_zephyr.bin ./build/zephyr.bin (Take the previously generated key from the esp-idf directory)
Flash the signed binary into the app sector.

**Expected behavior**
The signed application should start functioning as normal build.

**Impact**
Previously, I was trying to get some solution using the sysbuild. But struggling with signing the Zephyr application image for ESP32C3 chip since long. 

**Logs**
[image]

**Environment:**
- OS: Linux
- Toolchain: Zephyr SDK 0.16.8
- Zephyr version: 3.7.0 (LTS)

sylvioalves, 2024-09-23:
You would need to use MCUBoot to support SecureBoot. It seems you have been single default `SIMPLE_BOOT` bootloader.

rana-roy, 2024-09-23:
I tried building with sysbuild for the MCUboot. But could not able to get the proper guide to do it for espressif chip. I'll be very grateful if I can able to refer to any guide or document regarding signing up the app image instead of building the secure bootloader for ESP from ESP IDF separately.

rftafas, 2024-09-24:
> Hi @sylvioalves, Thank you for your quick response! I tried building with sysbuild for the MCUboot. But could not able to get the proper guide to do it for espressif chip. I'll be very grateful if I can able to refer to any guide or document regarding signing up the app image instead of building the secure bootloader for ESP from ESP IDF separately.

https://docs.mcuboot.com/readme-espressif.html

There are several ports of MCUboot (check the project's folder). Zephyr regularly uses Zephyr port. It do work for Espressif chips BUT... It is larger in memory usage (ok, it will be reclaimed later, still a problem for some) and lacks Secure boot and Flash Encryption.

The Sysbuild only builds Zephyr port. We are still working on a better integration of Espressif Port.

rftafas, 2024-09-24:
So far, I don't think there is a bug here. Either this should be closed or (better) moved to discussions.

rana-roy, 2024-09-24:
Thank you for the quick and guiding responses @sylvioalves and @rftafas!
It is quite clear now and I am trying to implement this. 
If I get any further queries, I will generate a ticket for discussion.

closed 2024-09-24 by rana-roy

zhang-wenchao, 2025-02-06:
Any update?  I'm in dire need of secure boot right now, and I'd appreciate any updates.

rftafas, 2025-02-06:
> Any update? I'm in dire need of secure boot right now, and I'd appreciate any updates.

What updates? You just need to follow the instructions on the link and use Espressif Port for this feature. There was no issue here.

zhang-wenchao, 2025-02-25:
@sylvioalves @rftafas 

Lack of an example or complete tutorial.

I think what you mean is that I can build using mcuboot espressif port separately. Then use zephyr sysbuild normally to generate zephyr.signed.bin. Write mcuboot to enable secure boot etc. first, then write zephyr signed app?

Is this the best way to enable secure boot in zephyr?

If so I could write a simple tutorial that would put an end to this current mess.

rftafas, 2025-02-25:
> Lack of an example or complete tutorial.

There is a tutorial, it is complete and it was provided by me on a link above. Please, follow it.
Repeating the link: https://docs.mcuboot.com/readme-espressif.html

> I think what you mean is that I can build using the MCUboot Espressif Port separately. Then use Zephyr sysbuild normally to generate zephyr.signed.bin. Write MCUboot to enable secure boot, etc. first, then write the Zephyr signed application?

I don't want to be rude, but you clearly didn't paid enough attention to the available documentation. The process is very well explained there, so please, go for it again. Follow it to the letter, it is common to jump a step and get bogus results. Also, make sure everything is properly installed.

The process to make it work, _including_ explanations on the security chain are there.

Of course, suggestions are welcome.

> Is this the best way to enable secure boot in Zephyr?

Espressif port currently is the only way. It is also explained on the link above. **_Zephyr port of MCUboot on Espressif chips DO NOT support any security features._**

### In case Someone needs a "overview" of the whole process - valid for 4.1 and older

1. Install Zephyr in home dir, follow getting started.
1. Build hello world. flash. (will be built without bootloader, a.k.a. Simpleboot). Hello World works.
1. add kconfig for mcuboot on proj.conf of hello world
1. do the Zephyr port pristine build with --sysbuild, flash, hello world works with Zephyr port of MCUboot.
1. download ESP-IDF with GIT clone home dir. Follow starting guide.
1. run hello world for ESP-IDF. flash. works.
1. clone MCUboot repo in home dir.
1. follow ALL instructions from https://docs.mcuboot.com/readme-espressif.html
1. flash MCUboot on the device using the ninja -C build/ flash.
1. go back to Zephyr folder.
1. do a pristine build of hello_world, WITHOUT --sysbuild while keeping the MCUboot kconfig (it is the only kconfig there).
1. run `west flash` and `west espressif monitor`

IMPORTANT:

- Please, open a new discussion or go to Discord for further questions, as this is a closed issue from OP, which seems to be ok with the answer / results.
- Once this all is a success, then you can start with Flash Encryption and Secure boot. READ THE DOCS, otherwise, you can brick your device.
- Espressif may have already have a better way to do it. For releases newer than 4.1, please, look after an updated information and HOW-TO for Zephyr.