zephyrproject-rtos/zephyr #78866: boards: nxp: frdm_mcxn947 flash driver build is failed.

butok, 2024-09-23:
**Describe the bug**

frdm_mcxn947 flash driver build is failed.

It 's stared failing since the following 9/20/2024 commit:
```
SHA-1: cbee39ef711e13c2acf1fae090e9c0c123986af2
* soc: nxp: Add MCXW71

Add MCXW71 SOC, which inherits some qualitiies
of kinetis heritage platforms.

```
**To Reproduce**
Steps to reproduce the behavior:
1. `west build -p always -b frdm_mcxn947/mcxn947/cpu0 tests/drivers/flash/common`
2. See error log:

```
Loading Zephyr default modules (Zephyr base).
-- Application: C:/_ddm/ZEPHYR/zephyr_latest/zephyr/tests/drivers/flash/common
-- CMake version: 3.28.1
-- Found Python3: C:/Python311/python.exe (found suitable version "3.11.4", minimum required is "3.8") found components: Interpreter
-- Cache files will be written to: C:/_ddm/ZEPHYR/zephyr_latest/zephyr/.cache
-- Zephyr version: 3.7.99 (C:/_ddm/ZEPHYR/zephyr_latest/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: frdm_mcxn947, qualifiers: mcxn947/cpu0
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.8 (C:/Users/nxa12205/zephyr-sdk-0.16.8)
-- Found toolchain: zephyr 0.16.8 (C:/Users/nxa12205/zephyr-sdk-0.16.8)
-- Found Dtc: C:/Users/nxa12205/.mcuxpressotools/dtc-msys2/tools/usr/bin/dtc.exe (found suitable version "1.6.1", minimum required is "1.4.6")
-- Found BOARD.dts: C:/_ddm/ZEPHYR/zephyr_latest/zephyr/boards/nxp/frdm_mcxn947/frdm_mcxn947_mcxn947_cpu0.dts
-- Generated zephyr.dts: C:/_ddm/ZEPHYR/zephyr_latest/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: C:/_ddm/ZEPHYR/zephyr_latest/zephyr/build/zephyr/include/generated/zephyr/devicetree_generated.h
-- Including generated dts.cmake file: C:/_ddm/ZEPHYR/zephyr_latest/zephyr/build/zephyr/dts.cmake
Parsing C:/_ddm/ZEPHYR/zephyr_latest/zephyr/Kconfig
Loaded configuration 'C:/_ddm/ZEPHYR/zephyr_latest/zephyr/boards/nxp/frdm_mcxn947/frdm_mcxn947_mcxn947_cpu0_defconfig'
Merged configuration 'C:/_ddm/ZEPHYR/zephyr_latest/zephyr/tests/drivers/flash/common/prj.conf'
Configuration saved to 'C:/_ddm/ZEPHYR/zephyr_latest/zephyr/build/zephyr/.config'
Kconfig header saved to 'C:/_ddm/ZEPHYR/zephyr_latest/zephyr/build/zephyr/include/generated/zephyr/autoconf.h'
-- Found GnuLd: c:/users/nxa12205/zephyr-sdk-0.16.8/arm-zephyr-eabi/arm-zephyr-eabi/bin/ld.bfd.exe (found version "2.38")
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/nxa12205/zephyr-sdk-0.16.8/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc.exe
Load components for MCXN947_cm33_core0:
driver_common component is included.
driver_reset component is included.
device_CMSIS component is included.
CMSIS_Include_core_cm component is included.
device_system component is included.
driver_lpflexcomm component is included.
driver_lpuart component is included.
driver_flexspi component is included.
driver_mcx_spc component is included.
driver_port component is included.
driver_cache_cache64 component is included.
driver_flashiap component is included.
-- Setting build type to 'MinSizeRel' as none was specified.
-- Using ccache: C:/Strawberry/c/bin/ccache.exe
-- Configuring done (31.9s)
-- Generating done (0.8s)
-- Build files have been written to: C:/_ddm/ZEPHYR/zephyr_latest/zephyr/build
-- west build: building application
[1/151] Generating include/generated/zephyr/version.h
-- Zephyr version: 3.7.99 (C:/_ddm/ZEPHYR/zephyr_latest/zephyr), build: v3.7.0-3184-gcbee39ef711e
[79/151] Building C object zephyr/drivers/flash/CMakeFiles/drivers__flash.dir/soc_flash_mcux.c.obj
FAILED: zephyr/drivers/flash/CMakeFiles/drivers__flash.dir/soc_flash_mcux.c.obj
ccache C:\Users\nxa12205\zephyr-sdk-0.16.8\arm-zephyr-eabi\bin\arm-zephyr-eabi-gcc.exe -DCPU_MCXN947VDF_cm33_core0 -DKERNEL -DK_HEAP_MEM_POOL_SIZE=0 -DLPFLEXCOMM_INIT_NOT_USED_IN_DRIVER=1 -DPICOLIBC_LONG_LONG_PRINTF_SCANF -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_START -D__ZEPHYR_SUPERVISOR__ -D__ZEPHYR__=1 -IC:/_ddm/ZEPHYR/zephyr_latest/zephyr/drivers/memc -IC:/_ddm/ZEPHYR/zephyr_latest/zephyr/build/zephyr/include/generated/zephyr -IC:/_ddm/ZEPHYR/zephyr_latest/zephyr/include -IC:/_ddm/ZEPHYR/zephyr_latest/zephyr/build/zephyr/include/generated -IC:/_ddm/ZEPHYR/zephyr_latest/zephyr/soc/nxp/mcx -IC:/_ddm/ZEPHYR/zephyr_latest/zephyr/soc/nxp/mcx/mcxn/. -IC:/_ddm/ZEPHYR/zephyr_latest/zephyr/subsys/testsuite/include -IC:/_ddm/ZEPHYR/zephyr_latest/zephyr/subsys/testsuite/coverage -IC:/_ddm/ZEPHYR/zephyr_latest/zephyr/subsys/testsuite/ztest/include -IC:/_ddm/ZEPHYR/zephyr_latest/modules/hal/cmsis/CMSIS/Core/Include -IC:/_ddm/ZEPHYR/zephyr_latest/zephyr/modules/cmsis/. -IC:/_ddm/ZEPHYR/zephyr_latest/modules/hal/nxp/mcux/mcux-sdk/devices/MCXN947 -IC:/_ddm/ZEPHYR/zephyr_latest/modules/hal/nxp/mcux/mcux-sdk/devices/MCXN947/drivers -IC:/_ddm/ZEPHYR/zephyr_latest/modules/hal/nxp/mcux/mcux-sdk/drivers/common -IC:/_ddm/ZEPHYR/zephyr_latest/modules/hal/nxp/mcux/mcux-sdk/drivers/lpflexcomm -IC:/_ddm/ZEPHYR/zephyr_latest/modules/hal/nxp/mcux/mcux-sdk/drivers/lpflexcomm/lpuart -IC:/_ddm/ZEPHYR/zephyr_latest/modules/hal/nxp/mcux/mcux-sdk/drivers/flexspi -IC:/_ddm/ZEPHYR/zephyr_latest/modules/hal/nxp/mcux/mcux-sdk/drivers/mcx_spc -IC:/_ddm/ZEPHYR/zephyr_latest/modules/hal/nxp/mcux/mcux-sdk/drivers/port -IC:/_ddm/ZEPHYR/zephyr_latest/modules/hal/nxp/mcux/mcux-sdk/drivers/cache/cache64 -IC:/_ddm/ZEPHYR/zephyr_latest/modules/hal/nxp/mcux/mcux-sdk/drivers/mcx_romapi -IC:/_ddm/ZEPHYR/zephyr_latest/zephyr/modules/hal_nxp/. -isystem C:/_ddm/ZEPHYR/zephyr_latest/zephyr/lib/libc/common/include -Os -DNDEBUG -Wshadow -fno-strict-aliasing -Os -imacros C:/_ddm/ZEPHYR/zephyr_latest/zephyr/build/zephyr/include/generated/zephyr/autoconf.h -fno-printf-return-value -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m33 -mthumb -mabi=aapcs -mfp16-format=ieee -mtp=soft --sysroot=C:/Users/nxa12205/zephyr-sdk-0.16.8/arm-zephyr-eabi/arm-zephyr-eabi -imacros C:/_ddm/ZEPHYR/zephyr_latest/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wdouble-promotion -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -fno-asynchronous-unwind-tables -ftls-model=local-exec -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=C:/_ddm/ZEPHYR/zephyr_latest/zephyr/tests/drivers/flash/common=CMAKE_SOURCE_DIR -fmacro-prefix-map=C:/_ddm/ZEPHYR/zephyr_latest/zephyr=ZEPHYR_BASE -fmacro-prefix-map=C:/_ddm/ZEPHYR/zephyr_latest=WEST_TOPDIR -ffunction-sections -fdata-sections -mcmse --specs=picolibc.specs -std=c99 -MD -MT zephyr/drivers/flash/CMakeFiles/drivers__flash.dir/soc_flash_mcux.c.obj -MF zephyr\drivers\flash\CMakeFiles\drivers__flash.dir\soc_flash_mcux.c.obj.d -o zephyr/drivers/flash/CMakeFiles/drivers__flash.dir/soc_flash_mcux.c.obj -c C:/_ddm/ZEPHYR/zephyr_latest/zephyr/drivers/flash/soc_flash_mcux.c
C:/_ddm/ZEPHYR/zephyr_latest/zephyr/drivers/flash/soc_flash_mcux.c:51:10: fatal error: fsl_flash.h: No such file or directory
   51 | #include "fsl_flash.h"
      |          ^~~~~~~~~~~~~
compilation terminated.
[92/151] Building C object zephyr/drivers/serial/CMakeFiles/drivers__serial.dir/uart_mcux_lpuart.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: 'C:\Users\nxa12205\.mcuxpressotools\cmake-3.28.1-windows-x86_64\bin\cmake.EXE' --build 'C:\_ddm\ZEPHYR\zephyr_latest\zephyr\build'
```

**Expected behavior**
No build errors

**Impact**

All flash dependent examples are failed for frdm_mcxn947 

**Environment (please complete the following information):**

 - OS: Windows
 - Toolchain Zephyr SDK
 - Commit SHA or Version used: cbee39ef711e13c2acf1fae090e9c0c123986af2

butok, 2024-09-23:
Briefly looked at the HAL, it seems that it can be caused by change in NXP HAL, by deleting zephyr_include_directories(..) in hal_nxp.cmake by

```
SHA-1: f4e26fad2cfd8b8e8988e835a28667573ed072cf

* mcux: MCXW71 plumbing

Add code to build MCXW71 in zephyr.

```

decsny, 2024-09-23:
if only we had CI

closed 2024-09-24 by aescolar