zephyrproject-rtos/zephyr #78906: mbedtls: redefinition of `DIV_ROUND_UP` with `CONFIG_POSIX_API=y`

JordanYates, 2024-09-24:
**Describe the bug**

Enabling the following configuration options with ECP curve448 results in build warnings:
 * `CONFIG_MBEDTLS_HAVE_TIME_DATE=y`
 * `CONFIG_POSIX_API=y`

This is particularly an issue with `WIFI_NM_WPA_SUPPLICANT`, which force enables the posix API:
https://github.com/zephyrproject-rtos/zephyr/blob/810f978bed4fbc35eec165eaaedd17745b1ec255/modules/hostap/Kconfig#L8-L12

**To Reproduce**

```
west build -p -b nrf52840dk/nrf52840 zephyr/tests/benchmarks/mbedtls/ -- -DCONFIG_MBEDTLS_HAVE_TIME_DATE=y -DCONFIG_POSIX_API=y
```

**Expected behavior**

No build warnings

**Logs and console output**

```
[3/277] Generating include/generated/zephyr/version.h
-- Zephyr version: 3.7.99 (/home/jordan/code/zephyr/zephyr), build: v3.7.0-3338-g810f978bed4f
[166/277] Building C object modules/mbedtls/CMakeFiles/mbedT.../code/zephyr/modules/crypto/mbedtls/library/ecp_curves.c.obj
/home/jordan/code/zephyr/modules/crypto/mbedtls/library/ecp_curves.c:5231: warning: "DIV_ROUND_UP" redefined
 5231 | #define DIV_ROUND_UP(X, Y) (((X) + (Y) -1) / (Y))
      |
In file included from /home/jordan/code/zephyr/zephyr/include/zephyr/sys/atomic.h:18,
                 from /home/jordan/code/zephyr/zephyr/include/zephyr/kernel_includes.h:25,
                 from /home/jordan/code/zephyr/zephyr/include/zephyr/kernel.h:17,
                 from /home/jordan/code/zephyr/zephyr/include/zephyr/posix/time.h:59,
                 from /home/jordan/code/zephyr/modules/crypto/mbedtls/include/mbedtls/platform_time.h:26,
                 from /home/jordan/code/zephyr/modules/crypto/mbedtls/include/mbedtls/platform_util.h:18,
                 from /home/jordan/code/zephyr/modules/crypto/mbedtls/include/mbedtls/ecp.h:27,
                 from /home/jordan/code/zephyr/modules/crypto/mbedtls/library/ecp_curves.c:14:
/home/jordan/code/zephyr/zephyr/include/zephyr/sys/util.h:359: note: this is the location of the previous definition
  359 | #define DIV_ROUND_UP(n, d) (((n) + (d) - 1) / (d))
      |
[277/277] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:      154076 B         1 MB     14.69%
             RAM:       92672 B       256 KB     35.35%
        IDT_LIST:          0 GB        32 KB      0.00%
Generating files from /home/jordan/code/zephyr/build/zephyr/zephyr.elf for board: nrf52840dk
```

ceolin, 2024-10-03:
Yep, that is a major / generic problem, without proper namespaces our macros can collide with other projects.
And it is hard to know the inclusion order to conditionally define, not to mention that these macros can have different behavior.

JordanYates, 2024-10-29:
> Yep, that is a major / generic problem, without proper namespaces our macros can collide with other projects. And it is hard to know the inclusion order to conditionally define, not to mention that these macros can have different behavior.

I think this is more an issue with the MbedTLS unconditionally defining a common macro name in the `.c` file. I can't see anything we can do in the Zephyr headers to prevent this apart from namespacing.

closed 2025-01-12 by github-actions[bot]