zephyrproject-rtos/zephyr #78934: FPGA ICE40: CDONE cannot be used with a PCAL6416

benediktibk, 2024-09-24:
**Describe the bug**
It is not possible to configure the CDONE GPIO of a ICE40 FPGA on the port expander PCAL6416. It fails during the call to gpio_pin_configure_dt of CDONE.

**To Reproduce**
Configure a ICE40 FPGA whose CDONE GPIO is on a PCAL6416.

**Expected behavior**
It should be possible to use the ICE40 with CDONE on a PCAL6416.

**Impact**
It is impossible to use the ICE40 with CDONE on a PCAL6416.

**Environment:**

Found in zephyr revision 45727b5fe141624a03866ff694c563830a26dfa4.

benediktibk, 2024-09-24:
Found a few more occurences matching `gpio_pin_configure_dt(.*, 0)`:

- boards/beagle/beagleconnect_freedom/board_antenna.c:65
- boards/beagle/beagleconnect_freedom/board_antenna.c:81
- boards/ti/common/board_antenna.c:66
- boards/ti/common/board_antenna.c:82

I am unsure if these should be fixed as well.

cfriedt, 2024-09-24:
> Found a few more occurences matching `gpio_pin_configure_dt(.*, 0)`:
>
> - boards/beagle/beagleconnect_freedom/board_antenna.c:65
> - boards/beagle/beagleconnect_freedom/board_antenna.c:81
> - [...]

My guess is that static flags in the gpio dt spec should be enough in those cases. I.e. the last parameter (`extra_flags`) is probably fine as zero. @vaishnavachath - please confirm.

benediktibk, 2024-09-24:
I assume at least these calls should be changed to `GPIO_DISCONNECTED`.

cfriedt, 2024-09-24:
I'm less inclined to believe that this is a bug in the fpga subsystem, since the driver works fine on other platforms. I think this bug is likely in the PCAL6416 driver.

benediktibk, 2024-09-24:
It just fails with the PCAL6416 driver because it it intentionally does not support GPIO_DISCONNECTED. Which is, according to @henrikbrixandersen, valid. Therefore the FPGA driver fails if it is used in combination with the PCAL6416 driver.

cfriedt, 2024-09-24:
@benediktibk - now I'm sort of thinking that the issue is in `gpio_pin_configure_dt()`.

Given that the parameter is called `extra_flags` it would imply that it has to be extra with respect to something else. Naturally that something else would be the flags that are part of the gpio dt spec.

The way that the argument is, would imply that 0 means no extra flags (semantically 'none'), just those that are in the gpio dt spec.

However, a value of 0 has a special meaning in the gpio API, and being disconnected is not the same as 'none', and in fact would conflict with the value in the DT spec (unless it was also exactly GPIO_DOSCONNECTED).

I wonder if there is a test for that narrow corner case.

That's why I was wondering if there is a convention of using a non-zero parameter in `extra_flags` elsewhere in Zephyr.

henrikbrixandersen, 2024-09-24:
If you don't specify a GPIO as being an input nor as being an output, it's disconnected.

What you want here is the GPIO to be configured as an input.

benediktibk, 2024-09-25:
I can remember my initial confusion with the GPIO flags when I was starting to use Zephyr. Why do I have to specify `ACTIVE_LOW/HIGH` in devicetree and `INPUT/OUTPUT` as extra flag in the call to configure the GPIO? This is definitely a questionable design choice, in my opinion.

However, for this issue I think we will have to accept the choice as it is and at least try to be consistent with the rest of the code base.

PR #78935 (2024-09-28): Fix gpio configuration of CDONE in FPGA ICE40
Configure the GPIO CDONE of the ICE40 as GPIO_INPUT.
Fixes #78934.
commit f54a97c1 (2024-09-28): drivers: fpga: configure CDONE for ICE40 as input

Configure the GPIO CDONE of the ICE40 as GPIO_INPUT.
Fixes #78934.
closed 2024-09-28 by mmahadevan108