zephyrproject-rtos/zephyr #78939: i.MX8MP-EVK SMP boot documentation not accurate

deribaucourt, 2024-09-24:
**Describe the bug**

The [documentation for imx8mp_evk](https://docs.zephyrproject.org/latest/boards/nxp/imx8mp_evk/doc/index.html) suggests to start zephyr on the A53 with SMP by running the u-boot command:

`cpu 2 release 0xc0000000`

This command is not available for this target. I also checked the u-boot-imx vendor forks, u-boot-imx & u-boot-fslc. This command requires an mp.c file to be defined. It is only available on [layerscape](https://github.com/Freescale/u-boot-fslc/blob/2024.07%2Bfslc/arch/arm/cpu/armv8/fsl-layerscape/mp.c) armv8 processors.

It looks like the [original author](https://github.com/zephyrproject-rtos/zephyr/pull/49651) @JiafeiPan did run Zephyr with SMP on the A53, but in all my attempts, Zephyr with SMP could not boot or produce any output. Maybe they could explain which u-boot version they were using at the time?

**To Reproduce**

**Expected behavior**

**Impact**

**Logs and console output**

**Environment (please complete the following information):**

 - u-boot-imx 2024.04+gde16f4f1722+p0 & 2020.04
 - Zephyr 3.4.0 (356c8cbe63ae01b3ab438382639d25bb418a0213) & Zephyr 3.7.0 (36940db938a8f4a1e919496793ed439850a221c2)

**Additional context**

mmahadevan108, 2024-09-24:
cc @JiafeiPan, can you take a look at the documentation for i.MX8MP-EVK

JiafeiPan, 2024-09-25:
Hi, @deribaucourt,  thanks for reporting this issue.
Acutrallly cpu command for mp.c supports in Real-time Edge uboot, you could use: https://github.com/nxp-real-time-edge-sw/real-time-edge-uboot/commits/uboot_v2023.04-2.9.0, branch: uboot_v2023.04-2.9.0.
And you could refer to Real-Time Edge Software at: https://www.nxp.com/rtedge, such as download pre-build images for v2.9 release on imx8mp: https://www.nxp.com/webapp/Download?colCode=REAL-TIME_EDGE_V2.9_IMX8MP-LPDDR4-EVK&appType=license, or download detailed document for how to run Zephyr and other multicore applications at user guide: https://www.nxp.com/docs/en/user-guide/REALTIMEEDGEUG.pdf

I will update Zephyr board document to provide more information about which u-boot could be used, any more questions feel free to let me know.

deribaucourt, 2024-09-25:
That's precisely what I needed to know.

Thanks a lot @JiafeiPan !

closed 2024-09-25 by deribaucourt