zephyrproject-rtos/zephyr #78943: Board nrf9131ek/nrf9131/ns fails to build minimal c library application

nordicjm, 2024-09-24:
**Describe the bug**
Board is failing to build sample as per log on https://github.com/zephyrproject-rtos/zephyr/actions/runs/10918300637/job/30303868974?pr=78621

**Expected behavior**
CI to not be broken

**Impact**
Showstopper, CI is broken on unrelated changes

@maxd-nordic please take a look

maxd-nordic, 2024-09-24:
TF-M wasn't a possibility in upstream-zephyr when the board was added AFAIK.
Maybe a TF-M expert can help here?

maxd-nordic, 2024-09-24:
Looks like ZRESTRICT is not defined.

dkalowsk, 2024-09-24:
is this a regression?  @d3zd3z this is the second TF-M issue we're seeing this morning.  Is it possible a TF-M update went wonky?

nordic-piks, 2024-09-26:
Triaged internally

tomi-font, 2024-09-26:
This is not a TF-M issue. Fixed by https://github.com/zephyrproject-rtos/zephyr/pull/78991.

closed 2024-09-26 by mmahadevan108