zephyrproject-rtos/zephyr #78950: Single slot Image compilation fails when trying to build for TF-M NS Application on MPS2 board

jayashree505, 2024-09-24:
**Describe the bug**

I am trying to  build a Non-Secure sample application for the mps2_an521 board. This build uses the TF-M BL2 port of MCUBoot. I want to enable the Single slot mechanism and therefore want TF-M and Zephyr to be merged as a single image to be loaded into a single MCUboot slot. When I enable the necessary macro on Zephyr west, the compilation fails. 

**To Reproduce**

```
west build -p -b mps2_an521 samples/hello_world/ -DCONFIG_MCUBOOT_BOOTLOADER_MODE_SINGLE_APP=y -DCONFIG_SINGLE_APPLICATION_SLOT=y
```
I get the undefined symbol error.
```
zephyr/build/zephyr/misc/generated/extra_kconfig_options.conf:3: warning: attempt to assign the value 'y' to the undefined symbol SINGLE_APPLICATION_SLOT
Parsing /home/js/workdisk/zephyr_pristine/zephyr/Kconfig
```
I have tried `-DCONFIG_MCUBOOT_SINGLE_APPLICATION_SLOT=y` and `-DSINGLE_APPLICATION_SLOT=y` macros. I am not sure if the macro I am using is incorrect or if the build fails becasue of a bug on the zephyr side. 

Thanks

dkalowsk, 2024-09-24:
@jayashree505 any chance you can bisect this?  It seems like a regression that likely got introduced recently.

jayashree505, 2024-09-24:
I tried the same command for other boards supporting Trustzone - lpcxpresso55s69_ns, stm32l562e_dk_ns, with `-DCONFIG_TFM_MCUBOOT_IMAGE_NUMBER=1` option included in build and I see the same behavior. 

The build fails with `Error: Image size (0x3fe1845) + trailer (0x1b0) exceeds requested size 0x3c000` Exceeded size error during the signing step. Are you asking if I can sign it manually? 

I am also confused about that. **In the Single image slot mechanism, should TF-M + Zephyr be merged and signed or should they be signed separately and merged?**

dkalowsk, 2024-09-24:
@d3zd3z FYI this is the other TF-M looking issue we saw today.  @ithinuel

closed 2024-12-16 by github-actions[bot]