zephyrproject-rtos/zephyr #78965: qemu_cortex_a9: Fails to run samples/hello_world on v3.7.0

mcscholtz, 2024-09-25:
**Describe the bug**
When trying to build and run `samples/hello_world` for the board `qemu_cortex_a9` I get the following error:
```
[0/1] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: cortex-a9
/bin/sh: line 1: QEMU-NOTFOUND: command not found
FAILED: zephyr/CMakeFiles/run_qemu /home/marius/projects/sample/zephyr/build/zephyr/CMakeFiles/run_qemu 
cd /home/marius/projects/sample/zephyr/build && QEMU-NOTFOUND -nographic -machine arm-generic-fdt-7series -dtb /home/marius/projects/sample/zephyr/boards/qemu/cortex_a9/fdt-zynq7000s.dtb -net none -pidfile qemu.pid -chardev stdio,id=con,mux=on -serial chardev:con -mon chardev=con,mode=readline -icount shift=3,align=off,sleep=on -rtc clock=vm -device loader,file=/home/marius/projects/sample/zephyr/build/zephyr/zephyr.elf,cpu-num=0
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/marius/projects/sample/zephyr/build --target run
```

I do not know if this is specific to my qemu installation or a wider issue, which is why I made a bug report with the patch instead of a PR

**To Reproduce**
```
cd zephyr
git checkout v3.7.0
west build -b qemu_cortex_a9 -p auto samples/hello_world
west build -t run
```

**Expected behavior**
It should run and print 'Hello World'

**Impact**
With the provided patch I can continue but I was surprised this error occured

**Logs and console output**
With the provided patch at the bottom I get the following output:
```
(.venv) ➜  zephyr git:(v3.7.0) ✗ west build -t run                                       
-- west build: running target run
[0/1] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: cortex-a9
qemu-system-arm: warning: nic cadence_gem.0 has no peer
qemu-system-arm: warning: nic cadence_gem.1 has no peer
*** Booting Zephyr OS build v3.7.0 ***
Hello World! qemu_cortex_a9/xc7z007s
```

**Environment (please complete the following information):**

I am using opensuse tumbleweed as the build environment.
Zephyr version checkout out at tag `v3.7.0`

**Additional context**
The following patch seems to fix the issue:
```
diff --git a/boards/qemu/cortex_a9/board.cmake b/boards/qemu/cortex_a9/board.cmake
index 45d2dcd7138..d3ccbe2e848 100644
--- a/boards/qemu/cortex_a9/board.cmake
+++ b/boards/qemu/cortex_a9/board.cmake
@@ -4,13 +4,13 @@
 #
 
 set(SUPPORTED_EMU_PLATFORMS qemu)
-set(QEMU_ARCH xilinx-aarch64)
+set(QEMU_ARCH arm)
 
 set(QEMU_CPU_TYPE_${ARCH} cortex-a9)
 
 set(QEMU_FLAGS_${ARCH}
   -nographic
-  -machine arm-generic-fdt-7series
+  -machine xilinx-zynq-a9
   -dtb ${CMAKE_CURRENT_LIST_DIR}/fdt-zynq7000s.dtb
   )
```

nordicjm, 2024-09-25:
This is not valid, the original board name is correct and has specific hardware configuration as listed on https://www.qemu.org/docs/master/system/arm/xlnx-zynq.html

ithinuel, 2024-09-25:
I don't see how changing the QEMU_ARCH affects the `QEMU-NOTFOUND`.
Either way, it seems like there's something off with your environment.

closed 2024-12-16 by github-actions[bot]