zephyrproject-rtos/zephyr #79027: NXP Boards that do not build hello_world

EmilioCBen, 2024-09-25:
**Describe the bug**
A small section of boards from boards/nxp do not seem to build hello_world.
This was tested on the latest commit of main1fe904539d32b5b7d18361d63acc7a47c9171400

frdm_mcxn947/mcxn947/cpu1
imx8mn_evk/mimx8mn6/m7
imx8mq_mimx8mq6/a53
imx8qm_mek/mimx8qm6/a72
imx8qm_mek/mimx8qm6/a53
imx8qm_mek/mimx8qm6/m4
imx8qxp_mek/mimx8qx6/a35
imx8qxp_mek/mimx8qx6/m4
imx8ulp_evk/mimx8ud7/f1_dsp
imx8ulp_evk/mimx8ud7/a35
mimxrt1050_evk/mimxrt1052/qspi
mimxrt1050_evk/mimxrt1052/hyperflash
mimxrt1060_evk/mimxrt1062/qspi
mimxrt1060_evk/mimxrt1062/hyperflash
mimxrt1170_evk/mimxrt1176/cm4/B
mimxrt1170_evk/mimxrt1176/cm7/B
s32z2xxdc2/s32z270/rtu0/B
s32z2xxdc2/s32z270/rtu0/D
s32z2xxdc2/s32z270/rtu1/B
s32z2xxdc2/s32z270/rtu1/D

**To Reproduce**
west build -p always -b <board_name> /path/to/samples/hello_world

**Expected behavior**
A successful build of hello world for <board_name>

**Impact**
Delay of CI confirmation and platform testing when PR for Issue [https://github.com/zephyrproject-rtos/zephyr/issues/78353] is posted.

**Logs and console output**
My Environment
--|
   |- zephyr/*
   | 
   |- builds/

Example output:  /> west build -p always -b imxulp_evk/mimx8ud7/a35 ./zephyr/samples/hello_world
-- west build: making build dir /home/emilio/zephyrproject/build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/emilio/zephyrproject/zephyr/samples/hello_world
-- CMake version: 3.29.2
-- Found Python3: /home/emilio/zephyrproject/.venv/bin/python3 (found suitable version "3.10.12", minimum required is "3.10") found components: Interpreter
-- Cache files will be written to: /home/emilio/.cache/zephyr
-- Zephyr version: 3.7.99 (/home/emilio/zephyrproject/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: imx8ulp_evk, qualifiers: mimx8ud7/a35
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.5 (/home/emilio/zephyr-sdk-0.16.5)
-- Found toolchain: zephyr 0.16.5 (/home/emilio/zephyr-sdk-0.16.5)
-- Found Dtc: /home/emilio/zephyr-sdk-0.16.5/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/emilio/zephyrproject/zephyr/boards/common/stub.dts
-- Generated zephyr.dts: /home/emilio/zephyrproject/build/zephyr/zephyr.dts
-- Generated pickled edt: /home/emilio/zephyrproject/build/zephyr/edt.pickle
-- Generated zephyr.dts: /home/emilio/zephyrproject/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/emilio/zephyrproject/build/zephyr/include/generated/zephyr/devicetree_generated.h
-- Including generated dts.cmake file: /home/emilio/zephyrproject/build/zephyr/dts.cmake
Parsing /home/emilio/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/emilio/zephyrproject/zephyr/samples/hello_world/prj.conf'
Configuration saved to '/home/emilio/zephyrproject/build/zephyr/.config'
Kconfig header saved to '/home/emilio/zephyrproject/build/zephyr/include/generated/zephyr/autoconf.h'
CMake Error at /home/emilio/zephyrproject/zephyr/cmake/modules/arch_v2.cmake:36 (message):
  ARCH not defined.  Check that BOARD=imx8ulp_evk, is selecting an
  appropriate SoC in Kconfig, SoC=, and that the SoC is selecting the correct
  architecture.
Call Stack (most recent call first):
  /home/emilio/zephyrproject/zephyr/cmake/modules/zephyr_default.cmake:133 (include)
  /home/emilio/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /home/emilio/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/bin/cmake -DWEST_PYTHON=/home/emilio/zephyrproject/.venv/bin/python3 -B/home/emilio/zephyrproject/build -GNinja -DBOARD=imx8ulp_evk/mimx8ud7/a35 -S/home/emilio/zephyrproject/zephyr/samples/hello_world

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA: 1fe904539d32b5b7d18361d63acc7a47c9171400

LaurentiuM1234, 2024-09-25:
Interesting.... `imx8`, `imx8x`, and `imx8ulp` should only support the ADSP-based builds, yet the soc.yml file contains other cpu clusters as well.....

dleach02, 2024-09-25:
@hakehuang

hakehuang, 2024-09-26:
@dleach02 , @EmilioCBen  those platform is not listed in the baords/nxp/<platyform>/board.yaml so they are not supposed to be supported, and I see #79028 is just remove those core, am I right?

EmilioCBen, 2024-09-26:
It seems there was some slight confusion, but confusion none the less.
It would seem that when you build for an invalid target of the boards I have listed,
these options do come up, so I only assumed they were suppose to build. I believe #79028
will take care of this.

hakehuang, 2024-09-26:
> It seems there was some slight confusion, but confusion none the less.
> It would seem that when you build for an invalid target of the boards I have listed,
> these options do come up, so I only assumed they were suppose to build. I believe #79028
> will take care of this. Under the same series of steps, the preview of how to generate a build
> with revision of a target say "s32z2xxdc2@D/s32z270/rtu0 is not a displayed example of a target.

this is also discussed in today's test working group meeting, and #77250 is to resolve this confusion.

JiafeiPan, 2024-09-27:
> this is also discussed in today's test working group meeting, and #77250 is to resolve this confusion.

So  #77250 is used to fix this issue, and no need for #79028, right? Per my understand, soc.yml lists all the hardware clusters but not clusters currently have been supported.

nashif, 2024-09-27:
#77250 will not solve the problem if I understand things correctly. This is an issue in hwmv2 in general, the build systems nad scripts will generate automatic board targets for clusters and consider them supported targets even if the board target was not implemented or configured. So removing the clusters is the proper solution IMO. FYI @tejlmand  @nordicjm

commit 8a1785b8 (2024-10-07): soc: nxp: soc.yml: remove unsupported cpu clusters for imx8 socs

Some imx8 socs have cpu cluster entries for cores that
are not currently supported. Remove them.

Fixes #79027.
closed 2024-10-07 by carlescufi