zephyrproject-rtos/zephyr #79038: STM32G473 Crash while starting ADC reads via DMA

PeeJay, 2024-09-26:
**Describe the bug**
Device crash while starting ADC reads via DMA.
It would seem that the DMA part of the ADC driver struct is never initialised, so dma_config calls an invalid pointer.
I also got a few warnings during compile, not sure if they are related.
```
-- west build: building application
[1/144] Generating include/generated/zephyr/version.h
-- Zephyr version: 3.7.99 (D:/cantest/zephyr), build: v3.7.0-3516-g57a8a7c3500a
[67/144] Building C object zephyr/drivers/adc/CMakeFiles/drivers__adc.dir/adc_stm32.c.obj
D:/cantest/zephyr/drivers/adc/adc_stm32.c:779:13: warning: 'dma_callback' defined but not used [-Wunused-function]
  779 | static void dma_callback(const struct device *dev, void *user_data,
      |             ^~~~~~~~~~~~
[86/144] Building C object zephyr/drivers/spi/CMakeFiles/drivers__spi.dir/spi_ll_stm32.c.obj
D:/cantest/zephyr/drivers/spi/spi_ll_stm32.c:126:13: warning: 'dma_callback' defined but not used [-Wunused-function]
  126 | static void dma_callback(const struct device *dma_dev, void *arg,
      |             ^~~~~~~~~~~~
[144/144] Linking CXX executable zephyr\zephyr.elf
```

**To Reproduce**
Run the ADC sequence sample app

**Logs and console output**
```
*** Booting Zephyr OS build v3.7.0-3516-g57a8a7c3500a ***
ADC sequence reading [0]:
[00:00:01.010,000] <err> os: ***** BUS FAULT *****
[00:00:01.015,000] <err> os:   Instruction bus error
[00:00:01.021,000] <err> os: r0/a1:  0x00000000  r1/a2:  0x00000000  r2/a3:  0x200002ec
[00:00:01.029,000] <err> os: r3/a4:  0x6ff7f6b5 r12/ip:  0x20000220 r14/lr:  0x08004b3b
[00:00:01.038,000] <err> os:  xpsr:  0x41000000
[00:00:01.043,000] <err> os: s[ 0]:  0x200035e4  s[ 1]:  0x00000000  s[ 2]:  0x0000000f  s[ 3]:  0x20003688
[00:00:01.054,000] <err> os: s[ 4]:  0x00000000  s[ 5]:  0xffffffff  s[ 6]:  0xffffffff  s[ 7]:  0x00000000
[00:00:01.064,000] <err> os: s[ 8]:  0x00000000  s[ 9]:  0xe000ed00  s[10]:  0x20000d68  s[11]:  0x00000000
[00:00:01.075,000] <err> os: s[12]:  0x0000000a  s[13]:  0x08009631  s[14]:  0x0800431c  s[15]:  0x61000000
[00:00:01.085,000] <err> os: fpscr:  0x00000000
[00:00:01.090,000] <err> os: Faulting instruction address (r15/pc): 0x70000000
[00:00:01.098,000] <err> os: >>> ZEPHYR FATAL ERROR 27: Unknown error on CPU 0
[00:00:01.106,000] <err> os: Current thread: 0x200007f0 (unknown)
[00:00:01.113,000] <err> os: Halting system
```
[image]

**Environment (please complete the following information):**
 - OS: Windows 10
 - Toolchain Zephyr SDK 0.16.5-1-win
 - v3.7.0-3516-g57a8a7c3500a
 - STM32G473

FRASTM, 2024-09-26:
hello, @PeeJay could you please share your DTS. The stm32G4 serie has a DMAMUX in front of the DMA. So configuration is done for DMAMUX instance.

PeeJay, 2024-09-26:
```
/dts-v1/;
#include <st/g4/stm32g473Xe.dtsi>
#include <st/g4/stm32g473v(b-c-e)tx-pinctrl.dtsi>
#include <zephyr/dt-bindings/input/input-event-codes.h>
#include <zephyr/dt-bindings/led/led.h>

/ {
// Analog IOs
&adc1 {
	#address-cells = <1>;
	#size-cells = <0>;

	pinctrl-0 = <&adc1_in15_pb0
				&adc1_in4_pa3
				&adc1_in3_pa2
				&adc1_in1_pa0
				&adc1_in9_pc3
				&adc1_in8_pc2
				&adc1_in7_pc1
				&adc1_in6_pc0
		    >;
	pinctrl-names = "default";
	st,adc-clock-source = <SYNC>;
	st,adc-prescaler = <4>;
	status = "okay";

	channel@f {
		reg = <15>;
		zephyr,gain = "ADC_GAIN_1";
		zephyr,reference = "ADC_REF_INTERNAL";
		zephyr,vref-mv = <2500>;
		zephyr,acquisition-time = <ADC_ACQ_TIME_DEFAULT>;
		zephyr,resolution = <12>;
		zephyr,oversampling = <32>;
	};
```
... etc ...

```

&dma1 {
	status = "okay";
};

&dmamux1 {
	status = "okay";
};

&clk_hsi {
	status = "okay";
};

&clk_hsi48 {
	status = "okay";
};

&pll {
	div-m = <4>;
	mul-n = <85>;
	div-p = <2>;
	div-q = <2>;
	div-r = <2>;
	clocks = <&clk_hsi>;
	status = "okay";
};

&rcc {
	clocks = <&pll>;
	clock-frequency = <DT_FREQ_M(170)>;
	ahb-prescaler = <1>;
	apb1-prescaler = <1>;
	apb2-prescaler = <1>;
};
```

FRASTM, 2024-09-26:
Don't you have a phandle with DMAMUX channel declaration for the ADC , like tests/drivers/adc/adc_api/boards/nucleo_g474re.overlay:
```
	dmas = <&dmamux1 0 5 (STM32_DMA_PERIPH_RX | STM32_DMA_MEM_16BITS |
		STM32_DMA_PERIPH_16BITS)>;
	dma-names = "dmamux";
```

PeeJay, 2024-09-26:
Ahh crud. Sorry! I looked everywhere to find documentation on how to setup the DMA correctly, but I never thought to look in there.
Just added it and now it's working. Thanks for the help!

closed 2024-09-26 by PeeJay