zephyrproject-rtos/zephyr #79057: Build parallelism not working with West and Cmake in Zephyr/nRF Connect SDK

aHaugl, 2024-09-26:
When trying to build with parallelism -o=-j1, -o=-j4, -o=-j10 or default (nothing specified) in nRF Connect SDK/Zephyr there are no changes to compilation time or cores used both with West build -o.. and when configuring and building projects with Cmake. 

All compilations result in ~identical use of available CPUs and identical time lapsed both when compiling projects in Windows10 and Ubuntu 22.04.

Is there anything under the hood that is overwritten or ignored when attempting to build with parallelism?

https://docs.nordicsemi.com/bundle/ncs-latest/page/zephyr/develop/west/build-flash-debug.html#build_parallelism

henrikbrixandersen, 2024-09-26:
Thank you for reporting this. However - unless you are able to reproduce this issue with upstream Zephyr `main` - please report issues with the nRF Connect SDK (NCS) on the [Nordic Semiconductor DevZone](https://devzone.nordicsemi.com/).

marc-hb, 2024-09-26:
I just retested this quickly and `west build --build-opt=-jN` works as intended in my workspace and configuration.

After you used `west` once, can you try to temporarily leave it aside and invoke `ninja` directly and see if that makes any difference? Like this:

```
ninja -C build/ clean
ninja -j1 -C build/ all
```

> ... unless you are able to reproduce this issue with upstream Zephyr main ...

Wherever any issue should be reported, it should always include the git SHA1s and any other, complete version information + step by step commands required for reproduction.

I'm afraid you didn't explain that either: is the problem that you can't make it faster, or can't make it less concurrent not to bury error messages?

> there are no changes to compilation time ...

Before testing, make also sure you (temporarily) uninstall `ccache`. It's so good and makes everything so much faster that it also makes it much harder to spot performance differences in this area.

> ... or cores used 

It's not that easy to spot how many cores are used because compiler processes are short-lived. On Linux, `top` is for instance not very useful, I prefer some carefully  timed `ps faux`.

aHaugl, 2024-09-27:
> Thank you for reporting this. However - unless you are able to reproduce this issue with upstream Zephyr `main` - please report issues with the nRF Connect SDK (NCS) on the [Nordic Semiconductor DevZone](https://devzone.nordicsemi.com/).

Hi Henrik.
I work for Nordic in Devzone and the issue originated in a Devzone case. The issue seem to be related to the build system in common for both Zephyr and NCS and were present in both SDKs.

> I just retested this quickly and `west build --build-opt=-jN` works as intended in my workspace and configuration.
That's interesting. I will try to reproduce the tests I set up with your steps to see if it still persists.

> It's not that easy to spot how many cores are used because compiler processes are short-lived. On Linux, `top` is for instance not very useful, I prefer some carefully timed `ps faux`.
I would imagine monitoring the cpu usage and overlaying the graphs combined with tracking compilation tiime should show a difference between -j1 and -j10, or am I mistaken in how small of an impakt parallelism has on the building procedure?

hellesvik-nordic, 2024-09-27:
I was the Ubuntu 22.04 test aHaugl reported.
So I downloaded Zephyr and ran the same tests for this. Indeed, with Zephyr I see -o=-j1 vs -o=-j12 behaving very differently. It is clear on `htop` that for 12 cores, all cores max out towards the end of the build, while they do indeed not with -j1.
For NCS, all cores max out for -j1 and for -j12, showing that this is indeed a NCS issue. We will import it internally and see if we can figure out what NCS does differently.

carlescufi, 2024-09-27:
Closing since this is NCS-specific

closed 2024-09-27 by carlescufi

marc-hb, 2024-09-27:
> Closing since this is NCS-specific

> ....

> see https://github.com/zephyrproject-rtos/zephyr/issues/79099

So it is not NCS-specific after all, correct? Cause --sysbuild is not NCS-specific.

hellesvik-nordic, 2024-09-30:
Correct. As you can see the new issue is also on Zephyr: https://github.com/zephyrproject-rtos/zephyr/issues/79099.
We thought it was NCS specific because NCS builds some projects with Sysbuild by default, and thus the tests were done a bit differently(with/out sysbuild) despite us thinking we built both the same way.