zephyrproject-rtos/zephyr #79059: BAP/USR/STR/BV-369-C test fails.

MicImbriani, 2024-09-26:
**Describe the bug**

The test BAP/USR/STR/BV-369-C fails. Looking at the traces, the ISO data being sent has payload 0, when it should instead be 40.

From my understanding, during the test procedure, PTS should send a MMI that would trigger hdl_wid_377. This would then in turn trigger the BTP_BAS_SEND callback in the tester app, which would fill up the ring buffer with the data. However, the MMI is not sent, and therefore the payload is 0. 
This lead to test failure.

Looking into the history of the test results reports:

- Test added: [week 33.7](https://lists.zephyrproject.org/g/Bluetooth-sig/message/1086?p=%2C%2C%2C20%2C0%2C0%2C0%3A%3ACreated%2C%2Cautopts+test+session%2C20%2C2%2C160%2C100846836)
- Test regressed: [week 39.6](https://lists.zephyrproject.org/g/Bluetooth-sig/message/1133?p=%2C%2C%2C20%2C0%2C0%2C0%3A%3ACreated%2C%2Cautopts+test+session%2C20%2C2%2C140%2C101672471)

The test has failed ever since. It was mistakenly marked to be related to the "[2nd CIS establishment issue](https://github.com/zephyrproject-rtos/zephyr/issues/60132)", but has now been corrected, as it is not affected by that issue.

**To Reproduce**

Run AutoPTS test

**Expected behavior**

IUT starts CIS stream, and test passes.

**Impact**

**Logs and console output**

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**

henrikbrixandersen, 2024-09-26:
Please use our bug template when reporting bugs. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md (e.g. how to reproduce, which subsystem this affects, ...).

MicImbriani, 2024-09-27:
> Please use our bug template when reporting bugs. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md (e.g. how to reproduce, which subsystem this affects, ...).

Hi Henrik. Apologies for omitting the information. I (mistakenly) assumed that this being an AutoPTS test, those information might be redundant. I updated the issue description now.

closed 2024-10-22 by aescolar