zephyrproject-rtos/zephyr #79099: Build options broken with sysbuild

carlescufi, 2024-09-27:
After investigating https://github.com/zephyrproject-rtos/zephyr/issues/79057, we have discovered that build options, both with `west` and `ninja`, are broken with sysbuild.

So, this works:
`west build -b nrf54h20dk/nrf54h20/cpuapp --build-opt=-j1 samples/hello_world` (built by a single core)
but this is broken:
`west build -b nrf54h20dk/nrf54h20/cpuapp --build-opt=-j1 --sysbuild samples/hello_world` (ignores the`-j1`, uses all cores)

The only way to limit the cores with sysbuild is:
`CMAKE_BUILD_PARALLEL_LEVEL=1 west build -b nrf54h20dk/nrf54h20/cpuapp --sysbuild samples/hello_world`

henrikbrixandersen, 2024-10-01:
Please use our bug template when reporting bugs. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

marc-hb, 2024-10-01:
> --build-opt=-j1 (ignores the-j1, uses all cores)

What happens when you invoke `ninja -C build/ -j1` directly?

carlescufi, 2024-10-08:
> west build -b nrf54h20dk/nrf54h20/cpuapp --build-opt=-j1 --sysbuild samples/hello_world

Still uses all the cores.

carlescufi, 2024-10-08:
> Please use our bug template when reporting bugs. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

Fixed, sorry!

tejlmand, 2025-03-05:
The limitation is with CMake and Ninja integration on external projects (Looking into how we can improve / work around this).

But CMake and Makefile integration works, so for example:
```
west build -b nrf54h20dk/nrf54h20/cpuapp --build-opt=-j1 samples/hello_world -- -G"Unix Makefiles"
```
only uses a single core.

or similar
```
west build -b nrf54h20dk/nrf54h20/cpuapp --build-opt="VERBOSE=1" samples/hello_world -- -G"Unix Makefiles"
```
will produce verbose output for all images.
It's possible to switch default generator for `west build` to Makefiles, and thereby remove the need for typing `-G...`.

As side-note, then this issue is not limited to `sysbuild` but is also present for other usage of external projects in Zephyr, such as TF-M, however the nature of sysbuild makes the issue more apparent, so keeping sysbuild as the main reference for the issue.

PR #86707 (2025-03-10): scripts: support number of jobs and verbosity in west build.
Fixes: #79099

Calling `west build --build-opt="-v" --build-opt="-j=<n>"` passes the build options to `cmake --build ... -- <build-opt>` which again is passed to the native build tool, such as ninja or make.

However, when ExternalProjects are used in CMake, such as in TF-M or sysbuild builds then those extra build options are only passed to the first image build and not those build as external projects.

CMake supports environment variables for those flags, so translate verbosity and number of jobs to those environment variables and [... cut]
closed 2025-03-10 by kartben