zephyrproject-rtos/zephyr #79102: Add twister support for Cadence toolchain on Xtensa

lyakh, 2024-09-27:
**All twister tests fail when using the Cadence toolchain on the qemu_xtensa platform**
SOF is built for Xtensa using the Cadence toolchain, for consistency it should also be possible to use it to run twister tests, including on the qemu_xtensa platform. Currently this doesn't work. A result from a local run:
```
ZEPHYR_TOOLCHAIN_VARIANT=xt-clang TOOLCHAIN_VER=RI-2022.10-linux XTENSA_SYSTEM=$HOME/xtensa/XtDevTools/install/builds/$TOOLCHAIN_VER/ace10_LX7HiFi4_2022_10/config XTENSA_TOOLCHAIN_PATH=$HOME/xtensa/XtDevTools/install/tools ../zephyr/scripts/twister -p qemu_xtensa -vv --force-toolchain
...
INFO    - 0 of 2863 test configurations passed (0.00%), 0 failed, 409 errored, 2454 skipped with 0 warnings in 3879.03 seconds
INFO    - In total 4262 test cases were executed, 12191 skipped on 1 out of total 2 platforms (50.00%)
INFO    - 0 test configurations executed on platforms, 409 test configurations were only built.
```

**How to solve**
The first error to solve should be this build failure:
```
$HOME/dev/zephyr/arch/xtensa/core/window_vectors.S: Assembler messages:
$HOME/dev/zephyr/arch/xtensa/core/window_vectors.S:105: Error: invalid register 'MISC0' for 'rsr' instruction
```
This should be relatively easy. But then I suspect we will hit more problems, e.g. FLIX instruction sequences, generated by the toolchain, which don't seem to be currently supported by QEMU. Adding FLIX support to QEMU might turn out a difficult time-consuming task, although they're already supported by GNU binutils.

**Alternative solutions**
As a temporary work-around a `-mno-generate-flix` flag can be added, but this might have a performance impact.

nashif, 2024-09-27:
why is this a twister problem, first, you should be able to build something directly with cmake/west, if that works, then twister will work as well.
Are you saying you are able to build with cmake/west directly, but twister still fails?

lyakh, 2024-09-27:
> why is this a twister problem, first, you should be able to build something directly with cmake/west, if that works, then twister will work as well. Are you saying you are able to build with cmake/west directly, but twister still fails?

@nashif we're consistently using that toolchain to build SOF+Zephyr firmware, so, yes, it's working for Intel ADSP targets. The build problem is probably related to the Xtensa SoC version, used for the qemu_xtensa platform. There is already code around those register definitions that are failing, I think they just aren't catching this case. But after a brief look at the code I didn't have a fast solution to it, might be a one-liner though. As for FLIX - that might be QEMU specific, so, it wouldn't show up when building.

nashif, 2024-09-27:
> @nashif we're consistently using that toolchain to build SOF+Zephyr firmware, so, yes, it's working for Intel ADSP targets.

right, I know this, but qemu_xtensa is not currently not supported with cadence toolchain, this is not related to twister, first we need to add this support, i.e. being able to build qemu_xtensa with xcc or xt-clang, then twister will just work.

dcpleung, 2024-09-27:
Errr... this is going to explode in board variants, as you will need a special board variant under `qemu_xtensa` for every SoC you want to support building through `xcc` and `xt-clang`.

I have to ask why. Cadence provides their own simulator (ISS) where you can run ELF images, and that comes with the toolchain installation.

dcpleung, 2024-09-27:
Not to mention that QEMU does not have the hardware support for the SoC. For example, the secondary core boot workflow is not going to work on QEMU. How do you plan to solve that? If this is going to be build only, then what's the point of building for QEMU...? Building for QEMU is that we can do testing.

jcmvbkbc, 2024-09-27:
> FLIX instruction sequences, generated by the toolchain, which don't seem to be currently supported by QEMU.
> Adding FLIX support to QEMU might turn out a difficult time-consuming task, although they're already supported by GNU binutils.

QEMU does support FLIX bundles for all instructions that it recognizes. If you encounter any issues with that please report them.