zephyrproject-rtos/zephyr #79108: Bluetooth: Audio: Change dependency of BT_BAP_SCAN_DELEGATOR to GATT_DYNAMIC_DB from 'select' to 'depends on'

babrsn, 2024-09-27:
**Is your enhancement proposal related to a problem? Please describe.**

Currently, ´BT_BAP_SCAN_DELEGATOR´ uses ´select´ to manage its dependency on ´GATT_DYNAMIC_DB´. A better approach is to change this from ´select´ to ´depends on´, which ensures that dependencies are updated in a single location and prevents the generation of invalid configurations. However, this change introduces a dependency loop that must be investigated and resolved.

Thalley, 2024-09-27:
@babrsn Please follow the templates when creating issues:

```
**Is your enhancement proposal related to a problem? Please describe.**
<!--
A clear and concise description of what the problem is.
-->

**Describe the solution you'd like**
<!--
A clear and concise description of what you want to happen.
-->

**Describe alternatives you've considered**
<!--
A clear and concise description of any alternative solutions or features you've considered.
-->

**Additional context**
<!--
Add any other context or graphics (drag-and-drop an image) about the feature request here.
-->
```

babrsn, 2024-09-30:
> @babrsn Please follow the templates when creating issues:
> 
> ```
> **Is your enhancement proposal related to a problem? Please describe.**
> 
> 
> **Describe the solution you'd like**
> 
> 
> **Describe alternatives you've considered**
> 
> 
> **Additional context**
> 
> ```

The issue description has been updated to follow the template.

babrsn, 2024-10-03:
@Thalley please assign it to @babrsn

PR #79360 (2024-10-23): Bluetooth: Audio: Update BT_BAP_SCAN_DELEGATOR dependency to GATT_DYN…
…AMIC_DB

Changed dependency of the BT_BAP_SCAN_DELEGATOR to GATT_DYNAMIC_DB from 'select' to 'depends on' and solved all loop dependencies caused by this change.

Fixes #79108
commit 7f820d59 (2024-10-23): Bluetooth: Audio: Update SCAN_DELEGATOR dependency to GATT_DYNAMIC_DB

Changed dependency of the BT_BAP_SCAN_DELEGATOR to GATT_DYNAMIC_DB
from 'select' to 'depends on' and solved all loop dependencies
caused by this change.

Fixes #79108
closed 2024-10-23 by carlescufi