zephyrproject-rtos/zephyr #79126: input: stm32 touch sensing controller peripheral implementation

arifbalik, 2024-09-27:
## Introduction

### Problem description

Some STM32 models have [TSC peripheral](http://www.st.com/resource/en/product_training/stm32l4_peripheral_touchsense.pdf) for capacitive touch sensing applications, this really accelerates application development and Zephyr has no support for it yet.

### Proposed change

I want to implement this hardware in input subsystem.

## Detailed RFC

There is no LL file for this peripheral and it is a rather simple peripheral to work with. Most of the devices has almost identical registers and configuration. So my plan is to create a unified `st,stm32-tsc.yaml` binding to define and control this peripheral.

I will be pushing the group capture values (readings from the sensing channels) to input subsystem so anyone can pick it up after it is enabled. 

There is also [STM32 Tocuh Library](https://www.st.com/en/embedded-software/32l1-touch-lib.html) but I am not planning to use it in this implementation as it is pretty high level and wouldn't pass zephyr coding guidelines.

### Proposed change (Detailed)

I already sketched out an initial binding file you can see below;

My configuration pretty much includes all available settings this peripheral can have;

-  Charge transfer pulse high/low duration
- Spread spectrum (enable/disable deviation and prescaler)
- Pulse generator prescaler
- Mac count feature (count value and interrupt settings)
- Default io modes
- Sync mode (enable and polarity)

These settings belong to the parent node which applies all groups. Then each sensing channel is a child node and has the following properties;

- Schmitt trigger mode
- Analog mode
- Channel type (sampling cap or electrode)

```yaml
# dts/bindings/input/st,stm32-tsc.yaml

description: STM32 Tocuh Sensing Controller (TSC) driver

compatible: "st,stm32-tsc"

include: [base.yaml, pinctrl-device.yaml]

properties:
  resets:
    type: phandle-array
    description: Reset controller

  cpth:
    type: int
    default: 1
    description: |
      Duration of the high state of the 
      charge transfer pulse (charge of Cx).
    enum: [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16]

  cptl:
    type: int
    default: 1
    description: |
      Duration of the low state of the 
      charge transfer pulse (transfer of charge from CX to CS).
    enum: [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16]

  ssd:
    type: int
    default: 0
    description: Spread spectrum deviation ([6:0] bits)

  spread-spectrum:
    type: boolean
    description: Spread spectrum enable

  spread-spectrum-prescaler:
    type: boolean
    description: Spread spectrum prescaler (fhclk/2 if enabled)

  pgpsc:
    type: int
    default: 1
    description: Pulse generator prescaler
    enum: [1, 2, 4, 8, 16, 32, 64, 128]

  mcv:
    type: int
    description: Max count value
    enum: [255, 511, 1023, 2047, 4095, 8191, 16383]

  iodef:
    type: string
    default: "push-pull-low"
    description: |
      I/O Default mode. Configuration of TSC IO when no acquisition is ongoing.
    enum: ["push-pull-low", "input-floating"]

  syncpol:
    type: string
    default: "falling-edge"
    description: Synchronization pin polarity.
    enum: ["falling-edge", "rising-edge"]

  synced-acquisition:
    type: boolean
    description: Synchronized acquisition enable (acquisition starts when START bit and signal on sync pin)

child-binding:
  description: STM32 TSC group configuration
  schmitt-trigger-hysteresis:
    type: boolean
    description: Schmitt trigger hysteresis enable

  analog-switch:
    type: boolean
    description: Analog switch enable

  channel-type:
    type: string
    default: "normal"
    description: Channel type
    enum: ["sampling", "normal"]
```

Here is an exmaple node for STM32U0x series;

```devicetree
tsc : tsc@40024000 {
            compatible = "st,stm32-tsc";
            reg = <0x40024000 0x400>;
            clocks = <&rcc STM32_CLOCK_BUS_AHB1 0x01000000>;
            resets = <&rctl STM32_RESET(AHB1, 24U)>;
            interrupts = <21 0>;
            status = "disabled";
        };

&tsc {
  status = "okay";
  cpth = <1>;
  cptl = <1>;
  ssd = <0>;
  spread-spectrum;
  spread-spectrum-prescaler;
  pgpsc = <1>;
  mcv = <255>;
  iodef = "push-pull-low";
  synced-acquisition;
  syncpol = "falling-edge";

  channel@0 {
    channel-type = "sampling";
  };

  channel@1 {
    channel-type = "normal";
    analog-switch;
  };

}
```

### Concerns and Unresolved Questions

- I'm not sure if the child node should represent a group or an individual channel. Maybe people from ST can help with that.
- All devices who has TSC only has 1 of this peripheral, I'm not sure whether i should implement is as a single instance per device or not.

erwango, 2024-09-30:
> I'm not sure if the child node should represent a group or an individual channel. Maybe people from ST can help with that.

I don't know well this IP, can it be done the same way than ADC ?

> All devices with TSC only has 1 of this peripheral, I'm not sure whether i should implement is as a single instance per device or not.

Please implement the driver as if multi-instance was possible. It's always possible to see new SoC with multiple IPs in future and it's always a pain to convert drivers to multi-instance.

arifbalik, 2024-10-18:
@erwango Sorry for late reply, i've been busy for the last week or two. I managed to piece together a working demo for you today. Though this is still work in progress.

Please check out my commit; https://github.com/arifbalik/zephyr/commit/693e7f8536de536a1732677b023be80249ba0fc6

I've tested this on a STM32U083C-DK. I only have this devkit for testing the TSC, but from the reference manuals they seem practically the same across models.

For the child binding I went in a direction like you suggested. Changed the binding file a little bit, and implemented the driver for multi instance TSC peripherals.

Now for the main `stm32u0.dtsi` i defnied the basic tsc peripheral which is common to all u0 series.

https://github.com/arifbalik/zephyr/blob/693e7f8536de536a1732677b023be80249ba0fc6/dts/arm/st/u0/stm32u0.dtsi#L390-L398

then in `stm32u083c_dk.dts` I enabled the peripheral;

https://github.com/arifbalik/zephyr/blob/693e7f8536de536a1732677b023be80249ba0fc6/boards/st/stm32u083c_dk/stm32u083c_dk.dts#L118-L154

Notice how i also defined `pinctrl` defines here, because i couldn't get hal_stm32 repo to generate TSC peripheral pin configs, but i think the issue is from the glob pattern you use to fetch the pins from the database, i will look into that later.

I wasn't sure where to put the driver, at first i put it in `input` but that didnt sound right so i put it in `drivers/misc`

I have also written a sample application to test it;

https://github.com/arifbalik/zephyr/blob/693e7f8536de536a1732677b023be80249ba0fc6/samples/drivers/misc/stm32_tsc/src/main.c

I also found a bug while trying to enable the TSC clock (AHB). I only saw this for u0 series;

https://github.com/zephyrproject-rtos/zephyr/blob/main/include/zephyr/dt-bindings/clock/stm32u0_clock.h#L12-L18

If you notice `STM32_PERIPH_BUS_MIN` is not set to the minimum offset number. I think this happened because of the line number of these defines. So whenever I try to enable something in AHB i got an error from the `clock_control_on` function until i changed the `STM32_PERIPH_BUS_MIN` to `STM32_CLOCK_BUS_AHB1`. I'll create a patch for this soon.

Please let me know what you think about the implementation so far :)

arifbalik, 2024-10-26:
@erwango opened a PR for the clock issue here

- #80471

PR #81187 (2025-03-17): drivers: input: added stm32 tsc peripheral support
This is an implementation for STM32 TSC (touch sensing controller) driver.  The details are discussed in 

- #79126

I've also added tests and a sample application for this and put it inside the same branch but this makes the PR rather large so I'm nor sure if I should split it in separate PRs, to make everything easier, I've separated them into different commits.

Closes #79126 

Note: I've written this driver for my blog post. Maybe it can help the review process;

https://arifbalik.github.io/self-monorepo/zephyr-device-driver/
closed 2025-03-17 by kartben