zephyrproject-rtos/zephyr #79137: NRF: Modifications on Hostap & Wifi causing a CI build failure

marwaiehm-st, 2024-09-27:
**Describe the bug**
There are some modifications on the module Hostap and the driver wifi  causing the CI build failure of this [PR](https://github.com/zephyrproject-rtos/zephyr/pull/79076) 
```c
/__w/zephyr/zephyr/modules/hostap/src/supp_api.c:355:9: error: duplicate case value
  355 |         case BAND_2_4_GHZ:
      |         ^~~~
/__w/zephyr/zephyr/modules/hostap/src/supp_api.c:351:9: note: previously used here
  351 |         case WPA_KEY_MGMT_IEEE8021X:
      |         ^~~~
/__w/zephyr/zephyr/modules/hostap/src/supp_api.c:352:9: error: case label value exceeds maximum value for type [-Werror=switch-outside-range]
  352 |         case WPA_KEY_MGMT_IEEE8021X_SUITE_B:
      |         ^~~~
/__w/zephyr/zephyr/modules/hostap/src/supp_api.c:353:9: error: case label value exceeds maximum value for type [-Werror=switch-outside-range]
  353 |         case WPA_KEY_MGMT_IEEE8021X_SUITE_B_192:
      |         ^~~~
cc1: all warnings being treated as errors
ninja: build stopped: subcommand failed.
``` 
CI compilation fails on these tests:
1. samples/net/wifi/sample.net.wifi.nrf7002ek on nrf5340dk/nrf5340/cpuapp
2. samples/net/wifi/sample.net.wifi.nrf70dk on nrf7002dk/nrf5340/cpuapp/nrf7001
3. samples/net/wifi/sample.net.wifi.nrf70dk on nrf7002dk/nrf5340/cpuapp
4. samples/net/wifi/sample.net.wifi.nrf7002ek on nucleo_h723zg
5. samples/net/wifi/sample.net.wifi.nrf7002eb on thingy53/nrf5340/cpuapp
**Note: All the tests have the same error logs**

**Logs and console output**
[image]
I sahred a file contains all the logs.
[CI_build_failure.log](https://github.com/user-attachments/files/17170884/CI_build_failure.log)

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used : https://github.com/zephyrproject-rtos/zephyr/commit/caf86128b753c826591c2e2ec1df70a0f368fb12

jukkar, 2024-09-28:
Already fixed by #79129

closed 2024-09-28 by jukkar

marwaiehm-st, 2024-09-28:
> Already fixed by #79129

Thank you