zephyrproject-rtos/zephyr #79147: mcuboot/mcumgr: board not upgradable after "Not enough free space to run swap upgrade" with slot1 big image

fabiobaltieri, 2024-09-28:
**Describe the bug**
Hi, this is the slot1 version of https://github.com/zephyrproject-rtos/zephyr/issues/58103, turns out was reported in one of the comments there (https://github.com/zephyrproject-rtos/zephyr/issues/58103#issuecomment-1664161553) and that's still a problem: during the initial programming of a device using mcuboot an image is programmed using most of the space in slot1, the device is no longer upgradable as it fails the swap size check regardless of the size of the slot2 image.

Reproduced this on an nrf52dk, just flashed a stock mcuboot and then an inflated version of samples/subsys/mgmt/mcumgr/smp_svr, just added this to the sample

```diff

--- a/samples/subsys/mgmt/mcumgr/smp_svr/src/main.c
+++ b/samples/subsys/mgmt/mcumgr/smp_svr/src/main.c
@@ -50,11 +50,16 @@ static struct fs_mount_t littlefs_mnt = {
 };
 #endif
 
+static const uint32_t x[1024*9] = {0xaa};
+
 int main(void)
 {
        int rc = STATS_INIT_AND_REG(smp_svr_stats, STATS_SIZE_32,
                                    "smp_svr_stats");
 
+       for (int i = 0; i < 128; i++)
+               printk("%02x\n", x[i]);
+
```

which results in

```
Memory region         Used Size  Region Size  %age Used
           FLASH:      219792 B       220 KB     97.56%
             RAM:       55268 B        64 KB     84.33%
        IDT_LIST:          0 GB        32 KB      0.00%
```

then built and flashed (`west build -p -b nrf52dk/nrf52832 samples/subsys/mgmt/mcumgr/smp_svr -DOVERLAY_CONFIG="overlay-bt.conf" -DCONFIG_MCUBOOT_SIGNATURE_KEY_FILE='"/home/fabiobaltieri/zephyrproject/bootloader/mcuboot/root-rsa-2048.pem"'`), then loaded a smaller imaged and flagged for test, reset and

```
I: Starting swap using move algorithm.
W: Not enough free space to run swap upgrade
W: required 225280 bytes but only 221184 are available
I: Bootloader chainload address offset: 0xc000
I: Jumping to the first image slot
````

no luck, device stuck with that firmware.

Also still, the new image is now stuck forever in slot2

```
~/zephyrproject/zephyr$ sudo /home/fabiobaltieri/go/bin/mcumgr --conntype ble --connstring peer_name='Zephyr' image erase
Error: 6
~/zephyrproject/zephyr$ sudo /home/fabiobaltieri/go/bin/mcumgr --conntype ble --connstring peer_name='Zephyr' image list
Images:
 image=0 slot=0
    version: 0.0.0
    bootable: true
    flags: active confirmed
    hash: da52ae009483d49fd4714b1d82640186a0ae1d375a4c11461e1155375a8a51e5
 image=0 slot=1
    version: 0.0.0
    bootable: true
    flags: pending
    hash: a4c910dc3a2ee2ee72aad29dce306e419c008368954fe40c1fc0935ec352b8a9
Split status: N/A (0)
```

Discussion points:
- can the build be more defensive and flag these situations when building for mcuboot?
- alternatively can mcuboot flag these in a warning at start? then one at least has a chance to catch it during development

**To Reproduce**
See above.

**Expected behavior**
Non upgradable images should fail the build.

**Impact**
It's bad, potential device lost in the field if this is not caught.

**Logs and console output**
See above.

**Environment (please complete the following information):**
Linux, Zephyr SDK 0.16.8, d11997dbcf2

fabiobaltieri, 2024-09-28:
cc @nordicjm @de-nordic, for the mcumgr part, I think the image locked up falls there (though I can probably look into that myself)

nordicjm, 2024-09-30:
This is a complete non-issue if you build with sysbuild, if you build without it then you have no way of knowing

fabiobaltieri, 2024-09-30:
> This is a complete non-issue if you build with sysbuild, if you build without it then you have no way of knowing

Is the image validate when using sysbuild?

nordicjm, 2024-09-30:
> > This is a complete non-issue if you build with sysbuild, if you build without it then you have no way of knowing
> 
> Is the image validate when using sysbuild?

Not sure of the question there. By using sysbuild, it takes into consideration the overhead of what MCUboot needs, so if you build this application for nrf52dk without sysbuild, you will see:
```
Memory region         Used Size  Region Size  %age Used
           FLASH:       34768 B       220 KB     15.43%
             RAM:       12784 B        64 KB     19.51%
        IDT_LIST:          0 GB        32 KB      0.00%
```
If you build it with sysbuild, you will see:
```
Memory region         Used Size  Region Size  %age Used
           FLASH:       34768 B       212 KB     16.02%
             RAM:       12784 B        64 KB     19.51%
        IDT_LIST:          0 GB        32 KB      0.00%
```

Notice how 220KiB has became 212KiB? That's because the overheads have been taken into consideration, you can't do that without sysbuild because it doesn't know the overheads, though you can manually adjust `CONFIG_ROM_END_OFFSET` to what you expect the overhead to be and it will do the same thing, so if you set it to `0x2000` without sysbuild for example then build, you will get the same:
```
Memory region         Used Size  Region Size  %age Used
           FLASH:       34768 B       212 KB     16.02%
             RAM:       12784 B        64 KB     19.51%
        IDT_LIST:          0 GB        32 KB      0.00%
```

fabiobaltieri, 2024-09-30:
Hey I see, `CONFIG_ROM_END_OFFSET` totally does the trick, and I see `CONFIG_MCUBOOT_UPDATE_FOOTER_SIZE`, so I guess I could just set those two.

In general though, would you recommend working with sysbuild enabled all the time even during development? Seems like a waste of resources to rebuild mcuboot every time but these are pretty vital bit of config that I never realized were missing.

nordicjm, 2024-09-30:
> Hey I see, `CONFIG_ROM_END_OFFSET` totally does the trick, and I see `CONFIG_MCUBOOT_UPDATE_FOOTER_SIZE`, so I guess I could just set those two.
>
> In general though, would you recommend working with [...]

Generally I would say in early development don't use MCUboot, that way you have more flash available for debug builds and easier debugging, when you get nearer to getting something stable, that's when MCUboot can be enabled and tested. At that point can build both separately and update the Kconfigs manually (can't do this if using some features which mandates sysbuild) or together, but for final builds always sysbuild.

fabiobaltieri, 2024-09-30:
Good stuff, thanks Jamie that does it for me. Would it make any sense to add a warning somewhere when building with `BOOTLOADER_MCUBOOT=y` and `CONFIG_ROM_END_OFFSET=0` or that's a valid configuration in any scenario?

nordicjm, 2024-09-30:
> Good stuff, thanks Jamie that does it for me. Would it make any sense to add a warning somewhere when building with `BOOTLOADER_MCUBOOT=y` and `CONFIG_ROM_END_OFFSET=0` to expose the potential problem or is that valid configuration in some scenarios? The swap may not be needed but maybe the TLV sector is always there?

TLV is always needed but I believe some people deal with dealing with those parts themselves outside the scope of the build system so would have them set to 0 (i.e. brix)

fabiobaltieri, 2024-09-30:
Cool, ok thanks for all the answers, that does it for me, would be nice to expose this somehow but I can't see an obvious way that is not potentially noisy, at least the information is recorded here, hopefully other people bumping into this will land here. Thanks!

closed 2024-09-30 by fabiobaltieri