zephyrproject-rtos/zephyr #79162: Lib JSON misbehaving

ofirshe, 2024-09-29:
### Discussed in https://github.com/zephyrproject-rtos/zephyr/discussions/79017

<div type='discussions-op-text'>

<sup>Originally posted by **ofirshe** September 25, 2024</sup>
Hey, 

I recently conducted a test and encountered an issue with the JSON encoding in my application, and I would appreciate your assistance. Below is a snippet of my code:

```
#define REGULATOR_NUM_DEVICES 17

// # include necessary header files 

/* Regulator */
typedef struct regulator_stats
{
    /* Voltage, in millivolts */
    bool is_read_vout_supported; 
    int32_t last_voltage_mv;
    int32_t max_voltage_mv;
    int32_t min_voltage_mv;
	
    /* Current, in milliamperes */
    bool is_read_iout_supported; 
    int32_t last_current_ma;
    int32_t max_current_ma;
    int32_t min_current_ma;

    uint32_t last_measurement_from_uptime_ms; 
} regulator_stats_t;

static const struct json_obj_descr regulator_stats_json_description[] =
{
    /* Voltage */
    JSON_OBJ_DESCR_PRIM(struct regulator_stats, is_read_vout_supported, JSON_TOK_TRUE),
    JSON_OBJ_DESCR_PRIM(struct regulator_stats, last_voltage_mv, JSON_TOK_NUMBER),
    JSON_OBJ_DESCR_PRIM(struct regulator_stats, max_voltage_mv, JSON_TOK_NUMBER),
    JSON_OBJ_DESCR_PRIM(struct regulator_stats, min_voltage_mv, JSON_TOK_NUMBER),

    /* Current */
    JSON_OBJ_DESCR_PRIM(struct regulator_stats, is_read_iout_supported, JSON_TOK_TRUE),
    JSON_OBJ_DESCR_PRIM(struct regulator_stats, last_current_ma, JSON_TOK_NUMBER),
    JSON_OBJ_DESCR_PRIM(struct regulator_stats, max_current_ma, JSON_TOK_NUMBER),
    JSON_OBJ_DESCR_PRIM(struct regulator_stats, min_current_ma, JSON_TOK_NUMBER),

    JSON_OBJ_DESCR_PRIM(struct regulator_stats, last_measurement_from_uptime_ms, JSON_TOK_NUMBER),
};

struct my_struct 
{
    regulator_stats_t regulator_devices_stats_arr[REGULATOR_NUM_DEVICES];
    uint32_t regulator_devices_stats_arr_len;
};

static struct my_struct ms;

static const struct json_obj_descr ms_desc[] = 
{
    /* Regulator Devices Stats */ 
    JSON_OBJ_DESCR_OBJ_ARRAY(struct my_struct, regulator_devices_stats_arr, REGULATOR_NUM_DEVICES,
                regulator_devices_stats_arr_len, regulator_stats_json_description,
                ARRAY_SIZE(regulator_stats_json_description)),
};

int main()
{
	int i;
	ssize_t encoded_len;
	char buffer[8192] = { 0 };

	for (i = 0; i < REGULATOR_NUM_DEVICES; i++)
	{
		ms.regulator_devices_stats_arr[i].is_read_vout_supported = ms.regulator_devices_stats_arr[i].is_read_iout_supported = false;
		ms.regulator_devices_stats_arr[i].max_voltage_mv = ms.regulator_devices_stats_arr[i].max_current_ma = INT32_MIN;
		ms.regulator_devices_stats_arr[i].min_voltage_mv = ms.regulator_devices_stats_arr[i].min_current_ma = INT32_MAX;
		ms.regulator_devices_stats_arr[i].last_voltage_mv = ms.regulator_devices_stats_arr[i].last_current_ma = 0;
		ms.regulator_devices_stats_arr[i].last_measurement_from_uptime_ms = 0;
	}
	ms.regulator_devices_stats_arr_len = REGULATOR_NUM_DEVICES;
	encoded_len = json_calc_encoded_len(ms_desc, ARRAY_SIZE(ms_desc), &ms);
	if (encoded_len < 0)
	{
		LOG_ERR("failed to calculate the string length of the encoded json; rc = %d", encoded_len);
	}
	json_obj_encode_buf(ms_desc, ARRAY_SIZE(ms_desc), &ms, buffer, sizeof(buffer));
	printf("%s\n", buffer);
}
```

I expected all 17 devices to be initialized with identical values like this:
`{"is_read_vout_supported":false,"last_voltage_mv":0,"max_voltage_mv":-2147483648,"min_voltage_mv":2147483647,"is_read_iout_supported":false,"last_current_ma":0,"max_current_ma":-2147483648,"min_current_ma":2147483647,"last_measurement_from_uptime_ms":0}
`
However, the console output shows that after the first device, the values become corrupted and shift between fields:
```
{"regulator_devices_stats_arr":[{"is_read_vout_supported":false,"last_voltage_mv":0,"max_voltage_mv":-2147483648,"min_voltage_mv":2147483647,"is_read_iout_supported":false,"last_current_ma":0,"max_current_ma":-2147483648,"min_current_ma":2147483647,"last_measurement_from_uptime_ms":0},
								{"is_read_vout_supported":false,"last_voltage_mv":0,"max_voltage_mv":0,"min_voltage_mv":-2147483648,"is_read_iout_supported":true,"last_current_ma":0,"max_current_ma":0,"min_current_ma":-2147483648,"last_measurement_from_uptime_ms":2147483647},
								{"is_read_vout_supported":true,"last_voltage_mv":0,"max_voltage_mv":0,"min_voltage_mv":0,"is_read_iout_supported":false,"last_current_ma":2147483647,"max_current_ma":0,"min_current_ma":0,"last_measurement_from_uptime_ms":-2147483648},
								{"is_read_vout_supported":false,"last_voltage_mv":2147483647,"max_voltage_mv":0,"min_voltage_mv":0,"is_read_iout_supported":false,"last_current_ma":-2147483648,"max_current_ma":2147483647,"min_current_ma":0,"last_measurement_from_uptime_ms":0},
								{"is_read_vout_supported":false,"last_voltage_mv":-2147483648,"max_voltage_mv":2147483647,"min_voltage_mv":0,"is_read_iout_supported":false,"last_current_ma":0,"max_current_ma":-2147483648,"min_current_ma":2147483647,"last_measurement_from_uptime_ms":0},
								{"is_read_vout_supported":false,"last_voltage_mv":0,"max_voltage_mv":-2147483648,"min_voltage_mv":2147483647,"is_read_iout_supported":false,"last_current_ma":0,"max_current_ma":0,"min_current_ma":-2147483648,"last_measurement_from_uptime_ms":2147483647},
								{"is_read_vout_supported":true,"last_voltage_mv":0,"max_voltage_mv":0,"min_voltage_mv":-2147483648,"is_read_iout_supported":true,"last_current_ma":0,"max_current_ma":0,"min_current_ma":0,"last_measurement_from_uptime_ms":-2147483648},
								{"is_read_vout_supported":false,"last_voltage_mv":2147483647,"max_voltage_mv":0,"min_voltage_mv":0,"is_read_iout_supported":false,"last_current_ma":2147483647,"max_current_ma":0,"min_current_ma":0,"last_measurement_from_uptime_ms":0},
								{"is_read_vout_supported":false,"last_voltage_mv":-2147483648,"max_voltage_mv":2147483647,"min_voltage_mv":0,"is_read_iout_supported":false,"last_current_ma":-2147483648,"max_current_ma":2147483647,"min_current_ma":0,"last_measurement_from_uptime_ms":0},
								{"is_read_vout_supported":false,"last_voltage_mv":0,"max_voltage_mv":-2147483648,"min_voltage_mv":2147483647,"is_read_iout_supported":false,"last_current_ma":0,"max_current_ma":-2147483648,"min_current_ma":2147483647,"last_measurement_from_uptime_ms":0},
								{"is_read_vout_supported":false,"last_voltage_mv":0,"max_voltage_mv":0,"min_voltage_mv":-2147483648,"is_read_iout_supported":true,"last_current_ma":0,"max_current_ma":0,"min_current_ma":-2147483648,"last_measurement_from_uptime_ms":2147483647},
								{"is_read_vout_supported":true,"last_voltage_mv":0,"max_voltage_mv":0,"min_voltage_mv":0,"is_read_iout_supported":false,"last_current_ma":2147483647,"max_current_ma":0,"min_current_ma":0,"last_measurement_from_uptime_ms":-2147483648},
								{"is_read_vout_supported":false,"last_voltage_mv":2147483647,"max_voltage_mv":0,"min_voltage_mv":0,"is_read_iout_supported":false,"last_current_ma":-2147483648,"max_current_ma":2147483647,"min_current_ma":0,"last_measurement_from_uptime_ms":0},
								{"is_read_vout_supported":false,"last_voltage_mv":-2147483648,"max_voltage_mv":2147483647,"min_voltage_mv":0,"is_read_iout_supported":false,"last_current_ma":0,"max_current_ma":-2147483648,"min_current_ma":2147483647,"last_measurement_from_uptime_ms":0},
								{"is_read_vout_supported":false,"last_voltage_mv":0,"max_voltage_mv":-2147483648,"min_voltage_mv":2147483647,"is_read_iout_supported":false,"last_current_ma":0,"max_current_ma":0,"min_current_ma":-2147483648,"last_measurement_from_uptime_ms":2147483647},
								{"is_read_vout_supported":true,"last_voltage_mv":0,"max_voltage_mv":0,"min_voltage_mv":-2147483648,"is_read_iout_supported":true,"last_current_ma":0,"max_current_ma":0,"min_current_ma":0,"last_measurement_from_uptime_ms":-2147483648},
								{"is_read_vout_supported":false,"last_voltage_mv":2147483647,"max_voltage_mv":0,"min_voltage_mv":0,"is_read_iout_supported":false,"last_current_ma":2147483647,"max_current_ma":0,"min_current_ma":0,"last_measurement_from_uptime_ms":0}]}
```
I am unsure if this is an issue with my implementation or a potential problem with the library. I am currently using the latest changes in the JSON library [878640f](https://github.com/zephyrproject-rtos/zephyr/commit/878640fa0817a5c6684758d68dcb824e24571612). 

Your help will be appreciated. 

Best regards. </div>

ofirshe, 2024-09-29:
Hey,

When I add __attribute__((packed, aligned(1))) to the struct regulator_stats, the issue gets resolved. However, I don't see any indication in the library that requires all structs to be packed. I also checked the [Zephyr json test source file](https://github.com/zephyrproject-rtos/zephyr/blob/main/tests/lib/json/src/main.c), and none of the structs there are packed either.

cfriedt, 2024-10-01:
Please use our bug template when reporting bugs. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

ofirshe, 2024-10-01:
@cfriedt, I have aligned with the bug template.

ofirshe, 2024-10-04:
Hi,

Could someone kindly help with this? We're utilizing JSON encoding extensively in our application, and it appears there's a problem with the encoding.

Any assistance would be greatly appreciated.

cfriedt, 2024-10-05:
@ofirshe - the main stack size is gigantic. You should be able to do what you need to with a stack size of maybe 4096 bytes (just a guess).

Also, the 8k buffer is larger than most stack sizes in Zephyr. Consider making it static, or allocating it from the heap?

ofirshe, 2024-10-05:
Hi @cfriedt,

You're right—the current approach isn't clean. Allocating this buffer on the stack isn't ideal, and I agree with your point about the main stack size; I just set an arbitrary number to avoid stack corruption. However, even if you move the buffer to the data segment, the same issue with the JSON encoding persists.

ofirshe, 2024-10-10:
Hi,

Could someone kindly take a look at the JSON encoding issue I'm experiencing?

Thanks again.

ofirshe, 2024-10-10:
Hi,

I recently pulled the latest json.c and json.h files from the Zephyr project based on commit [03029e7bd7](https://github.com/zephyrproject-rtos/zephyr/commit/03029e7bd7077834bb9c6b61bad19497b3f5558f), and created a basic application outside of the Zephyr OS. I ran the application on Linux and encountered the same issue as described previously.
And again, I found that adding `__attribute__((packed, aligned(1)))` to the `regulator_stats struct` resolves the issue.

To replicate this, you can clone the project, run make to compile, and execute ./main to see the result.
[json_encoding_failure_app.zip](https://github.com/user-attachments/files/17329334/json_encoding_failure_app.zip)

Best regards,
Ofir.

ofirshe, 2024-10-12:
Hi @cfriedt,

I experimented with various permutations of the field order in the regulator_stats struct and found that the following order works without the need for the `packed` attribute:
```
typedef struct regulator_stats
{
    /* Voltage, in millivolts */
    int32_t last_voltage_mv;
    int32_t max_voltage_mv;
    int32_t min_voltage_mv;
	
    /* Current, in milliamperes */
    int32_t last_current_ma;
    int32_t max_current_ma;
    int32_t min_current_ma;
    bool is_read_iout_supported;
    bool is_read_vout_supported;

    uint32_t last_measurement_from_uptime_ms; 
} regulator_stats_t;
``` 

Regardless of alignment considerations, the JSON library should support any valid field order and type combination, as long as the types are documented as supported. Since the documentation doesn’t mention any restrictions, this behavior appears to be unintended.

cfriedt, 2024-10-12:
> I experimented with various permutations of the field order in the regulator_stats struct and found that the following order works without the need for the `packed` attribute:

My first guess is that this would work as well.
```
typedef struct regulator_stats
{
     /* Voltage, in millivolts */
     int32_t last_voltage_mv;
     int32_t max_voltage_mv;
     int32_t min_voltage_mv;
     bool is_read_vout_supported;
 	
     /* Current, in milliamperes */
     int32_t last_current_ma;
     int32_t max_current_ma;
     int32_t min_current_ma;
     bool is_read_iout_supported;

     uint32_t last_measurement_from_uptime_ms; 
} regulator_stats_t;
```

> Regardless of alignment considerations, the JSON library should support any valid field order and type combination.

Maybe.

I'm not incredibly familiar with the JSON lib, but it might do what is relatively normal in C functions when given a pointer, which is assume that the target of the pointer is naturally aligned. Some architectures with certain micro-architectural enhancements are able to catch this and perform an unaligned fetch. However, others cannot - many in the embedded space. So expecting the JSON library (effectively the compiler) to detect and (actually) un-optimize for that use case might not be very tenable. It might be possible, but could require a lot of work / redesign of the macro system.

I would guess the easiest thing to do would be to trigger a `BUILD_ASSERT()` when struct members are not naturally aligned.

ofirshe, 2024-10-12:
Hi @cfriedt,

In case no attributes are applied to the struct, I believe the compiler should automatically enforce proper alignment by adding any necessary padding between fields. To verify this, I added a check for the alignment of each field, and indeed, every field address was aligned to a 4-byte boundary—no alignment issues were detected. This leads me to believe that alignment isn't the problem here.

Below is the modified loop, where I also checked the field addresses:
```
    for (i = 0; i < REGULATOR_NUM_DEVICES; i++)
    {
        // Initialize values
        ms.regulator_devices_stats_arr[i].is_read_vout_supported = ms.regulator_devices_stats_arr[i].is_read_iout_supported = false;
        ms.regulator_devices_stats_arr[i].max_voltage_mv = ms.regulator_devices_stats_arr[i].max_current_ma = INT32_MIN;
        ms.regulator_devices_stats_arr[i].min_voltage_mv = ms.regulator_devices_stats_arr[i].min_current_ma = INT32_MAX;
        ms.regulator_devices_stats_arr[i].last_voltage_mv = ms.regulator_devices_stats_arr[i].last_current_ma = 0;
        ms.regulator_devices_stats_arr[i].last_measurement_from_uptime_ms = 0;

        // Get the start address of the struct
        uintptr_t struct_addr = (uintptr_t)&ms.regulator_devices_stats_arr[i];
        printf("Start address of regulator_stats[%d]: %p\n", i, (void*)struct_addr);

        // Check alignment of every field in the struct
        uintptr_t addr_is_read_vout_supported = (uintptr_t)&ms.regulator_devices_stats_arr[i].is_read_vout_supported;
        uintptr_t addr_last_voltage_mv = (uintptr_t)&ms.regulator_devices_stats_arr[i].last_voltage_mv;
        uintptr_t addr_max_voltage_mv = (uintptr_t)&ms.regulator_devices_stats_arr[i].max_voltage_mv;
        uintptr_t addr_min_voltage_mv = (uintptr_t)&ms.regulator_devices_stats_arr[i].min_voltage_mv;
        uintptr_t addr_is_read_iout_supported = (uintptr_t)&ms.regulator_devices_stats_arr[i].is_read_iout_supported;
        uintptr_t addr_last_current_ma = (uintptr_t)&ms.regulator_devices_stats_arr[i].last_current_ma;
        uintptr_t addr_max_current_ma = (uintptr_t)&ms.regulator_devices_stats_arr[i].max_current_ma;
        uintptr_t addr_min_current_ma = (uintptr_t)&ms.regulator_devices_stats_arr[i].min_current_ma;
        uintptr_t addr_last_measurement_from_uptime_ms = (uintptr_t)&ms.regulator_devices_stats_arr[i].last_measurement_from_uptime_ms;

        // Check alignment for 4-byte boundaries (for int32_t and bool fields)
        if (addr_is_read_vout_supported % 4 != 0)
            printf("Error: regulator_stats[%d].is_read_vout_supported at %p is not aligned\n", i, (void*)addr_is_read_vout_supported);
        if (addr_last_voltage_mv % 4 != 0)
            printf("Error: regulator_stats[%d].last_voltage_mv at %p is not aligned\n", i, (void*)addr_last_voltage_mv);
        if (addr_max_voltage_mv % 4 != 0)
            printf("Error: regulator_stats[%d].max_voltage_mv at %p is not aligned\n", i, (void*)addr_max_voltage_mv);
        if (addr_min_voltage_mv % 4 != 0)
            printf("Error: regulator_stats[%d].min_voltage_mv at %p is not aligned\n", i, (void*)addr_min_voltage_mv);
        if (addr_is_read_iout_supported % 4 != 0)
            printf("Error: regulator_stats[%d].is_read_iout_supported at %p is not aligned\n", i, (void*)addr_is_read_iout_supported);
        if (addr_last_current_ma % 4 != 0)
            printf("Error: regulator_stats[%d].last_current_ma at %p is not aligned\n", i, (void*)addr_last_current_ma);
        if (addr_max_current_ma % 4 != 0)
            printf("Error: regulator_stats[%d].max_current_ma at %p is not aligned\n", i, (void*)addr_max_current_ma);
        if (addr_min_current_ma % 4 != 0)
            printf("Error: regulator_stats[%d].min_current_ma at %p is not aligned\n", i, (void*)addr_min_current_ma);
        if (addr_last_measurement_from_uptime_ms % 4 != 0)
            printf("Error: regulator_stats[%d].last_measurement_from_uptime_ms at %p is not aligned\n", i, (void*)addr_last_measurement_from_uptime_ms);
    }
```
Furthermore, to support my assumption, when the `packed` attribute is used, the JSON encoding works despite the compiler not adding padding, which results in unaligned access for some fields.

As mentioned earlier, the JSON library should support any valid field order and type combination as long as the types are documented as supported. Since no restrictions are mentioned in the documentation, this issue seems unintended. Would you know who might be able to assist in resolving this?

cfriedt, 2024-10-12:
In that case, I would agree - there likely is something wrong with the JSON library.

ofirshe, 2024-10-12:
Thanks, @cfriedt. I'm also not very familiar with the code within that library; @d3zd3z, could you please assist in resolving this issue? It appears to be a core problem with the library that shouldn't be overlooked.

ofirshe, 2024-10-15:
I believe I've identified the problem. It indeed relates to alignment issues.

For this struct and its JSON description:
```
typedef struct regulator_stats {
    /* Voltage, in millivolts */
    bool is_read_iout_supported;
    int32_t last_voltage_mv;
    int32_t max_voltage_mv;
    int32_t min_voltage_mv;

    /* Current, in milliamperes */
    bool is_read_vout_supported;
    int32_t last_current_ma;
    int32_t max_current_ma;
    int32_t min_current_ma;

    uint32_t last_measurement_from_uptime_ms;
} regulator_stats_t;

static const struct json_obj_descr regulator_stats_json_description[] = {
    /* Voltage */
    JSON_OBJ_DESCR_PRIM(struct regulator_stats, is_read_vout_supported, JSON_TOK_TRUE),
    JSON_OBJ_DESCR_PRIM(struct regulator_stats, last_voltage_mv, JSON_TOK_NUMBER),
    JSON_OBJ_DESCR_PRIM(struct regulator_stats, max_voltage_mv, JSON_TOK_NUMBER),
    JSON_OBJ_DESCR_PRIM(struct regulator_stats, min_voltage_mv, JSON_TOK_NUMBER),

    /* Current */
    JSON_OBJ_DESCR_PRIM(struct regulator_stats, is_read_iout_supported, JSON_TOK_TRUE),
    JSON_OBJ_DESCR_PRIM(struct regulator_stats, last_current_ma, JSON_TOK_NUMBER),
    JSON_OBJ_DESCR_PRIM(struct regulator_stats, max_current_ma, JSON_TOK_NUMBER),
    JSON_OBJ_DESCR_PRIM(struct regulator_stats, min_current_ma, JSON_TOK_NUMBER),

    JSON_OBJ_DESCR_PRIM(struct regulator_stats, last_measurement_from_uptime_ms, JSON_TOK_NUMBER),
};
```

The `get_elem_size()` function retrieves a size of 32, while the actual size is 36.

If we break it down in terms of potential memory layout, it would look something like this:

- bool is_read_iout_supported (1 byte)
- Padding (3 bytes)
- int32_t last_voltage_mv (4 bytes)
- int32_t max_voltage_mv (4 bytes)
- int32_t min_voltage_mv (4 bytes)
- bool is_read_vout_supported (1 byte)
- Padding (3 bytes)
- int32_t last_current_ma (4 bytes)
- int32_t max_current_ma (4 bytes)
- int32_t min_current_ma (4 bytes)
- uint32_t last_measurement_from_uptime_ms (4 bytes)

The total would be: `1 + 3 + 4 + 4 + 4 + 1 + 3 + 4 + 4 + 4 + 4 = 36`.

It seems that the `get_elem_size()` function does not account for the padding between the fields.

ofirshe, 2024-10-15:
@cfriedt @d3zd3z Should I create a new issue that summarizes all the messages from this current discussion, to make the problem I found more concrete and focused (and close this one)?

cfriedt, 2024-10-15:
@ofirshe - you can probably update your summary here. No need to create a new issue.

ofirshe, 2024-10-15:
@cfriedt I edited the main post.

ofirshe, 2024-10-22:
Hi, 

Could someone please assist in resolving this issue? I use the JSON library extensively, and I'm concerned that this problem may be affecting other parts of my application as well.

cfriedt, 2024-10-22:
@ofirshe - can you make a pull request to fix the issue?

ofirshe, 2024-10-27:
Hey @cfriedt,

I'm not very familiar with the library's code and am unsure how best to address this edge case. Given its complexity, I believe someone with more experience in the codebase would be better suited to handle it.

cfriedt, 2024-10-27:
@ofirshe - ok.

The issue has low priority, and there are workarounds. The maintainer likely will get to it eventually.

Thanks for your patience.

ofirshe, 2024-11-13:
Hey @cfriedt, 

Is there a chance this issue will be addressed soon? It seems quite tricky, and developers who aren’t careful about alignment issues might run into hidden problems with the JSON library.

cfriedt, 2024-11-13:
@d3zd3z ?

ofirshe, 2024-11-24:
@cfriedt @d3zd3z ?

ofirshe, 2024-12-06:
Hello @cfriedt,

As mentioned earlier, those who are not careful with alignment issues might run into subtle problems with the JSON library. I'm not sure if a workaround exists, as you suggested [here](https://github.com/zephyrproject-rtos/zephyr/issues/79162#issuecomment-2439959694).

Would it be possible for someone else to take a look at this? I noticed from git blame that many others were involved in the development of the library. Is there someone you would recommend who could assist?

Best regards,
Ofir.

cfriedt, 2024-12-06:
Definitely the assignee is the best for now, since @d3zd3z is the maintainer. You could 'at' others who have worked on JSON potentially here to attract attention and maybe another person could help.

ofirshe, 2025-01-20:
@d3zd3z Could you please provide an update on when this bug is estimated to be addressed?

closed 2025-02-04 by ofirshe