zephyrproject-rtos/zephyr #79178: Wrong logic in NXP driver for watchdog

VinFar, 2024-09-30:
**Describe the bug**

I am working on a i.MX from NXP and implemented the watchdog using the Zephyr API. During development the watchdog was not behaving as expected and I looked into the driver code and probably found a bug

**To Reproduce**

**In the following file the logic is wrong. The watchdog options for sleep and debug get set when the corresponding bit is 0, which is not correct according the [Zephyr API](https://docs.zephyrproject.org/latest/doxygen/html/group__watchdog__interface.html).**

https://github.com/zephyrproject-rtos/zephyr/blob/2815b6d710ad9c7ba25725ae34c402fc9153c427/drivers/watchdog/wdt_mcux_imx_wdog.c#L44-L48

I corrected it for my i.MX RT, but found two more files in Zephyr 3.6 where this issue is also existent:

https://github.com/zephyrproject-rtos/zephyr/blob/2815b6d710ad9c7ba25725ae34c402fc9153c427/drivers/watchdog/wdt_mcux_wdog.c#L45-L49

https://github.com/zephyrproject-rtos/zephyr/blob/2815b6d710ad9c7ba25725ae34c402fc9153c427/drivers/watchdog/wdt_mcux_wdog32.c#L53-L57

**Expected behavior**

The correct way would be invert the logic, just like this:

```
 data->wdog_config.workMode.enableStop = 
 	(options & WDT_OPT_PAUSE_IN_SLEEP) == WDT_OPT_PAUSE_IN_SLEEP; 
  
 data->wdog_config.workMode.enableDebug = 
 	(options & WDT_OPT_PAUSE_HALTED_BY_DBG) == WDT_OPT_PAUSE_HALTED_BY_DBG; 
```
I fixed it locally, but it would be nice to have it in the next Zephyr release included :) 

**Impact**

The watchdog runs just fine and also triggers fine. But it does not stop when the device is put to sleep, which can cause miss-triggers. Also when debugging the code and halting the core wit the debugger the watchdog also triggers.

This is annyoing, especially when you don't know the cause of the reset. 

**Logs and console output**

N/A

**Environment (please complete the following information):**

Probably all i.MX processors, also all S32 processors. I am not sure, which processor are using the file `wdt_mcux_wdog.c` and `wdt_mxuc_wdog32.c`, but this bug could have a big scope

**Should I create a PR for this to be fixed? I can open one for the 3 above mentioned files**

pdgendt, 2024-09-30:
If you could submit a PR, that'd be great. If these bugs are confirmed, they should be backported too.

dleach02, 2024-10-24:
This bug was with v3.6 but was fixed in v3.7

closed 2024-10-24 by dleach02