zephyrproject-rtos/zephyr #79195: test: tests/lib/multi_heap/libraries.multi_heap FAILED on up2

KamilxPaszkiet, 2024-09-30:
**Describe the bug**
Test libraries.multi_heap failed
Repro 100%

**To Reproduce**
`twister --device-testing --hardware-map up2.map -s tests/lib/multi_heap/libraries.multi_heap`

**Logs and console output**
```
DEBUG   - DEVICE: *** Booting Zephyr OS build v3.7.0-3643-gf810e1318f38 ***
DEBUG   - DEVICE: Running TESTSUITE mheap_api
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_k_aligned_alloc
DEBUG   - DEVICE: PASS - test_k_aligned_alloc in 0.001 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_malloc_in_isr
DEBUG   - DEVICE: PASS - test_malloc_in_isr in 0.001 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_malloc_in_thread
DEBUG   - DEVICE: PASS - test_malloc_in_thread in 0.001 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_mheap_calloc
DEBUG   - DEVICE: Assertion failed at WEST_TOPDIR/zephyr/tests/lib/multi_heap/src/test_mheap_api.c:190: mheap_api_test_mheap_calloc: (mem is not NULL)
DEBUG   - DEVICE: calloc operation failed
DEBUG   - DEVICE: FAIL - test_mheap_calloc in 0.017 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_mheap_malloc_align4
DEBUG   - DEVICE: PASS - test_mheap_malloc_align4 in 0.001 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_mheap_malloc_free
DEBUG   - DEVICE: Assertion failed at WEST_TOPDIR/zephyr/tests/lib/multi_heap/src/test_mheap_api.c:86: mheap_api_test_mheap_malloc_free: block_fail is not NULL
DEBUG   - DEVICE: FAIL - test_mheap_malloc_free in 0.015 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_mheap_realloc
DEBUG   - DEVICE: Assertion failed at WEST_TOPDIR/zephyr/tests/lib/multi_heap/src/test_mheap_api.c:132: mheap_api_test_mheap_realloc: block2 is not NULL
DEBUG   - DEVICE: FAIL - test_mheap_realloc in 0.014 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_mheap_threadsafe
DEBUG   - DEVICE: PASS - test_mheap_threadsafe in 0.001 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_multi_heap
DEBUG   - DEVICE: PASS - test_multi_heap in 0.001 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_sys_heap_mem_pool_assign
DEBUG   - DEVICE: Assertion failed at WEST_TOPDIR/zephyr/tests/lib/multi_heap/src/test_mheap_api.c:259: mheap_api_test_sys_heap_mem_pool_assign: ((char *)z_thread_malloc(BLK_SIZE_MAX * 2) is not NULL)
DEBUG   - DEVICE: overflow check failed
DEBUG   - DEVICE: FAIL - test_sys_heap_mem_pool_assign in 0.021 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: TESTSUITE mheap_api failed.
DEBUG   - DEVICE: ------ TESTSUITE SUMMARY START ------
DEBUG   - DEVICE: SUITE FAIL -  60.00% [mheap_api]: pass = 6, fail = 4, skip = 0, total = 10 duration = 0.073 seconds
DEBUG   - DEVICE: - PASS - [mheap_api.test_k_aligned_alloc] duration = 0.001 seconds
DEBUG   - DEVICE: - PASS - [mheap_api.test_malloc_in_isr] duration = 0.001 seconds
DEBUG   - DEVICE: - PASS - [mheap_api.test_malloc_in_thread] duration = 0.001 seconds
DEBUG   - DEVICE: - FAIL - [mheap_api.test_mheap_calloc] duration = 0.017 seconds
DEBUG   - DEVICE: - PASS - [mheap_api.test_mheap_malloc_align4] duration = 0.001 seconds
DEBUG   - DEVICE: - FAIL - [mheap_api.test_mheap_malloc_free] duration = 0.015 seconds
DEBUG   - DEVICE: - FAIL - [mheap_api.test_mheap_realloc] duration = 0.014 seconds
DEBUG   - DEVICE: - PASS - [mheap_api.test_mheap_threadsafe] duration = 0.001 seconds
DEBUG   - DEVICE: - PASS - [mheap_api.test_multi_heap] duration = 0.001 seconds
DEBUG   - DEVICE: - FAIL - [mheap_api.test_sys_heap_mem_pool_assign] duration = 0.021 seconds
DEBUG   - DEVICE: ------ TESTSUITE SUMMARY END ------
```
**Expected behavior**
Test passed 

**Environment:**

 - Linux Ubuntu 22.04
 - Toolchain: Zephyr SDK 16.8
 - Zephyr version: v3.7.0-3643-gf810e1318f38

dkalowsk, 2024-10-01:
@edersondisouza can you please assign this to the right Intel person

KamilxPaszkiet, 2024-10-02:
> `west boards -n up2` returns no boards here. Which board is this?

Sorry for my mistake. I meant up_squared board.

kwd-doodling, 2024-10-05:
@henrikbrixandersen @dkalowsk @KamilxPaszkiet 
could you assign the right owner? Although I'm one Intel guy, but I have no idea what up_squared board is.

laurenmurphyx64, 2024-11-13:
As an FYI, the last time I can see this test work was two years ago, it broke sometime before 11 months ago it was moved to the folder it currently resides in (`2ce20e5df64dece42b122359ff77328c622a9a72`)

PR #85256 (2025-03-07): tests: lib: multi_heap fix up_squared test failures
Fixes test failures on up_squared caused by ACPI adding to heap size (K_HEAP_MEM_POOL_SIZE). The test assumed the heap size always matched the Kconfig CONFIG_HEAP_MEM_POOL_SIZE.

Fixes #79195