zephyrproject-rtos/zephyr #79197: test: message_bus.zbus.hlp_priority_boost FAILED on up2

KamilxPaszkiet, 2024-09-30:
**Describe the bug**
Test message_bus.zbus.hlp_priority_boost failed
Repro 100%

**To Reproduce**
`twister --device-testing --hardware-map up2.map -s tests/subsys/zbus/hlp_priority_boost/message_bus.zbus.hlp_priority_boost`

**Logs and console output**
```
DEBUG   - DEVICE: *** Booting Zephyr OS build v3.7.0-3643-gf810e1318f38 ***
DEBUG   - DEVICE: Running TESTSUITE hlp_priority_boost
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_priority_elevation
DEBUG   - DEVICE: msub1 level: 0
DEBUG   - DEVICE: sub1 level: 0
DEBUG   - DEVICE: Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/zbus/hlp_priority_boost/src/main.c:135: hlp_priority_boost_test_priority_elevation: (prio == 1 is false)
DEBUG   - DEVICE: msub1 level: 0
DEBUG   - DEVICE: The priority must be 1, but it is 8sub1 level: 0
DEBUG   - DEVICE: FAIL - test_priority_elevation in 0.125 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: TESTSUITE hlp_priority_boost failed.
DEBUG   - DEVICE: ------ TESTSUITE SUMMARY START ------
DEBUG   - DEVICE: SUITE FAIL -   0.00% [hlp_priority_boost]: pass = 0, fail = 1, skip = 0, total = 1 duration = 0.125 seconds
DEBUG   - DEVICE: - FAIL - [hlp_priority_boost.test_priority_elevation] duration = 0.125 seconds
DEBUG   - DEVICE: ------ TESTSUITE SUMMARY END ------
```
**Expected behavior**
Test passed 

**Environment:**

 - Linux Ubuntu 22.04
 - Toolchain: Zephyr SDK 16.8
 - Zephyr version: v3.7.0-3643-gf810e1318f38

dkalowsk, 2024-10-01:
@edersondisouza believe this is an Intel board, can you please re-assign to the right person.

rodrigopex, 2024-10-01:
@dkalowsk Should I act now or wait for further investigation from the Intel side?

dkalowsk, 2024-10-01:
> @dkalowsk Should I act now or wait for further investigation from the Intel side?

@rodrigopex if you want sure, but believe you got assigned due to the label for Zbus.  I'd wait for @edersondisouza because I don't think there are any UP2 boards in the tree right now.

KamilxPaszkiet, 2024-10-02:
> `west boards -n up2` returns no boards here. Which board is this?

Sorry for my mistake. I meant up_squared board.

edersondisouza, 2024-12-04:
So I bisected this to 027a1c30cd4423ab26575e6285efadecb13a00cc - assigning to @dcpleung for now, but he's on vacations, so I may end up revisiting it.

dcpleung, 2025-01-08:
@rodrigopex how does the priority boosting work under SMP? If the consumer is running on another thread, would the listener callback be called after the consumers have already processed the messages? and thus the priority boost is no longer necessary?

rodrigopex, 2025-01-08:
@dcpleung 

> @rodrigopex how does the priority boosting work under SMP? If the consumer is running on another thread, would the listener callback be called after the consumers have already processed the messages? and thus the priority boost is no longer necessary?

I did not have access to hardware with SMP capability, so I could not perform the proper validation in SMP environments. In theory, that should behave the same. In zbus, we only use semaphores and `k_spin_lock`, which are meant to work in SMP. 

Listeners are function calls executed by the publisher during the notification process. Priority boost works only for subscribers, which are meant to run in separate threads. The priority boost temporarily elevates the priority of the publisher thread during the VDED execution based on the channel's subscriber list priorities. That would avoid priority inversion and starvation. 

Even in SMP, a priority boost will be necessary since we can face the same issues with deadlock and starvation as a single core architecture.

PR #86948 (2025-03-14): tests: zbus/hlp_priority_boost: use 1cpu setup
The zbus hlp_priority_boost test requires thread priority boosting. If it is running SMP, it is possible that the message has already been processed before the check for priority boosting is done, thus failing the test. So change the test to use 1 CPU only.

Fixes #79197