zephyrproject-rtos/zephyr #79270: LE Audio: Scan Delegator: Bad broadcast code

jthm-ot, 2024-10-01:
Describe the bug
When a broadcast assistant connected to the bap_broadcast_sink sample adds an encrypted source, then the bap_broadcast_sink sample does not set the bad_code data correctly (bap_scan_delegator / receive_state_updated) in the receive state updated notification, when a wrong broadcast code is proved by the assistant. The bad_code data apparently only includes zeros!

To Reproduce
Steps to reproduce the behavior:

Disable scan self In the bap_broadcast_sink sample (CONFIG_SCAN_SELF=n)
Build and flash bap_broadcast_sink sample (e.g. nRF5340 Audio DK board)
Add an encrypted source from the broadcast assistant
Provide a wrong broadcast code from the broadcast assistant when requested

Expected behavior
The recv_state_updated callback in the bap_broadcast_sink sample should have these parameters:

recv_state->encrypt_state = BT_BAP_BIG_ENC_STATE_BAD_CODE
recv_state->bad_code = <wrong broadcast code>

Impact

The BAP specification is not fulfilled (6.5.7. Setting Broadcast_Codes) 

Zephyr SHA used: https://github.com/zephyrproject-rtos/zephyr/commit/eaf92a1494cc0e245f14f22ad1d1d42f363a07b3

Thalley, 2024-10-01:
I think this was fixed by https://github.com/zephyrproject-rtos/zephyr/pull/77657 - Can you verify?

jthm-ot, 2024-10-14:
Correct. Thanks. I have verified the fix

closed 2024-10-14 by Thalley