zephyrproject-rtos/zephyr #79273: VLAN not working with eth_nxp_enet driver

anhuba, 2024-10-01:
When switching from v3.6 to v3.7 we realized that VLAN is not working on our SOC_MIMXRT1064 board anymore.
After analyzing the problem we found out the problem is the default setting of the new nxp_enet driver (with the eth_mcux in v3.6 it worked fine)

The new driver has the HW acceleration (ETH_NXP_ENET_HW_ACCELERATION) enabled by default.
If using CONFIG_NET_VLAN the checksums for IP, TCP/UDP are calculated in SW as the VLAN interface does not use the information from the lower interface. The HW then calculates the checksums with non-zeroed values which results in checksum with the value 0xffff instead of the correct value.

If CONFIG_ETH_NXP_ENET_HW_ACCELERATION is disabled in the config and everything calculated in SW it works fine.

To make use of the HW acceleration and VLAN at the same time, I have added the following lines, which work fine in our setup:

```
--- A/zephyr/subsys/net/ip/net_if.c	2024-10-01 13:31:51.000000000 
+++ B/zephyr/subsys/net/ip/net_if.c	2024-10-01 11:02:28.000000000 
@@ -5169,12 +5169,16 @@
 			      enum net_if_checksum_type chksum_type)
 {
 #if defined(CONFIG_NET_L2_ETHERNET)
 	struct ethernet_config config;
 	enum ethernet_config_type config_type;
 
+	if (IS_ENABLED(CONFIG_NET_VLAN) && net_eth_is_vlan_interface(iface)) {
+		iface = net_eth_get_vlan_main(iface);
+	}
+
 	if (net_if_l2(iface) != &NET_L2_GET_NAME(ETHERNET)) {
 		/* For VLANs, figure out the main Ethernet interface and
 		 * get the offloading capabilities from it.
 		 */
 		if (IS_ENABLED(CONFIG_NET_VLAN) && net_eth_is_vlan_interface(iface)) {
 			iface = net_eth_get_vlan_main(iface);

```

jukkar, 2024-10-01:
This should be fixed by this https://github.com/zephyrproject-rtos/zephyr/pull/78745

anhuba, 2024-10-01:
oops did not find the fix...
already fixed in 00bb90a42e7f583704090c5e38e1b3fd933cf409

closed 2024-10-01 by anhuba