zephyrproject-rtos/zephyr #79276: test: portability.posix.env.newlib/portability.posix.common.newlib Fatal error on nsim

KamilxPaszkiet, 2024-10-01:
**Describe the bug**
Tests:
tests/posix/env/portability.posix.env.newlib
and
tests/posix/common/portability.posix.common.newlib
end with error.
I think these are similar problems.

**To Reproduce**
`twister -p nsim_em -s tests/posix/env/portability.posix.env.newlib`
`twister -p nsim_em -s tests/posix/common/portability.posix.common.newlib`

**Logs and console output**
tests/posix/env/portability.posix.env.newlib:
```
*** Booting Zephyr OS build v3.7.0-3583-g0a20f24d7090 ***
Running TESTSUITE env
===================================================================
START - test_getenv
 PASS - test_getenv in 0.001 seconds
===================================================================
START - test_getenv_r
 PASS - test_getenv_r in 0.002 seconds
===================================================================
START - test_setenv
 PASS - test_setenv in 0.001 seconds
===================================================================
START - test_unsetenv
 PASS - test_unsetenv in 0.002 seconds
===================================================================
START - test_watertight
E: ***** Exception vector: 0x6, cause code: 0x1, parameter 0x2
E: Address 0x80000b40
E: EV_ProtV
E: Memory read protection violation (stack checking scheme)
E:  r0: 0x80000b44  r1: 0x0  r2: 0x8000009c  r3: 0x80000b47
E:  r4: 0x80808080  r5: 0x3d3d3d3d  r6: 0x3d6f7869  r7: 0x0
E:  r8: 0x1010101  r9: 0x0 r10: 0x80002cd0 r11: 0x80808080
E: r12: 0x3 r13: 0x0  pc: 0x7ae0
E:  blink: 0x14c status32: 0x80086a06
E: lp_end: 0x7bea lp_start: 0x7bdc lp_count: 0x0
E: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
E: Current thread: 0x8000009c (test_watertight)
E: Halting system
Console now belongs to simulator.
```
tests/posix/common/portability.posix.common.newlib:
```
*** Booting Zephyr OS build v3.7.0-3583-g0a20f24d7090 ***
Running TESTSUITE barrier
===================================================================
START - test_barrier
 PASS - test_barrier in 0.001 seconds
===================================================================
TESTSUITE barrier succeeded
Running TESTSUITE clock
===================================================================
START - test_clock_getcpuclockid
 PASS - test_clock_getcpuclockid in 0.001 seconds
===================================================================
START - test_clock_getres
 PASS - test_clock_getres in 0.001 seconds
===================================================================
START - test_clock_gettime
 PASS - test_clock_gettime in 0.001 seconds
===================================================================
START - test_clock_settime
 PASS - test_clock_settime in 0.001 seconds
===================================================================
START - test_gettimeofday
[... 318 lines cut ...]
===================================================================
START - test_signal_strsignal
E: ***** Exception vector: 0x6, cause code: 0x1, parameter 0x2
E: Address 0x800041b0
E: EV_ProtV
E: Memory read protection violation (stack checking scheme)
E:  r0: 0x800041b4  r1: 0x800040f0  r2: 0x800016e0  r3: 0x800041b7
E:  r4: 0x1010101  r5: 0xb  r6: 0x1  r7: 0x1010101
E:  r8: 0x8000412c  r9: 0x0 r10: 0x0 r11: 0x0
E: r12: 0xa r13: 0x27394 pc: 0x186fc
E:  blink: 0x776e status32: 0x80086a06
E: lp_end: 0x1c3cc lp_start: 0x1c3c4 lp_count: 0x0
E: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
E: Current thread: 0x800012c4 (test_signal_strsignal)
E: Halting system
Console now belongs to simulator.
```
**Expected behavior**
Tests passed 

**Environment:**

 - Linux Ubuntu 22.04
 - Toolchain: Zephyr SDK 16.8
 - Zephyr version: v3.7.0-3583-g0a20f24d7090

gbarkadiusz, 2024-11-27:
Observed also on `nsim/nsim_hs/mpuv6.`
`nsim/nsim_hs/mpuv6        tests/posix/common/portability.posix.common.newlib FAILED Timeout (nsim 2.060s)`
_twister-out/nsim_nsim_hs_mpuv6/tests/posix/common/portability.posix.common.newlib/handler.log_
```
...
===================================================================
START - test_sigismember_oor
 PASS - test_sigismember_oor in 0.001 seconds
===================================================================
START - test_signal_strsignal
E: ***** Exception vector: 0x6, cause code: 0x1, parameter 0x2
E: Address 0x80003da0
E: EV_ProtV
E: Memory read protection violation (stack checking scheme)
E:  r0: 0x80003da4  r1: 0x80003ce0  r2: 0x800016cc  r3: 0x80003da7
E:  r4: 0xb  r5: 0x1  r6: 0x2e  r7: 0x0
E:  r8: 0x0  r9: 0x0 r10: 0x0 r11: 0x2a1bd
E: r12: 0xa r13: 0x2758c  pc: 0x16fb4
E:  blink: 0x76ba status32: 0x80086a02
E: lp_end: 0x1a9ac lp_start: 0x1a9a4 lp_count: 0x0
E: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
E: Current thread: 0x800012b4 (test_s)
E: Halting system
```

ruuddw, 2024-11-27:
Looks like a problem with the 'optimized' newlib strlen implementation that compares words instead of bytes.
These instructions trigger the problem:
```
strlen       or           %r3,%r0,7
strlen+0x04  ld           %r2,[%r3,-7] 
```
ARC hardware stack bounds checking not only ensures the stack stays within bounds but also raises an exception on non-sp based memory references to the unallocated stack (between stack_top and sp).
The word alignment (`or 7`) and correction (-7 offset) points to a valid address (4 byte` buf`), but `r3` points to the unallocated stack.

Not sure yet who's to blame and where to fix: compiler, newlib (or even hardware to not check the base but base+offset...)?

cfriedt, 2024-11-27:
This sounds like a newlib problem

ruuddw, 2024-12-02:
@keith-packard This looks like a similar problem as fixed in https://github.com/picolibc/picolibc/pull/508
Any suggestions? I like the notion of "PICOLIBC_NO_OUT_OF_BOUNDS_READS". Here, the trigger is hardware stack protection, but it could have been any protection (e.g. mem domains).

keith-packard, 2024-12-02:
The only reason we aren't hitting this with picolibc is because the SDK only includes -Os versions, which don't use the arc assembly code...

I'm a bit confused though -- is the stack not 8-byte aligned in both base and size? If it were, how could these accesses be out-of-bounds? Or is the issue that SP is actually 0x80003da4 and we're reading below that address?

I assume that switching to the C versions would solve this problem? That code never reads before the string and only reads long values on long-aligned addresses -- it will read bytes past the end of the string, but never past the longword containing the end of the string. If so, fixing the newlib build to skip the arc assembly code would be fairly easy. We'll want to also make the same changes in picolibc. Alternatively, some brave volunteer could offer to fix the arc assembly code, once we understand the cause.

ruuddw, 2024-12-03:
Correct, this is not about stack allocation/alignment, it is about the value of the stack pointer + ARC hardware stack checks. Non-SP based accesses to the 'unallocated' stack (a better description would be the unused part of the total allocated stack) trigger an exception.
It sounds like the "PICOLIBC_NO_OUT_OF_BOUNDS_READS" was added for a different problem (stack alignment).

I agree that switching to the C version may be the better solution. I'm not sure both C implementations (the performance optimized one, as well as the simple per byte loop) would work. The "-Os" flag I assume picks the byte loop one? That should always work, but misses the optimization when ARC hardware stack checking is not configured in the hardware or not enabled.

keith-packard, 2024-12-03:
Reading the ARC asm code, it sure looks like the issue should only occur when the buffer is not 8-byte aligned and sits at the same address as SP. If that's true, then the C version should work fine -- it aligns the source address by doing byte-by-byte compares, then performs long-at-a-time compares for the remainder of the string. As long as the stack is allocated on long-aligned boundaries, then it will not access any out-of-bounds addresses.

Performance will be reduced as the C code operates on longs while the asm code operates on pairs of 4-byte values, and the asm code takes advantage of additional instructions which the C code does not.

I suggest testing using picolibc as a module and configuring the build to optimize for speed instead of size. I'm unable to test this myself as I don't have access to 'nsim'.

ruuddw, 2024-12-05:
Thanks @keith-packard, this looks like a problem for the assembly optimized code only. The standard C version doesn't trigger the problem (neither the speed or size optimized variant). But currently only when size optimization is chosen the C version is used. The assembly code and C code have similar optimizations, but the C version doesn't seem to do out of bound reads (as you explained, that may only happen with unaligned pointers). I guess we should either have a new 'no out of bounds' config option that doesn't pick the assembly version when set, or completely drop the ARC assembly versions of strlen in favor of the standard C version (I didn't compare the compiler generated vs assembly variants).

keith-packard, 2024-12-05:
I can easily disable the assembly versions in picolibc; it looks like there would be only a modest performance improvement when using them. Does someone want to make the equivalent changes in newlib? Or should we disable newlib as an option on arc targets?

abrodkin, 2024-12-16:
> I can easily disable the assembly versions in picolibc; it looks like there would be only a modest performance improvement when using them. Does someone want to make the equivalent changes in newlib? Or should we disable newlib as an option on arc targets?

I guess that was already done here https://github.com/picolibc/picolibc/pull/889, right?

keith-packard, 2024-12-16:
> I guess that was already done here [picolibc/picolibc#889](https://github.com/picolibc/picolibc/pull/889), right?

Yup.

closed 2025-03-01 by github-actions[bot]