zephyrproject-rtos/zephyr #79283: max32672 cpp build errors in weekly full twister run

MaureenHelm, 2024-10-01:
**Describe the bug**

`max32672fthr` and `max32672evkit` boards are failing multiple cpp and cxx tests in the weekly full twister run due to build warnings that get promoted to errors.
https://github.com/zephyrproject-rtos/zephyr/runs/30826322038
https://github.com/zephyrproject-rtos/zephyr/actions/runs/11089211666/job/30810087093#step:13:7071

**To Reproduce**

```console
$ west build -p -b max32672fthr samples/cpp/cpp_synchronization/ -T sample.cpp.synchronization
```

**Expected behavior**

Build without warnings

**Impact**

Noise in the weekly full twister run burying other failures.

**Logs and console output**

```console
INFO    - 5051/6958 max32672fthr              samples/cpp/cpp_synchronization/sample.cpp.synchronization  ERROR Build failure (build)
INFO    - /__w/zephyr/zephyr/twister-out/max32672fthr/samples/cpp/cpp_synchronization/sample.cpp.synchronization/build.log
ERROR   - Loading Zephyr default modules (Zephyr base).
-- Application: /__w/zephyr/zephyr/samples/cpp/cpp_synchronization
-- CMake version: 3.22.1
-- Found Python3: /usr/bin/python (found suitable version "3.10.12", minimum required is "3.10") found components: Interpreter 
-- Cache files will be written to: /root/.cache/zephyr
-- Zephyr version: 3.7.99 (/__w/zephyr/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: max32672fthr, qualifiers: max32672
-- Found host-tools: zephyr 0.16.8 (/opt/toolchains/zephyr-sdk-0.16.8)
-- Found toolchain: zephyr 0.16.8 (/opt/toolchains/zephyr-sdk-0.16.8)
-- Found Dtc: /opt/toolchains/zephyr-sdk-0.16.8/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /__w/zephyr/zephyr/boards/adi/max32672fthr/max32672fthr.dts
-- Generated zephyr.dts: /__w/zephyr/zephyr/twister-out/max32672fthr/samples/cpp/cpp_synchronization/sample.cpp.synchronization/zephyr/zephyr.dts
-- Generated pickled edt: /__w/zephyr/zephyr/twister-out/max32672fthr/samples/cpp/cpp_synchronization/sample.cpp.synchronization/zephyr/edt.pickle
-- Generated zephyr.dts: /__w/zephyr/zephyr/twister-out/max32672fthr/samples/cpp/cpp_synchronization/sample.cpp.synchronization/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /__w/zephyr/zephyr/twister-out/max32672fthr/samples/cpp/cpp_synchronization/sample.cpp.synchronization/zephyr/include/generated/zephyr/devicetree_generated.h
-- Including generated dts.cmake file: /__w/zephyr/zephyr/twister-out/max32672fthr/samples/cpp/cpp_synchronization/sample.cpp.synchronization/zephyr/dts.cmake
Parsing /__w/zephyr/zephyr/Kconfig
Loaded configuration '/__w/zephyr/zephyr/boards/adi/max32672fthr/max32672fthr_defconfig'
Merged configuration '/__w/zephyr/zephyr/samples/cpp/cpp_synchronization/prj.conf'
Merged configuration '/__w/zephyr/zephyr/twister-out/max32672fthr/samples/cpp/cpp_synchronization/sample.cpp.synchronization/zephyr/misc/generated/extra_kconfig_options.conf'
Configuration saved to '/__w/zephyr/zephyr/twister-out/max32672fthr/samples/cpp/cpp_synchronization/sample.cpp.synchronization/zephyr/.config'
Kconfig header saved to '/__w/zephyr/zephyr/twister-out/max32672fthr/samples/cpp/cpp_synchronization/sample.cpp.synchronization/zephyr/include/generated/zephyr/autoconf.h'
-- Found GnuLd: /opt/toolchains/zephyr-sdk-0.16.8/arm-zephyr-eabi/arm-zephyr-eabi/bin/ld.bfd (found version "2.38") 
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /opt/toolchains/zephyr-sdk-0.16.8/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Setting build type to 'MinSizeRel' as none was specified.
-- Using ccache: /usr/local/bin/ccache
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    TC_NAME
    TC_RUNID

-- Build files have been written to: /__w/zephyr/zephyr/twister-out/max32672fthr/samples/cpp/cpp_synchronization/sample.cpp.synchronization
[1/139] Generating include/generated/zephyr/version.h
-- Zephyr version: 3.7.99 (/__w/zephyr/zephyr), build: v3.7.0-3607-g0d118ec1abe0
[2/139] Preparing syscall dependency handling

[3/139] Generating misc/generated/syscalls_subdirs.trigger
[4/139] Generating misc/generated/syscalls.json, misc/generated/struct_tags.json
[... 15 lines cut ...]
                 from /__w/zephyr/zephyr/include/zephyr/arch/arm/arch.h:30,
                 from /__w/zephyr/zephyr/include/zephyr/arch/cpu.h:19,
                 from /__w/zephyr/zephyr/include/zephyr/kernel_includes.h:36,
                 from /__w/zephyr/zephyr/include/zephyr/kernel.h:17,
                 from /__w/zephyr/zephyr/samples/cpp/cpp_synchronization/src/main.cpp:22:
/__w/zephyr/modules/hal/cmsis/CMSIS/Core/Include/core_cm4.h:222: error: "__I" redefined [-Werror]
  222 |   #define   __I     volatile             /*!< Defines 'read only' permissions */
      | 
In file included from /__w/zephyr/modules/hal/adi/MAX/Libraries/CMSIS/Device/Maxim/MAX32672/Include/max32672.h:35:
/__w/zephyr/modules/hal/adi/MAX/Libraries/CMSIS/Device/Maxim/MAX32672/Include/trimsir_regs.h:53: note: this is the location of the previous definition
   53 | #define __I  volatile const
      | 
cc1plus: all warnings being treated as errors
ninja: build stopped: subcommand failed.
```

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain: Zephyr SDK 0.16.8
 - Commit SHA: 0d118ec1abe0ca7178eb8292212121064c5cb3fe

**Additional context**

@hfakkiz can you take a look?

mmahadevan108, 2024-11-06:
@MaureenHelm , can this issue be closed?

ttmut, 2024-11-11:
The failing test seems to have passed in recent runs so I believe this can be closed.

https://github.com/zephyrproject-rtos/zephyr/actions/runs/11761688794/job/32763760993#step:13:757

MaureenHelm, 2024-11-13:
Fixed by #79662

closed 2024-11-13 by MaureenHelm