zephyrproject-rtos/zephyr #79284: Nominate myself (jonas-rem) as Contributor

jonas-rem, 2024-10-01:
# Background

The [TSC Project Roles] defines the main roles for the Zephyr Project, including
Maintainer, Collaborator, and Contributor.

By default anyone that contributes code or documentation is a Contributor, but
with the lowest [GitHub Permission Level] of Read. For example, Contributors
with Read permission do not have the permission to add reviewers to a pull
request.

Use this template to nominate a GitHub user for the Contributor role with
Triage permission level, which allows the user to add reviewers to a pull
request and be added as a reviewer by other users.

# Nomination

## GitHub User

Provide the following information about the GitHub user:

1. Jonas Remmert
1. jonas-rem
1. PHYTEC

## Supporting Documents

- https://github.com/zephyrproject-rtos/zephyr/pull/76975
- https://github.com/zephyrproject-rtos/zephyr/pull/54734
- https://github.com/zephyrproject-rtos/zephyr/pull/65416
- https://github.com/zephyrproject-rtos/zephyr/pull/58179
- https://github.com/zephyrproject-rtos/zephyr/pull/49724

[TSC Project Roles]: <https://docs.zephyrproject.org/latest/project/project_roles.html>
[GitHub Permission Level]: <https://docs.github.com/en/organizations/managing-access-to-your-organizations-repositories/repository-roles-for-an-organization>

stephanosio, 2024-10-04:
@jonas-rem has been added as a Contributor with Triage permission level.

jonas-rem, 2024-10-04:
> @jonas-rem has been added as a Contributor with Triage permission level.

Thanks! :)

dnltz, 2024-10-08:
@stephanosio can you please promote @jonas-rem to Collaborator? He just got added to the recently added `PHYTEC Platforms`.

https://github.com/zephyrproject-rtos/zephyr/blob/41e4b5323ac3888c45763ed7574c892cf3907fb5/MAINTAINERS.yml#L3090

jonas-rem, 2024-10-23:
@stephanosio now that I am a collaborator with Triage permission level, my review indicators should show green (previously grey), correct? I reviewed and tested #79850, but far as I understood my review is not counted. I accepted the invitation, is there anything else required?

stephanosio, 2024-10-23:
@jonas-rem has been added as a Collaborator.

avisconti, 2024-10-23:
@jonas-rem are you a Collaborator now? Congratulations!

avisconti, 2024-10-23:
@stephanosio I don't see @jonas-rem here:
https://github.com/orgs/zephyrproject-rtos/teams/collaborators

Am I missing something?

avisconti, 2024-10-23:
> @avisconti thanks! It's mainly to better support the PHYTEC boards, for which we recently added a separate Platform.
> 
> Yes, it's still grey.. not sure why that is. I found no pending notification or any other hint in my GitHub Account. In my review comments, it's even showing `collaborator` which should reflect the Zephyr Organisation setting. Let me know if you have any hint.

I don't see you among the collaborators (https://github.com/orgs/zephyrproject-rtos/teams/collaborators). Maybe it's just a matter of synchronization, I don't know

jonas-rem, 2024-10-23:
@avisconti when following your link, I now saw an invitation in my GitHub Account, accepted it and not it seems to work. Strange, I got an invitation some time back, which I also accepted. After that I haven't seen any other hint in the GitHub UI until now.

Anyways it works now :), thanks for checking!