zephyrproject-rtos/zephyr #79306: LwM2M client sample: Add support for `greater than`, `less than` and `step` observer attributes

alanthompsonch, 2024-10-02:
**Is your enhancement proposal related to a problem? Please describe.**

LwM2M defines the greater than (gt), less than (lt) and step (st) attributes for observers (Client SHOULD support these). These attributes allow for observers to report values in a more specific manner, only reporting if the conditions are met. However, the LwM2M client sample does not support this mechanism. The client sample can recognize and set the attributes, but it doesn't do anything with the information and ignores the attributes.

**Describe the solution you'd like**

I'd like for these observer attributes to be implemented in the LwM2M client sample / the LwM2M engine according to the LwM2M technical specification ([See Table 5.1.2.-2](https://www.openmobilealliance.org/release/LightweightM2M/V1_1_1-20190617-A/OMA-TS-LightweightM2M_Core-V1_1_1-20190617-A.pdf#%BBy%E6e%B6%5C%FEj%3A%A91%DF%F8c%CFw3%BD%23%5E))

**Describe alternatives you've considered**

It's possible to implement a similar logic on the application level, but you lose all the flexibility provided by the LwM2M Specification. A flexible implementation belongs in the LwM2M engine

**Additional context**

I opened an issue on the nordic forum with some details: [Forum Post](https://devzone.nordicsemi.com/f/nordic-q-a/114073/lwm2m-client-does-not-respect-observer-attributes)

rlubos, 2024-11-04:
@alanthompsonch Are you planning/willing to work on this?

rlubos, 2025-06-24:
@alanthompsonch In case you're still interested, this is being implemented in https://github.com/zephyrproject-rtos/zephyr/pull/92097

closed 2025-06-26 by danieldegrasse