zephyrproject-rtos/zephyr #79312: test: tests/lib/cpp/cxx/cpp.main.* BUILD FAILURE on ace20_sim/ace30_sim/ace15_sim

KamilxPaszkiet, 2024-10-02:
**Describe the bug**
Tests:

- tests/lib/cpp/cxx/cpp.main.minimal
- tests/lib/cpp/cxx/cpp.main.picolibc

BUILD FAILURE on ace20_sim/ace30_sim/ace15_sim

**To Reproduce**
`west twister -p intel_adsp/ace20_lnl/sim -s tests/lib/cpp/cxx/cpp.main.minimal`
or
`west build -p -b intel_adsp/ace20_lnl/sim tests/lib/cpp/cxx/ -T cpp.main.minimal`

**Logs and console output**
```
/__w/os.rtos.zephyr.zephyr/os.rtos.zephyr.zephyr/tests/lib/cpp/cxx/src/main.cpp:148:1: error: initialization of flexible array member is not allowed
DEVICE_DT_DEFINE(DT_NODELABEL(test_dev1_dfr), NULL, NULL, NULL, NULL, POST_KERNEL, 33, NULL);
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/__w/os.rtos.zephyr.zephyr/os.rtos.zephyr.zephyr/include/zephyr/device.h:198:2: note: expanded from macro 'DEVICE_DT_DEFINE'
        Z_DEVICE_DEFINE(node_id, Z_DEVICE_DT_DEV_ID(node_id),                  \
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/__w/os.rtos.zephyr.zephyr/os.rtos.zephyr.zephyr/include/zephyr/device.h:1163:11: note: expanded from macro 'Z_DEVICE_DEFINE'
                              (Z_DEVICE_DT_METADATA_DEFINE(node_id, dev_id);))))\
                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/__w/os.rtos.zephyr.zephyr/os.rtos.zephyr.zephyr/include/zephyr/device.h:967:17: note: expanded from macro 'Z_DEVICE_DT_METADATA_DEFINE'
                .nodelabels = DT_NODELABEL_STRING_ARRAY(node_id),       \
                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
note: (skipping 10 expansions in backtrace; use -fmacro-backtrace-limit=0 to see all)
/__w/os.rtos.zephyr.zephyr/os.rtos.zephyr.zephyr/include/zephyr/sys/util_internal.h:64:39: note: expanded from macro '__COND_CODE'
        __GET_ARG2_DEBRACKET(one_or_two_args _if_code, _else_code)
                                             ^~~~~~~~
/__w/os.rtos.zephyr.zephyr/os.rtos.zephyr.zephyr/include/zephyr/sys/util_internal.h:69:65: note: expanded from macro '__GET_ARG2_DEBRACKET'
#define __GET_ARG2_DEBRACKET(ignore_this, val, ...) __DEBRACKET val
                                                                ^~~
/__w/os.rtos.zephyr.zephyr/os.rtos.zephyr.zephyr/include/zephyr/sys/util_internal.h:72:26: note: expanded from macro '__DEBRACKET'
#define __DEBRACKET(...) __VA_ARGS__
                         ^~~~~~~~~~~
/__w/os.rtos.zephyr.zephyr/os.rtos.zephyr.zephyr/include/zephyr/device.h:885:14: note: initialized flexible array member 'nodelabels' is here
        const char *nodelabels[];
                    ^
/__w/os.rtos.zephyr.zephyr/os.rtos.zephyr.zephyr/tests/lib/cpp/cxx/src/main.cpp:154:1: error: initialization of flexible array member is not allowed
DEVICE_DT_DEFINE(DT_NODELABEL(test_dev0_boot), NULL,
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/__w/os.rtos.zephyr.zephyr/os.rtos.zephyr.zephyr/include/zephyr/device.h:198:2: note: expanded from macro 'DEVICE_DT_DEFINE'
        Z_DEVICE_DEFINE(node_id, Z_DEVICE_DT_DEV_ID(node_id),                  \
        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/__w/os.rtos.zephyr.zephyr/os.rtos.zephyr.zephyr/include/zephyr/device.h:1163:11: note: expanded from macro 'Z_DEVICE_DEFINE'
                              (Z_DEVICE_DT_METADATA_DEFINE(node_id, dev_id);))))\
                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/__w/os.rtos.zephyr.zephyr/os.rtos.zephyr.zephyr/include/zephyr/device.h:967:17: note: expanded from macro 'Z_DEVICE_DT_METADATA_DEFINE'
                .nodelabels = DT_NODELABEL_STRING_ARRAY(node_id),       \
                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
note: (skipping 10 expansions in backtrace; use -fmacro-backtrace-limit=0 to see all)
/__w/os.rtos.zephyr.zephyr/os.rtos.zephyr.zephyr/include/zephyr/sys/util_internal.h:64:39: note: expanded from macro '__COND_CODE'
        __GET_ARG2_DEBRACKET(one_or_two_args _if_code, _else_code)
                                             ^~~~~~~~
/__w/os.rtos.zephyr.zephyr/os.rtos.zephyr.zephyr/include/zephyr/sys/util_internal.h:69:65: note: expanded from macro '__GET_ARG2_DEBRACKET'
#define __GET_ARG2_DEBRACKET(ignore_this, val, ...) __DEBRACKET val
                                                                ^~~
/__w/os.rtos.zephyr.zephyr/os.rtos.zephyr.zephyr/include/zephyr/sys/util_internal.h:72:26: note: expanded from macro '__DEBRACKET'
#define __DEBRACKET(...) __VA_ARGS__
                         ^~~~~~~~~~~
/__w/os.rtos.zephyr.zephyr/os.rtos.zephyr.zephyr/include/zephyr/device.h:885:14: note: initialized flexible array member 'nodelabels' is here
        const char *nodelabels[];
                    ^
2 errors generated.
ninja: build stopped: subcommand failed.

```
**Expected behavior**

Build without warnings

**Environment:**

 - Linux Ubuntu 22.04
 - Toolchain: Zephyr SDK 16.8
 - Zephyr version: v3.7.0-3643-gf810e1318f38

dcpleung, 2024-10-02:
I have no issue building `tests/lib/cpp/cxx` via twister for both ace15 and ace20. Zephyr SDK does not have toolchain for ace30 currently so it can't be built yet. I have tried both latest main and SHA f810e1318f38 but they both built fine without any issues. There may be something wrong with your environment.

KamilxPaszkiet, 2024-10-03:
> I have no issue building `tests/lib/cpp/cxx` via twister for both ace15 and ace20. Zephyr SDK does not have toolchain for ace30 currently so it can't be built yet. I have tried both latest main and SHA [f810e13](https://github.com/zephyrproject-rtos/zephyr/commit/f810e1318f385ade00b12a5bdddc4324c177f02a) but they both built fine without any issues. There may be something wrong with your environment.

I mean ace30/20/15 sim, not hardware ;)

dcpleung, 2024-10-03:
Yes, I tried it locally building for *_sim and they were no errors.

dcpleung, 2024-10-03:
```
INFO    - Using Ninja..
INFO    - Zephyr version: v3.7.0-3880-g73a3438b8204
INFO    - Using 'zephyr' toolchain.
INFO    - Building initial testsuite list...
INFO    - Writing JSON report /tmp/twister/testplan.json
INFO    - JOBS: 32
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    -  5/30 intel_adsp/ace15_mtpm/sim tests/lib/cpp/cxx/cpp.main.newlib_nano             SKIPPED (runtime filter)
INFO    -  6/30 intel_adsp/ace20_lnl/sim  tests/lib/cpp/cxx/cpp.main.newlib_nano             SKIPPED (runtime filter)
INFO    -  7/30 intel_adsp/ace20_lnl/sim  tests/lib/cpp/libcxx/cpp.libcxx.glibcxx.newlib_nano PASSED (build)
INFO    -  8/30 intel_adsp/ace15_mtpm/sim tests/lib/cpp/libcxx/cpp.libcxx.glibcxx.picolibc   PASSED (build)
INFO    -  9/30 intel_adsp/ace20_lnl/sim  tests/lib/cpp/cxx/cpp.main.cpp2B                   PASSED (build)
INFO    - 10/30 intel_adsp/ace15_mtpm/sim tests/lib/cpp/cxx/cpp.main.cpp2B                   PASSED (build)
INFO    - 11/30 intel_adsp/ace20_lnl/sim  tests/lib/cpp/libcxx/cpp.libcxx.glibcxx.picolibc   PASSED (build)
INFO    - 12/30 intel_adsp/ace15_mtpm/sim tests/lib/cpp/cxx/cpp.main.cpp17                   PASSED (build)
INFO    - 13/30 intel_adsp/ace15_mtpm/sim tests/lib/cpp/cxx/cpp.main.cpp2A                   PASSED (build)
INFO    - 14/30 intel_adsp/ace20_lnl/sim  tests/lib/cpp/cxx/cpp.main.cpp2A                   PASSED (build)
INFO    - 15/30 intel_adsp/ace15_mtpm/sim tests/lib/cpp/cxx/cpp.main.cpp20                   PASSED (build)
INFO    - 16/30 intel_adsp/ace20_lnl/sim  tests/lib/cpp/libcxx/cpp.libcxx.glibcxx.newlib     PASSED (build)
INFO    - 17/30 intel_adsp/ace20_lnl/sim  tests/lib/cpp/cxx/cpp.main.cpp14                   PASSED (build)
INFO    - 18/30 intel_adsp/ace15_mtpm/sim tests/lib/cpp/cxx/cpp.main.minimal                 PASSED (build)
INFO    - 19/30 intel_adsp/ace15_mtpm/sim tests/lib/cpp/cxx/cpp.main.newlib                  PASSED (build)
INFO    - 20/30 intel_adsp/ace15_mtpm/sim tests/lib/cpp/libcxx/cpp.libcxx.glibcxx.newlib     PASSED (build)
INFO    - 21/30 intel_adsp/ace15_mtpm/sim tests/lib/cpp/libcxx/cpp.libcxx.glibcxx.newlib_nano PASSED (build)
INFO    - 22/30 intel_adsp/ace20_lnl/sim  tests/lib/cpp/cxx/cpp.main.cpp20                   PASSED (build)
INFO    - 23/30 intel_adsp/ace20_lnl/sim  tests/lib/cpp/cxx/cpp.main.picolibc                PASSED (build)
INFO    - 24/30 intel_adsp/ace20_lnl/sim  tests/lib/cpp/cxx/cpp.main.minimal                 PASSED (build)
INFO    - 25/30 intel_adsp/ace15_mtpm/sim tests/lib/cpp/cxx/cpp.main.cpp14                   PASSED (build)
INFO    - 26/30 intel_adsp/ace15_mtpm/sim tests/lib/cpp/cxx/cpp.main.cpp98                   PASSED (build)
INFO    - 27/30 intel_adsp/ace20_lnl/sim  tests/lib/cpp/cxx/cpp.main.newlib                  PASSED (build)
INFO    - 28/30 intel_adsp/ace20_lnl/sim  tests/lib/cpp/cxx/cpp.main.cpp17                   PASSED (build)
INFO    - 29/30 intel_adsp/ace20_lnl/sim  tests/lib/cpp/cxx/cpp.main.cpp98                   PASSED (build)
INFO    - 30/30 intel_adsp/ace15_mtpm/sim tests/lib/cpp/cxx/cpp.main.picolibc                PASSED (build)

INFO    - 15 test scenarios (30 test instances) selected, 6 configurations skipped (4 by static filter, 2 at runtime).
INFO    - 24 of 30 test configurations passed (100.00%), 0 failed, 0 errored, 6 skipped with 0 warnings in 30.34 seconds
INFO    - In total 84 test cases were executed, 26 skipped on 2 out of total 8 platforms (25.00%)
INFO    - 0 test configurations executed on platforms, 24 test configurations were only built.
```

KamilxPaszkiet, 2024-10-09:
No repro for now
Thx @dcpleung

closed 2024-10-09 by KamilxPaszkiet