zephyrproject-rtos/zephyr #79326: Build warnings on subsys/net/lib/http/http_client.c

FaBjE, 2024-10-02:
**Describe the bug**
Depending on compile optimizations settings, subsys/net/lib/http/http_client.c builds with warnings

**Expected behavior**
No build warnings

**Impact**
Either annoyance for users, or broken CI when this is triggered in CI (it is triggered downstream with some samples).

**Logs and console output**
```
[559/734] Building C object zephyr/subsys/net/lib/http/CMakeFiles/subsys__net__lib__http.dir/http_client.c.obj
/Zephyr/subsys/net/lib/http/http_client.c: In function 'http_send_data.constprop':
/Zephyr/subsys/net/lib/http/http_client.c:112:33: warning: 'strncpy' specified bound depends on the length of the source argument [-Wstringop-truncation]
  112 |                                 strncpy(send_buf + end_of_send,
      |                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  113 |                                         data + end_of_data,
      |                                         ~~~~~~~~~~~~~~~~~~~
  114 |                                         remaining_len);
      |                                         ~~~~~~~~~~~~~~
/Zephyr/subsys/net/lib/http/http_client.c:87:41: note: length computed here
   87 |                         remaining_len = strlen(data + end_of_data);
      |                                         ^~~~~~~~~~~~~~~~~~~~~~~~~~
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: docker.io/zephyrprojectrtos/zephyr-build:v0.26.11
 - Zephyr Version: v3.7.0-17-gf84f55dfb0d3

jukkar, 2024-10-03:
@FaBjE I just tried the http-client sample but cannot see this error. Is there a way to reproduce it, can you give more information?

jukkar, 2024-10-04:
Compilation with west that reproduces the issue:
```
west build -p -b stm32h747i_disco/stm32h747xx/m7 \
    -d ../build/http-client samples/net/sockets/http_client/ -- -DCONFIG_SPEED_OPTIMIZATIONS=y
```
The above is using GCC 12.2.0.
Tried with native_sim but it is silent about the warning. There the GCC is 11.4.0.
The SDK version is 0.16.8 in both cases.
Also one needs to set `CONFIG_SPEED_OPTIMIZATIONS=y`, if that is left out there is no warning.

PR #79422 (2024-10-04): net: lib: http-client: Use memcpy() to avoid gcc warning
GCC 12.2.0 can give this warning (version 11.4.0 did not), when `CONFIG_SPEED_OPTIMIZATIONS=y`
```
subsys/net/lib/http/http_client.c: In function 'http_send_data.constprop': subsys/net/lib/http/http_client.c:114:33: warning: 'strncpy' specified \
  bound depends on the length of the source argument [-Wstringop-truncation]
  114 |                                 strncpy(send_buf + end_of_send,
      |                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  115 |                                         data + end_of_data,
      | [... cut]
closed 2024-10-04 by mmahadevan108