zephyrproject-rtos/zephyr #79327: lpcxpresso55s69/lpc55s69/cpu0/ns build fails

ozersa, 2024-10-02:
**Describe the bug**
Build failing
[image]

**To Reproduce**
1- Download final version of zephyr
2- run west update to update all component
3- try to build hello world example with ns mode

`west build -b lpcxpresso55s69/lpc55s69/cpu0/ns  .\samples\hello_world\ -p `

**Expected behavior**
Build shall success, zephyr + tfm and MCUBoot shall be build

**Impact**

**Logs and console output**
```
PS C:\Users\sozer\zephyrproject\zephyr> west build -b lpcxpresso55s69/lpc55s69/cpu0/ns  .\samples\hello_world\ -p 
-- west build: making build dir C:\Users\sozer\zephyrproject\zephyr\build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: C:/Users/sozer/zephyrproject/zephyr/samples/hello_world
-- CMake version: 3.26.4
-- Found Python3: C:/Python/Python312/python.exe (found suitable version "3.12.7", minimum required is "3.10") found components: Interpreter 
-- Cache files will be written to: C:/Users/sozer/zephyrproject/zephyr/.cache
-- Zephyr version: 3.7.99 (C:/Users/sozer/zephyrproject/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: lpcxpresso55s69, qualifiers: lpc55s69/cpu0/ns
-- Found host-tools: zephyr 0.16.1 (C:/Users/sozer/zephyr-sdk-0.16.1)
-- Found toolchain: zephyr 0.16.1 (C:/Users/sozer/zephyr-sdk-0.16.1)
-- Found Dtc: C:/ProgramData/chocolatey/bin/dtc.exe (found suitable version "1.5.0", minimum required is "1.4.6") 
-- Found BOARD.dts: C:/Users/sozer/zephyrproject/zephyr/boards/nxp/lpcxpresso55s69/lpcxpresso55s69_lpc55s69_cpu0_ns.dts
-- Generated zephyr.dts: C:/Users/sozer/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated pickled edt: C:/Users/sozer/zephyrproject/zephyr/build/zephyr/edt.pickle
-- Generated zephyr.dts: C:/Users/sozer/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: C:/Users/sozer/zephyrproject/zephyr/build/zephyr/include/generated/zephyr/devicetree_generated.h
-- Including generated dts.cmake file: C:/Users/sozer/zephyrproject/zephyr/build/zephyr/dts.cmake
Parsing C:/Users/sozer/zephyrproject/zephyr/Kconfig
Loaded configuration 'C:/Users/sozer/zephyrproject/zephyr/boards/nxp/lpcxpresso55s69/lpcxpresso55s69_lpc55s69_cpu0_ns_defconfig'
Merged configuration 'C:/Users/sozer/zephyrproject/zephyr/samples/hello_world/prj.conf'
Configuration saved to 'C:/Users/sozer/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to 'C:/Users/sozer/zephyrproject/zephyr/build/zephyr/include/generated/zephyr/autoconf.h'
-- Found GnuLd: c:/users/sozer/zephyr-sdk-0.16.1/arm-zephyr-eabi/arm-zephyr-eabi/bin/ld.bfd.exe (found version "2.38") 
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/sozer/zephyr-sdk-0.16.1/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc.exe
Load components for LPC55S69_cm33_core0:
driver_common component is included.
driver_reset component is included.
device_CMSIS component is included.
CMSIS_Include_core_cm component is included.
device_system component is included.
driver_lpc_gpio component is included.
driver_pint component is included.
driver_inputmux component is included.
driver_inputmux_connections component is included.
driver_flexcomm component is included.
driver_flexcomm_usart component is included.
driver_lpc_iocon component is included.
-- Setting build type to 'MinSizeRel' as none was specified.
CMake Warning at C:/Users/sozer/zephyrproject/zephyr/modules/trusted-firmware-m/CMakeLists.txt:527 (message):
  TFM_DUMMY_PROVISIONING is enabled:

        The device will be provisioned using dummy keys and is NOT secure!
        This is not suitable for production

-- Configuring done (30.7s)
-- Generating done (0.4s)
-- Build files have been written to: C:/Users/sozer/zephyrproject/zephyr/build
←[92m-- west build: building application
[1/246] Generating include/generated/zephyr/version.h
-- Zephyr version: 3.7.99 (C:/Users/sozer/zephyrproject/zephyr), build: v3.7.0-3804-gf98fde07b304
[2/246] Generating ../../tfm/CMakeCache.txt
FAILED: tfm/CMakeCache.txt C:/Users/sozer/zephyrproject/zephyr/build/tfm/CMakeCache.txt
cmd.exe /C "cd /D C:\Users\sozer\zephyrproject\zephyr\build\tfm && "C:\Program Files\CMake\bin\cmake.exe" -GNinja -DTFM_TOOLCHAIN_FILE=C:/Users/sozer/zephyrproject/modules/tee/tf-m/trusted-firmware-m/toolchain_GNUARM.cmake -DCROSS_COMPILE=C:/Users/sozer/zephyr-sdk-0.16.1/arm-zephyr-eabi/bin/arm-zephyr-eabi -DCMAKE_BUILD_TYPE=MinSizeRel -DTFM_PLATFORM=nxp/lpcxpresso55s69 -DCONFIG_TFM_BUILD_LOG_QUIET=ON -DCONFIG_TFM_MEMORY_USAGE_QUIET=OFF -DPython3_EXECUTABLE=C:/Python/Python312/python.exe -DCONFIG_TFM_SPM_BACKEND="IPC" -DBL2=TRUE -DMCUBOOT_IMAGE_VERSION_S=0.0.0+0 -DMCUBOOT_IMAGE_VERSION_NS=0.0.0+0 -DMCUBOOT_SIGNATURE_TYPE=EC-P256 -DMCUBOOT_USE_PSA_CRYPTO=ON -DMCUBOOT_BUILTIN_KEY=ON -DMCUBOOT_HW_KEY=OFF -DMCUBOOT_KEY_S=C:/Users/sozer/zephyrproject/modules/tee/tf-m/trusted-firmware-m/bl2/ext/mcuboot/root-EC-P256.pem -DMCUBOOT_KEY_NS=C:/Users/sozer/zephyrproject/modules/tee/tf-m/trusted-firmware-m/bl2/ext/mcuboot/root-EC-P256_1.pem -DTFM_ISOLATION_LEVEL=2 -DTFM_PROFILE=profile_medium -DMCUBOOT_IMAGE_NUMBER=2 -DTFM_DUMMY_PROVISIONING=ON -DTFM_EXCEPTION_INFO_DUMP=ON -DMCUBOOT_LOG_LEVEL=INFO -DTFM_PARTITION_LOG_LEVEL=TFM_PARTITION_LOG_LEVEL_INFO -DTFM_SPM_LOG_LEVEL=TFM_SPM_LOG_LEVEL_INFO -DTFM_PARTITION_NS_AGENT_MAILBOX=OFF -DTFM_PARTITION_PROTECTED_STORAGE=ON -DTFM_PARTITION_INTERNAL_TRUSTED_STORAGE=ON -DTFM_PARTITION_CRYPTO=ON -DTFM_PARTITION_PLATFORM=ON -DTFM_PARTITION_INITIAL_ATTESTATION=OFF -DTFM_PARTITION_FIRMWARE_UPDATE=OFF -DTFM_PLATFORM_NXP_HAL_FILE_PATH=C:/Users/sozer/zephyrproject/modules/tee/tf-m/trusted-firmware-m/platform/ext/target/nxp/ -DMCUBOOT_PATH=C:/Users/sozer/zephyrproject/bootloader/mcuboot -DCONFIG_TFM_ENABLE_FP=OFF -DTFM_TESTS_REVISION_CHECKS=OFF -DMBEDCRYPTO_PATH=C:/Users/sozer/zephyrproject/modules/crypto/mbedtls -DCMSIS_PATH=C:/Users/sozer/zephyrproject/modules/tee/tf-m/trusted-firmware-m/platform/ext/cmsis C:/Users/sozer/zephyrproject/modules/tee/tf-m/trusted-firmware-m"
-- Could NOT find Threads (missing: Threads_FOUND)
-- Could NOT find Threads (missing: Threads_FOUND)
CMake Warning (dev) at platform/ext/target/nxp/lpcxpresso55s69/CMakeLists.txt:17:
  Syntax Warning in cmake code at column 58

  Argument not separated from preceding token by whitespace.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Error at cmake/utils.cmake:67 (message):
  INVALID CONFIG: BL2 AND ( NOT MCUBOOT_UPGRADE_STRATEGY = DIRECT_XIP AND
  MCUBOOT_DIRECT_XIP_REVERT )
Call Stack (most recent call first):
  config/check_config.cmake:28 (tfm_invalid_config)
  CMakeLists.txt:70 (include)

-- Configuring incomplete, errors occurred!
[3/246] Generating misc/generated/syscalls.json, misc/generated/struct_tags.json
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' --build 'C:\Users\sozer\zephyrproject\zephyr\build'
PS C:\Users\sozer\zephyrproject\zephyr> 
```

**Environment (please complete the following information):**

 - OS: Windows
 - Toolchain: Zephyr SDK
- Zephyr SHA: https://github.com/zephyrproject-rtos/zephyr/commit/f98fde07b304fadf98d62659de3a4bbf0dd99149

**Additional context**

ozersa, 2024-10-02:
@MaureenHelm fyi.

mmahadevan108, 2024-10-03:
@butok, can you help take a look at this issue.

butok, 2024-10-07:
(10/06/2034) Commit SHA-1: d022d315d2e006ad1eb13f3f72a00a99b4ff7fb0

The issue was not been reproduced. 
The build has passed.

```
>west build -p always -b lpcxpresso55s69/lpc55s69/cpu0/ns samples/hello_world

-- west build: making build dir C:\_ddm\ZEPHYR\zephyr_vs\zephyr\build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: C:/_ddm/ZEPHYR/zephyr_vs/zephyr/samples/hello_world
-- CMake version: 3.28.1
-- Found Python3: C:/Python311/python.exe (found suitable version "3.11.4", minimum required is "3.10") found components: Interpreter
-- Cache files will be written to: C:/_ddm/ZEPHYR/zephyr_vs/zephyr/.cache
-- Zephyr version: 3.7.99 (C:/_ddm/ZEPHYR/zephyr_vs/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: lpcxpresso55s69, qualifiers: lpc55s69/cpu0/ns
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.8 (C:/Users/nxa12205/zephyr-sdk-0.16.8)
-- Found toolchain: zephyr 0.16.8 (C:/Users/nxa12205/zephyr-sdk-0.16.8)
-- Found Dtc: C:/Users/nxa12205/.mcuxpressotools/dtc-msys2/tools/usr/bin/dtc.exe (found suitable version "1.6.1", minimum required is "1.4.6")
-- Found BOARD.dts: C:/_ddm/ZEPHYR/zephyr_vs/zephyr/boards/nxp/lpcxpresso55s69/lpcxpresso55s69_lpc55s69_cpu0_ns.dts
-- Generated zephyr.dts: C:/_ddm/ZEPHYR/zephyr_vs/zephyr/build/zephyr/zephyr.dts
-- Generated pickled edt: C:/_ddm/ZEPHYR/zephyr_vs/zephyr/build/zephyr/edt.pickle
-- Generated zephyr.dts: C:/_ddm/ZEPHYR/zephyr_vs/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: C:/_ddm/ZEPHYR/zephyr_vs/zephyr/build/zephyr/include/generated/zephyr/devicetree_generated.h
-- Including generated dts.cmake file: C:/_ddm/ZEPHYR/zephyr_vs/zephyr/build/zephyr/dts.cmake
Parsing C:/_ddm/ZEPHYR/zephyr_vs/zephyr/Kconfig
Loaded configuration 'C:/_ddm/ZEPHYR/zephyr_vs/zephyr/boards/nxp/lpcxpresso55s69/lpcxpresso55s69_lpc55s69_cpu0_ns_defconfig'
Merged configuration 'C:/_ddm/ZEPHYR/zephyr_vs/zephyr/samples/hello_world/prj.conf'
Configuration saved to 'C:/_ddm/ZEPHYR/zephyr_vs/zephyr/build/zephyr/.config'
Kconfig header saved to 'C:/_ddm/ZEPHYR/zephyr_vs/zephyr/build/zephyr/include/generated/zephyr/autoconf.h'
-- Found GnuLd: c:/users/nxa12205/zephyr-sdk-0.16.8/arm-zephyr-eabi/arm-zephyr-eabi/bin/ld.bfd.exe (found version "2.38")
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/nxa12205/zephyr-sdk-0.16.8/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc.exe
Load components for LPC55S69_cm33_core0:
driver_common component is included.
driver_reset component is included.
device_CMSIS component is included.
CMSIS_Include_core_cm component is included.
device_system component is included.
driver_lpc_gpio component is included.
driver_pint component is included.
driver_inputmux component is included.
driver_inputmux_connections component is included.
driver_flexcomm component is included.
driver_flexcomm_usart component is included.
driver_lpc_iocon component is included.
-- Setting build type to 'MinSizeRel' as none was specified.
-- Using ccache: C:/Strawberry/c/bin/ccache.exe
CMake Warning at C:/_ddm/ZEPHYR/zephyr_vs/zephyr/modules/trusted-firmware-m/CMakeLists.txt:527 (message):
  TFM_DUMMY_PROVISIONING is enabled:

        The device will be provisioned using dummy keys and is NOT secure!
        This is not suitable for production

-- Configuring done (30.8s)
-- Generating done (0.9s)
-- Build files have been written to: C:/_ddm/ZEPHYR/zephyr_vs/zephyr/build
-- west build: building application
[1/246] Generating include/generated/zephyr/version.h
-- Zephyr version: 3.7.99 (C:/_ddm/ZEPHYR/zephyr_vs/zephyr), build: v3.7.0-3955-gd022d315d2e0
[6/246] Generating ../../tfm/CMakeCache.txt
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.41.0.windows.1")
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
[... 395 lines cut ...]
-- Installing: C:/_ddm/ZEPHYR/zephyr_vs/zephyr/build/tfm/api_ns/platform/include/tfm_plat_defs.h
-- Installing: C:/_ddm/ZEPHYR/zephyr_vs/zephyr/build/tfm/api_ns/platform/include/tfm_plat_device_id.h
-- Installing: C:/_ddm/ZEPHYR/zephyr_vs/zephyr/build/tfm/api_ns/platform/include/tfm_plat_ns.h
-- Installing: C:/_ddm/ZEPHYR/zephyr_vs/zephyr/build/tfm/api_ns/platform/include/tfm_plat_nv_counters.h
-- Installing: C:/_ddm/ZEPHYR/zephyr_vs/zephyr/build/tfm/api_ns/platform/include/tfm_plat_otp.h
-- Installing: C:/_ddm/ZEPHYR/zephyr_vs/zephyr/build/tfm/api_ns/platform/include/tfm_plat_provisioning.h
-- Installing: C:/_ddm/ZEPHYR/zephyr_vs/zephyr/build/tfm/api_ns/platform/include/tfm_plat_rotpk.h
-- Installing: C:/_ddm/ZEPHYR/zephyr_vs/zephyr/build/tfm/api_ns/platform/include/tfm_plat_test.h
-- Installing: C:/_ddm/ZEPHYR/zephyr_vs/zephyr/build/tfm/api_ns/CMakeLists.txt
-- Installing: C:/_ddm/ZEPHYR/zephyr_vs/zephyr/build/tfm/api_ns/cmake/toolchain_ns_GNUARM.cmake
-- Installing: C:/_ddm/ZEPHYR/zephyr_vs/zephyr/build/tfm/api_ns/cmake/toolchain_ns_ARMCLANG.cmake
-- Installing: C:/_ddm/ZEPHYR/zephyr_vs/zephyr/build/tfm/api_ns/cmake/toolchain_ns_IARARM.cmake
-- Installing: C:/_ddm/ZEPHYR/zephyr_vs/zephyr/build/tfm/api_ns/platform/include/fih.h
-- Installing: C:/_ddm/ZEPHYR/zephyr_vs/zephyr/build/tfm/api_ns/cmake/spe_export.cmake
[... cut ...] ns.cmake
[246/246] Linking C executable zephyr\zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       15194 B        96 KB     15.46%
             RAM:        4200 B       136 KB      3.02%
        USB_SRAM:          0 GB        16 KB      0.00%
        IDT_LIST:          0 GB        32 KB      0.00%
Generating files from C:/_ddm/ZEPHYR/zephyr_vs/zephyr/build/zephyr/zephyr.elf for board: lpcxpresso55s69
image.py: sign the payload
image.py: sign the payload
```

closed 2024-10-07 by butok

ozersa, 2024-10-07:
Yes the build works with final zephyr point that you mention,
It has been fixed by somehow after the SHA I mentioned.
https://github.com/zephyrproject-rtos/zephyr/compare/f98fde07b304fadf98d62659de3a4bbf0dd99149...d022d315d2e006ad1eb13f3f72a00a99b4ff7fb0
 Thanks.