zephyrproject-rtos/zephyr #79361: EDF updates don't work when made from ISR context

alexpaschoaletto, 2024-10-03:
**Describe the bug**

The EDF scheduler is not working as expected when the deadlines are updated within ISRs. The deadline values get correctly updated, but the threads don't execute in the expected order. I have first noticed something was going on on #79124, but after a bit of digging through the kernel files I'm convinced it's an actual bug.

**To Reproduce**

Run this code on your desired environment, preferably one with 64-bit timers (e.g. RISC-V targets). It might look lengthy, but it's actually quite simple. What it does:

- creates two identical threads, each with a private message queue;
- creates a master timer which periodically sends a message to each thread (.e.g.: every two seconds). The threads are unblocked when they receive a message, execute their cycle and get back to waiting for new messages to repeat it all.

Basically the timer ensures the threads are periodic, activating and sending the messages at every specified period. On every activation a new absolute deadline is generated for each thread. All extra context can be found within the comments.

Kconfig options:
```c
CONFIG_SYS_CLOCK_TICKS_PER_SEC=10000
CONFIG_SCHED_DEADLINE=y
CONFIG_LOG=y       /* makes printk be handled asynchronously by a background thread (reducing the overhead) */

# QEMU only
CONFIG_QEMU_ICOUNT=n
CONFIG_STDOUT_CONSOLE=y
```

source code:
```c
#include <zephyr/kernel.h>
/*
	Remember that for "pure" EDF results, all
	user threads must have the same static priority.
	That happens because EDF will be a tie-breaker
	among two or more ready tasks of the same static
	priority. An arbitrary positive number is chosen here.
*/

#define EDF_PRIORITY 5
#define INACTIVE -1
#define MSEC_TO_CYC(msec)       ((msec * CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC) / MSEC_PER_SEC)
#define MSEC_TO_USEC(msec)      (msec * 1000)

typedef struct {
    int id;
    int32_t rel_deadline_msec;
    int32_t period_msec;
    uint32_t wcet_msec;
    k_tid_t thread;
    struct k_msgq queue;
} edf_t;

// edf-related structs with the properties for each thread ////////////////////////////////

edf_t task1 = {
	.id = 1,
	.rel_deadline_msec = 10,
	.wcet_msec = 1
};

edf_t task2 = {
	.id = 2,
	.rel_deadline_msec = 50,
	.wcet_msec = 5
};

// timer-related functions ///////////////////////////////////////////////////////////////////

void grow(edf_t *task){
	task->wcet_msec += 1;
	task->rel_deadline_msec += 10;
}

void trigger(edf_t *task){
	const char t = 1;
	k_msgq_put(&task->queue, &t, K_NO_WAIT);
	int deadline = (int) MSEC_TO_CYC(task->rel_deadline_msec);
	k_thread_deadline_set(task->thread, deadline);
}

void trigger_all(struct k_timer *timer){
	printk("\n");
	trigger(&task1);
	trigger(&task2);
	grow(&task1);   //task 1 progressively gains +10ms of relative deadline
}

// thread function /////////////////////////////////////////////////////////////////////////

void thread_function(void *task_props, void *a2, void *a3){
	char buffer[10];
	char message;
	uint32_t deadline = 0;
	uint64_t start = 0;
	uint64_t end = 0;

	edf_t *task = (edf_t *) task_props;
	task->thread = k_current_get();
	k_msgq_init(&task->queue, buffer, sizeof(char), 10);
	printk("[ %d ] ready.\n", task->id);

	for(;;){
		/*
			The periodic thread has no period, ironically.
			What makes it periodic is the master timer
			associated with it that regularly sends new
			messages to the thread queue (e.g. every two
			seconds).
		*/
		k_msgq_get(&task->queue, &message, K_FOREVER);
		start = k_cycle_get_64();
		deadline = task->thread->base.prio_deadline;
		k_busy_wait(MSEC_TO_USEC(task->wcet_msec));
		end = k_cycle_get_64();
		printk("[ %d ]\tstart %llu\t end %llu\t dead %u\n", task->id, start, end, deadline);
	}
}

// defines and main //////////////////////////////////////////////////////////////////////

K_THREAD_DEFINE(
	task1_thread, 2048,
	thread_function,
	&task1, NULL, NULL,
	EDF_PRIORITY, 0, INACTIVE
);

K_THREAD_DEFINE(
	task2_thread, 2048,
	thread_function,
	&task2, NULL, NULL,
	EDF_PRIORITY, 0, INACTIVE
);

K_TIMER_DEFINE(master_timer, trigger_all, NULL);

int main(void){
         k_sleep(K_SECONDS(1));
        printk("\nexecuting on: %s\n\n", CONFIG_BOARD_TARGET);

	k_thread_start(task1_thread);
	k_thread_start(task2_thread);
	k_sleep(K_SECONDS(1));
	
	k_timer_start(&master_timer, K_SECONDS(1), K_SECONDS(2));
	return 0;
}
```

Congratulations, now your EDF should be NOT working! which is why I started this bug report in the first place.

**Expected behavior**
Thread 1 starts with a relative deadline of 10ms and thread 2 of 50ms, so when both are fired at the same time thread 1 should take the lead and execute first. However, since thread 1 is subject to the `grow()` function inside `trigger_all()`, it will gain +10ms of relative deadline at every trigger. So the expected outcome is that thread 2 eventually takes the lead:
```c
[ 1 ]   start 30149527   end 30171369    dead 30237806       // <-- thread 1 has the earliest deadline
[ 2 ]   start 30175266   end 30225378    dead 30645602

// a few cycles later...

[ 2 ]   start 130116686  end 130166893   dead 130613698
[ 1 ]   start 130167576  end 130237728   dead 130713353    // <-- no longer the earliest deadline
```

**Observed behavior**
Thread 2 remains the first one to run at ALL times, regardless of the deadline values. That happens, apparently, because thread 2 is the last inside ``trigger_all()` to be, well, triggered - so the scheduler basically behaves like a LIFO. If we trigger thread 2 and then 1, the execution order would be 1-2 forever.
```c
[ 2 ]   start 30112071   end 30162828    dead 30611068
[ 1 ]   start 30164161   end 30184363    dead 30209562     // <- but this is the earliest deadline... :(

[ 2 ]   start 50121647   end 50171696    dead 50621379
[ 1 ]   start 50171874   end 50201914    dead 50321308    // <- here aswell
```

**Impact**
This bug prevents periodic EDF threads from being timely updated, so it limits the usage of this algorithm in the application. For me, specifically, it is a showstopper for the feature #78000 that I've opened an RFC a few weeks ago, since it depends on updating its own thread's deadline inside a timer callback to work accordingly. 

**Environment (please complete the following information):**

 - OS: Windows 11 23H2
 - Zephyr 3.7.99, SDK 0.16.8
 - Commit SHA: main is at caf86128b75

alexpaschoaletto, 2024-10-03:
I have already done some testings and might have found the problem: the kernel function responsible for deciding which thread to execute upon the completion of an interrupt is picking from the ready queue cache _instead_ of looking for which has the highest priority (see line [986](https://github.com/zephyrproject-rtos/zephyr/blob/c710f8892beef3e486dded2f0ad4394e4a4d9aa0/kernel/sched.c#L986) of sched.c):

```c
void *z_get_next_switch_handle(void *interrupted)
{
	z_check_stack_sentinel();	
  
  #ifdef CONFIG_SMP
       /* not enabled, so we don't care for this #ifdef */
       ...
  #else
	z_sched_usage_switch(_kernel.ready_q.cache);
	_current->switch_handle = interrupted;
	set_current(_kernel.ready_q.cache);                <--- here!
	return _current->switch_handle;
  #endif /* CONFIG_SMP */
}
```

If one decides to compare the cached value with what `next_up` (the kernel function that choses the highest priority thread) gives by the moment `set_current` is called, will see there is a difference. `next_up` always returns the correct value, but the cache reflects what's been triggered the last in the timer callback.

 ### Possible solution
 placing this line right at the top of `z_get_next_switch_handle`:
 ```c
 void *z_get_next_switch_handle(void *interrupted)
{
	z_check_stack_sentinel();
	update_cache(1);                        <-- here!
	...

 ```
It updates the cache to the right thread before invoking `set_current`. After I've done this, the EDF worked as expected. However, I can't say this is the de facto solution - perhaps it would be best to assure the cache is updated correctly elsewhere in the code. Alfter all, we know for certain the cache gets updated mid-ISR when the threads are triggered, since they execute in LIFO fashion. The proposed solution also does not account for SMP cases, so it might be incomplete. But there you go.

alexpaschoaletto, 2024-10-13:
Ok, so I went through a deeper analysis on the error and have some new findings to share. I'm pretty sure I know _where_ the problem is happening, but still haven't figured out _why_.

### 1. Changing the order of execution WORKS
forget the possible solution I've listed above and just change the order of events of the `trigger` function:
```c
void trigger(edf_t *task){            <-- original function, NOT working
	...
	k_msgq_put(&task->queue, &t, K_NO_WAIT);
	k_thread_deadline_set(task->thread, deadline);
}

void trigger(edf_t *task){            <-- modified function, WORKING
	...
	k_thread_deadline_set(task->thread, deadline);
	k_msgq_put(&task->queue, &t, K_NO_WAIT);
}
```
You'll see the EDF now works. What happens underneath to justify that? In the `k_msgq_put` function... nothing really. As of my investigation, it runs through the exact same steps on both cases. But on the NON-working version, `k_thread_deadline_set` is called AFTER the thread has been unpended by the message queue. That means the thread is ALREADY in the runqueue by then. As a consequence, on the NON-working version `k_thread_deadline_set` performs an extra queue-dequeue routine to make sure the runqueue is correctly ordered:
 ```c
void z_impl_k_thread_deadline_set(k_tid_t tid, int deadline){
    ...
	K_SPINLOCK(&_sched_spinlock) {
		if (z_is_thread_queued(thread)) {                        <-- NON-working version enters here
			dequeue_thread(thread);
			thread->base.prio_deadline = newdl;
			queue_thread(thread);
		} else {                                                       <-- WORKING version enters here
			thread->base.prio_deadline = newdl;
		}
	}
}
 ```
However, this looks more like a symptom than a cause - forcing the `else` case for both versions doesn't help.

### 2. Problem seems to happen when thread is unpended
The `k_msgq_put` function internally calls `z_ready_thread(pending_thread)` on line [147](https://github.com/zephyrproject-rtos/zephyr/blob/c710f8892beef3e486dded2f0ad4394e4a4d9aa0/kernel/msg_q.c#L147) of msg_q.c, and, as of my investigation, it is the precise moment where the cache gets wrongly updated in the first place. Which makes sense, since the actual code of this function is:
```c
  static void ready_thread(struct k_thread *thread){
	...
	queue_thread(thread);
	update_cache(0);
	....
}
```
...so just change `update_cache` argument to 1 and there you go! right?
Unfortunately, no. This doesn't help us because `runq_best`, called internally by `update_cache`, **is returning the wrong thread at this point anyway**. So picking the highest priority thread or simply keeping the current one at the cache are virtually returning the same value. 

Are you noticing a pattern already?
 
 ### 3. queue_thread looks guilty
This function is a common reference in both cases, so I went to investigate that. Theoretically, after a thread is inserted in the ready queue it will ALREADY be positioned according to its priority. So calling `runq_best` right after `runq_add` should give us the highest priority thread. So, for the sake of testing, I made a few tweaks to the kernel code (can also post if anyone interested) to log the result of `runq_best` right after a thread is added in the ready queue. It will look like this:
```c
[ 1 ]   best
```
meaning, for instance, that the thread 1 is the highest priority. Since two user threads exist, we should see this log twice - and the second time should indicate which thread will actually run first. Running the different options for the `trigger` function noted above, then:

when we run the WORKING option, this is what we see in the console.
```c
[ 1 ]   best
[ 1 ]   best
[ 1 ]   start 90157290   end 90207401    dead 90556188
[ 2 ]   start 90207748   end 90257862    dead 90656534

[ 1 ]   best
[ 1 ]   best
[ 1 ]   start 110147238  end 110207370   dead 110646194
[ 2 ]   start 110211329  end 110261478   dead 110646517

[ 1 ]   best
[ 2 ]   best
[ 2 ]   start 130145405  end 130195510   dead 130643665
[ 1 ]   start 130195902  end 130266000   dead 130743338

[ 1 ]   best
[ 2 ]   best
[ 2 ]   start 150145326  end 150195506   dead 150643707
[ 1 ]   start 150195922  end 150275990   dead 150843194
``` 

But on the NON-working option, this is what we see:
```c
[ 1 ]   best
[ 2 ]   best
[ 2 ]   start 30171369   end 30222904    dead 30669903
[ 1 ]   start 30224723   end 30244793    dead 30267378

[ 1 ]   best
[ 2 ]   best
[ 2 ]   start 50159273   end 50209328    dead 50658973
[ 1 ]   start 50209717   end 50239777    dead 50358850

[ 1 ]   best
[ 2 ]   best
[ 2 ]   start 70160683   end 70210834    dead 70660134
[ 1 ]   start 70213051   end 70253162    dead 70459899

[ 1 ]   best
[ 2 ]   best
[ 2 ]   start 90156768   end 90206927    dead 90656149
[ 1 ]   start 90207629   end 90257732    dead 90555902
```

Notice that `runq_best` is giving wrong results, which then affect the execution order of the threads. To me this is looking like a problem with the priority queue insertion algorithm, since all `runq_best` really seems to do is picking the head of whatever is the data structure being used. I have inspected the code for the `runq_add` function but it all looks correct to me, so I really don't know yet what is wrong, exactly. Maybe it is related to optimization? I don't know.

Anyway, I hope these observations prove themselves useful. I'll keep going with my investigation.

peter-mitsis, 2024-10-14:
I suspect that this is boiling down to a design decision that is coming back to bite us--that being  k_thread_deadline_set() is not a scheduling point. It was assumed that k_thread_deadline_set() when called would set the deadline to a point far enough in the future that there would be intervening schedule points that would get us "back on track". That suspicion is also supported by your observation that changing the execution order works as k_msgq_put() would be a scheduling point (as it would wake the thread that was waiting on the message queue).

alexpaschoaletto, 2024-10-15:
> It was assumed that k_thread_deadline_set() when called would set the deadline to a point far enough in the future that there would be intervening schedule points that would get us "back on track"

Fair observation, but isn't a relative deadline of 50 milliseconds far enough already? I understand your concern but think the problem might be elsewhere.

By the way, I have new clues:
1. building the code with `CONFIG_NO_OPTIMIZATIONS=y` doesn't make a difference, so maybe optimization is not a problem.
2. building with a different data structure for the runqueue (`CONFIG_SCHED_SCALABLE` instead of `CONFIG_SCHED_DUMB`) also makes no difference. Which means the problem is likely not in the `runq_add` implementation.
3. When the NON-working version of the `trigger` function is used, the runqueue _does_ get correctly ordered _eventually_. Which is weird.

Let me elaborate on the latter. I have added the following function to `kernel/sched.c` and also made it available to the application, so multiple testpoints could be placed. All it does is log the runqueue as it is, and whatever is the best (highest priority) thread on it.  

```c
// thread->custom_data holds the task->id value, which is easier to read than the thread address

void k_log_runq(){
  #ifdef CONFIG_THREAD_CUSTOM_DATA
	sys_dlist_t *runqueue = thread_runq(_current);
	struct k_thread *t;
	SYS_DLIST_FOR_EACH_CONTAINER(runqueue, t, base.qnode_dlist) {
		if(t != NULL) printk("[ %d ] ", (int) t->custom_data);
	}
	printk("--> best: [ %d ]\n", (int) runq_best()->custom_data);
  #endif
}
```
So when this function is called we expect to see something like this:
```c
[ 2 ] [ 1 ] --> best: [ 2 ]		//runqueue has threads 2 and 1, and thread 2 is the highest priority
```

Now let's place the above function on two spots: 
1. in `k_msg_put`, right after it unpends the thread and reschedules.
```c
/* 1 - inside k_msgq_put */
int z_impl_k_msgq_put(struct k_msgq *msgq, const void *data, k_timeout_t timeout){
	...
	if (pending_thread != NULL) {
		...
		z_reschedule(&msgq->lock, key);
		k_log_runq();			<-- HERE!
		return 0;
	} 
	...
}
```
2. in the application function `trigger_all`, right after it triggers both threads.
```c
/* 2 - inside trigger_all */
void trigger_all(struct k_timer *timer){
	printk("\n");
	trigger(&task1);
	trigger(&task2);
	k_log_runq();			<-- HERE!
	...
}
```

## Results

WORKING option:
```c
[ 1 ] --> best: [ 1 ]							<-- msgq_put thread 1
[ 1 ] [ 2 ] --> best: [ 1 ]						<-- msgq_put thread 2
[ 1 ] [ 2 ] --> best: [ 1 ]						<-- end of trigger_all
[ 1 ]   start 50328328   end 50358404    dead 50526612
[ 2 ]   start 50359263   end 50409326    dead 50827318
```

NON-working option:
```c
[ 1 ] --> best: [ 1 ]							<-- msgq_put thread 1
[ 2 ] [ 1 ] --> best: [ 2 ]						<-- msgq_put thread 2	( WRONG )
[ 1 ] [ 2 ] --> best: [ 1 ]						<-- end of trigger_all  ( CORRECT )
[ 2 ]   start 50205780   end 50255995    dead 50704081
[ 1 ]   start 50257306   end 50287495    dead 50403207
```

In both examples thread 1 has the highest priority. However, in the NON-working option the run queue is incorrectly assembled by `msgq_put`, only to be rectified when `trigger_all` is about to finish. Unfortunately, this correction doesn't reflect on whichever runs first - thread 2 runs before thread 1 nonetheless - as if the queue was still in order [ 2 ] [ 1 ].

alexpaschoaletto, 2024-10-17:
> I suspect that this is boiling down to a design decision that is coming back to bite us--that being k_thread_deadline_set() is not a scheduling point.

Ok, by reading your comment again it came to mind that since this behavior is apparently by design then there's likely nothing to "fix" here - although I shoud point that the execution order still looks plain wrong specially when you add more threads in the game (they will go LIFO all the way). 

If that is the case, suppose I have a thread already executing when an ISR calls `k_thread_deadline_set` for it. How can I proactively signalize the scheduler that I want to reschedule if `k_yield()` is not available within ISRs? Will `k_sched_lock()` and `k_sched_unlock()` do the trick?

peter-mitsis, 2024-12-18:
@alexpaschoaletto - I have two ideas on how to do a work around for this.

Option 1. Enable SMP, but set the number of CPUs to 1. Assuming that the architecture in question supports SMP, this is likely the better of the two options. SMP does not use the ready queue cache, so this ought to get around the issue.

```
    CONFIG_SMP=y
    CONFIG_MP_MAX_NUM_CPUS=1
```

Option 2. Force a reschedule to occur when exiting the ISR by suspending the interrupted thread and then immediately resuming it. However, this approach is prone to unwanted side effects and is the riskier of the two options.

```
   k_thread_deadline_set( ... );
    k_thread_suspend(arch_current_thread());
    k_thread_resume(arch_current_thread());
```

alexpaschoaletto, 2024-12-19:
@peter-mitsis although I do appreciate the suggestions, honestly I'd rather see this solved in a "proper" way instead of with work arounds. What a timing to suggest an SMP-related solution, though, as I've been searching about it in the past week and so far it looks like Zephyr [only has (very) few targets](https://github.com/zephyrproject-rtos/zephyr/issues/83177) that support it (that shouldn't stop your suggestion from working, though, as it only uses SMP to skip the cache value).

andyross, 2024-12-19:
Just saw this.  Gotta be honest, this is way too many words to read, so I skipped some. :)

Correct me if I'm wrong, but the summary of the issue here is:

1. k_thread_deadline_set() isn't a reschedule point.  As mentioned, that's by design.  We can argue about it, but not here.
2. There is no easy API (as there is with k_yield() in a thread) to force a reschedule from interrupt context for situations where it's desired.

Solving 1 will be a big fight.  But 2 should be relatively straightforward.  Just add a `k_isr_reschedule()` or whatever (or just overload k_yield() to mean "yield on ISR exit" when called in what is currently an invalid context), which is a tiny wrapper around update_cache() on UP and a noop on SMP.

peter-mitsis, 2024-12-19:
@andyross - We are on the same page. I started working on just such a routine yesterday afternoon.

andyross, 2024-12-19:
FWIW, the second option could probably be as simple as (completely untested, obviously):

```patch
diff --git a/kernel/sched.c b/kernel/sched.c
index 3323adb05e2..280b1a785a1 100644
--- a/kernel/sched.c
+++ b/kernel/sched.c
@@ -1060,7 +1060,10 @@ bool k_can_yield(void)
 
 void z_impl_k_yield(void)
 {
-       __ASSERT(!arch_is_in_isr(), "");
+       if (arch_is_in_isr()) {
+               update_cache(1);
+               return;
+       }
 
        SYS_PORT_TRACING_FUNC(k_thread, yield);
 
```

alexpaschoaletto, 2024-12-19:
> Just saw this. Gotta be honest, this is way too many words to read, so I skipped some. :)
>
> Correct me if I'm wrong, but the summary of the issue here is:
>
> 1. k_thread_deadline_set() isn't a [...]

Yessir, that's the memo. I tend to give way too many details sometimes 😅 

> Solving 1 will be a big fight. But 2 should be relatively straightforward. Just add a `k_isr_reschedule()` or whatever (or just overload k_yield() to mean "yield on ISR exit" when called in what is currently an invalid context), which is a tiny wrapper around update_cache() on UP and a noop on SMP.

I'd be happy with any solution, really. I just want to know the proper way of implementing the code.

peter-mitsis, 2024-12-19:
FYI, I'm presently working on test code for `k_reschedule()`--a means to manually force a schedule point.

alexpaschoaletto, 2024-12-20:
> FYI, I'm presently working on test code for `k_reschedule()`--a means to manually force a schedule point.

Thank you! That will be really great.

PR #83249 (2024-12-24): kernel: Add k_reschedule()
Adds the routine k_reschedule() to force a schedule point. It is callable at both thread and ISR level.

Most applications will never use this.

Fixes #79361
closed 2024-12-24 by kartben