zephyrproject-rtos/zephyr #79380: Twister failing due to missing nrf-regtool

rettichschnidi, 2024-10-03:
**Describe the bug**

**To Reproduce**

Steps to reproduce the behavior:
1. `twister --all -s sample.drivers.crypto.mbedtls`
2. Encounter various build errors for `nrf54h20dk` and `nrf9280pdk` boards:
  ```
  INFO    - The following issues were found (showing the top 10 items):
INFO    - 1) samples/drivers/crypto/sample.drivers.crypto.mbedtls on nrf9280pdk/nrf9280/cpuapp error (Cmake build failure)
INFO    - 2) samples/drivers/crypto/sample.drivers.crypto.mbedtls on nrf9280pdk/nrf9280/cpuppr/xip error (Cmake build failure)
INFO    - 3) samples/drivers/crypto/sample.drivers.crypto.mbedtls on nrf9280pdk/nrf9280/cpurad error (Cmake build failure)
INFO    - 4) samples/drivers/crypto/sample.drivers.crypto.mbedtls on nrf54h20dk/nrf54h20/cpuapp error (Cmake build failure)
INFO    - 5) samples/drivers/crypto/sample.drivers.crypto.mbedtls on nrf54h20dk/nrf54h20/cpuppr error (Cmake build failure)
INFO    - 6) samples/drivers/crypto/sample.drivers.crypto.mbedtls on nrf54h20dk/nrf54h20/cpuppr/xip error (Cmake build failure)
INFO    - 7) samples/drivers/crypto/sample.drivers.crypto.mbedtls on nrf54h20dk/nrf54h20/cpurad error (Cmake build failure)
INFO    - 8) samples/drivers/crypto/sample.drivers.crypto.mbedtls on nrf54h20dk/nrf54h20/cpuflpr error (Cmake build failure)
INFO    - 9) samples/drivers/crypto/sample.drivers.crypto.mbedtls on nrf54h20dk/nrf54h20/cpuflpr/xip error (Cmake build failure)
  ```

**Expected behavior**

Running twister successfully on all supported architectures without needing to install extra/undocumented dependencies.

**Impact**

Can not run Twister for all architecture without encountering errors.

**Logs and console output**

The errors in build.log files look like this:

```
CMake Error at <snip>/zephyr/modules/hal_nordic/CMakeLists.txt:15 (find_package):
  Could not find a configuration file for package "nrf-regtool" that is
  compatible with requested version "5.6.0".

  The following configuration files were considered but not accepted:

    <snip>/modules/hal_nordic/nrf-regtool/nrf-regtoolConfig.cmake, version: unknown
```

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain: Zephyr SDK 0.16.8
 - Commit: 44d101acabb5f3c0916ba9c193361c32e16c015c

**Workaround**

* `pip install nrf-regtool`

nordic-piks, 2024-10-10:
Triaged internally

carlescufi, 2024-10-10:
@rettichschnidi thanks for raising this. This was [discussed](https://discord.com/channels/720317445772017664/1196766845332619305/1196766847962464276) on Discord since there was no clear precedent we ended up deciding to add this to the [Zephyr Docker image](https://github.com/zephyrproject-rtos/docker-image/commit/23e5da1e65edadba9c71bb4d2b776a9c2a054f30)

dleach02, 2024-10-16:
Still seeing this failure?

dottspina, 2024-10-17:
Yes, I came across this while trying to reproduce a CI error with:

```
west build -p -b nrf54h20dk/nrf54h20/cpuapp tests/drivers/pwm/pwm_api -T drivers.pwm.pwm_fast
```

Just to let you know:
- the latest version of `nrf-regtool` (7.0.0) is incompatible with the test case above: "Error: Invalid value: Generating from DTS is only supported for peripheral BICR"
- works fine with `pip install nrf-regtool==5.6.0`

nashif, 2024-11-01:
@carlescufi  putting something in a docker does not solve the  confusion and and problems this is causing when someone tries something locally, i.e. test development, build tests or whatever. This is not documented anywhere, installing the package based on answers on discord does not solve the problem, seems like you have to know which version you need etc etc.

57300, 2024-11-01:
FYI, for build-only tests, you can disable the check locally with `CONFIG_NRF_REGTOOL_GENERATE_UICR=n`

For CI, unfortunately the docker solution is faulty sometimes, in that it may end up using an outdated version of nrf-regtool. I don't know why updating the docker image wouldn't fix it for everyone.

57300, 2024-11-02:
nrf-regtool requirement was dropped again: 0b18f31246cd5d0bcdd10a3ad6e3bd108bf3f95d
We'll consider adding a note about this package to our board documentation, as well as follow-up changes to prevent users from programming a non-booting application when nrf-regtool is not installed. Not sure the latter can make it into 4.0.

pdgendt, 2024-11-02:
Created #80782 as an alternative.

PR #80839 (2024-11-06): west: runners: nrf: Fix for check for missing UICR
On nRF54H and nRF92, booting certain cores requires programming a UICR, which is normally generated using nrf-regtool. This should be considered an optional dependency, because we do not wish to force non-Nordic users to install it just to work with Zephyr, or just for build-only tests.

When nrf-regtool is not installed, a CMake warning will be displayed, but people ignore warnings all the time. As the last line of defense, check for missing UICR in the nrfutil flash runner, to prevent our users from unintentionally programming unbootable fir [... cut]
closed 2024-11-06 by mmahadevan108