zephyrproject-rtos/zephyr #79451: The STM32  standby_shutdown sample has a simple mistake

MZachmann, 2024-10-05:
In the sample application for the STM32 power management standby at    samples/boards/stm32/power_mgmt/standby_shutdown 

I'm pretty certain that the line in main.c line 49 that is currently:

LL_PWR_IsWakeUpPinPolarityLow(LL_PWR_WAKEUP_PIN2);

should be

LL_PWR_SetWakeUpPinPolarityLow(LL_PWR_WAKEUP_PIN2);

The first line requests the polarity and doesn't use the output which seems wrong. The second line (correctly) tells the os the pin polarity.

FRASTM, 2024-10-07:
Tanks @MZachmann to point this. I do not see any requirement to read (access) the Power register CR4 (what is done by the LL_PWR_IsWakeUpPinPolarityLow function) before the wakeUp pin configuration.

However there is a "recommended" sequence, for the stm32L4 serie, for example :
```
/* The Following Wakeup sequence is highly recommended prior to each Standby mode entry
     mainly when using more than one wakeup source this is to not miss any wakeup event.
     - Disable all used wakeup sources,
     - Clear all related wakeup flags, 
     - Re-enable all used wakeup sources,
     - Enter the Standby mode.
  */
```

FRASTM, 2024-10-07:
Second this sample should be upgraded thanks to the new PR  "[STM32: configure wake-up pins with GPIOs from devicetree for exiting Poweroff](https://github.com/zephyrproject-rtos/zephyr/pull/73058 )"

closed 2024-12-22 by github-actions[bot]