zephyrproject-rtos/zephyr #79473: XIAO ESP32-C3 - bt_disable() causes crash

Teodor12, 2024-10-06:
Hello!

I'm creating a hobby BLE project with my XIAO ESP32-C3.

To be concise: I'm calling the bt_disable() function, and my whole application crashes. This doesn't happen when I build and flash the program on my **ESP32 devkitc wroom/wrover** board.

[image]

Steps to reproduce(pseudocode):
1. bt_enable();
2. bt_le_adv_start(BT_LE_ADV_CONN_NAME, advertisment_data, ARRAY_SIZE(advertisment_data), NULL, 0);
3. Wait a bit.
4. bt_le_adv_stop();
5. bt_disable();

I read that there were some issues with nrf-boards using bt_disable.

Thank you for your help!

sylvioalves, 2024-11-28:
@Teodor12 this is fixed in v4.0.0 or latest main. Can you re-check? I'll close the issue but feel free to open again.

Teodor12, 2024-11-28:
Greetings @sylvioalves!

The issue is still present with the board.

sylvioalves, 2024-11-28:
Can you provide a basic code? I was able to make it work here. Have you tested on v4.0.0?

closed 2025-02-12 by sylvioalves