zephyrproject-rtos/zephyr #79475: W500 Ethernet Shield not working with ESP32

MoritzPoehlandt, 2024-10-06:
Hi,
I'm trying to use the Zephyr DHCPv4 sample with an ESP32-WROOM and the W5500 Ethernet shield. I have added the specific overlay and configuration files to the project. These are similar to the Acreli_eth_w5500 Ethernet shield setup, but I'm using the default SPI3 pins for the ESP32.

```
&spi3 {
    status = "okay";
    cs-gpios = <&gpio0 5 GPIO_ACTIVE_LOW>;
    spi_w5500: w5500@0 {
        compatible = "wiznet,w5500";
        reg = <0x0>;
        spi-max-frequency = <10000000>;
        int-gpios = <&gpio0 4 GPIO_ACTIVE_LOW>;
        reset-gpios = <&gpio0 15 GPIO_ACTIVE_LOW>;
    };
};
```

```
CONFIG_SPI=y
CONFIG_NET_L2_ETHERNET=y
CONFIG_ETH_W5500=y
CONFIG_ETH_W5500_TIMEOUT=1000
CONFIG_NET_MGMT=y
```

**Logs and console output**
```
[00:00:00.163,000] <err> eth_w5500: Unable to read RTR register
[00:00:00.163,000] <dbg> net_core: net_init: (main): Priority 90
[00:00:00.163,000] <dbg> net_pkt: net_pkt_init: (main): Allocating 14 RX (840 bytes), 14 TX (840 bytes), 0 RX data (0 bytes) and 0 TX data (0 bytes) buffers
[00:00:00.163,000] <dbg> net_core: l3_init: (main): Network L3 init done
[00:00:00.163,000] <dbg> net_if: net_if_init: (main): 
[00:00:00.163,000] <dbg> net_if: init_iface: (main): On iface 0x3ffb3088
[00:00:00.164,000] <dbg> net_if: update_operational_state: (main): iface 1 (0x3ffb3088), oper state DOWN admin DOWN carrier OFF dormant OFF
[00:00:00.164,000] <dbg> net_if: net_if_post_init: (main): 
[00:00:00.164,000] <dbg> net_if: net_if_up: (main): iface 1 (0x3ffb3088)
[00:00:00.164,000] <dbg> net_if: net_if_up: (main): Device w5500@0 (0x3f4083e0) is not ready
[00:00:00.164,000] <dbg> net_conn: net_conn_change_callback: (main): [3] connection handler 0x3ffb6468 changed callback
[00:00:00.164,000] <dbg> net_conn: net_conn_change_remote: (main): [3] connection handler 0x3ffb6468 changed remote
[00:00:00.164,000] <dbg> net_conn: conn_register_debug: (main): [0x3ffb6468/17/1/0x1d] remote -/67 
[00:00:00.164,000] <dbg> net_conn: conn_register_debug: (main):   local 0.0.0.0/68 cb 0x400e4df0 ud 0
0x400e4df0: net_dhcpv4_input at /Users/

*** Booting Zephyr OS build v3.7.0-3952-gb6aed5c505e6 ***
[00:00:00.164,000] <inf> net_dhcpv4_client_sample: Run dhcpv4 client
[00:00:00.165,000] <inf> net_dhcpv4_client_sample: Start on w5500@0: index=1
```

**Environment (please complete the following information):**

 - OS: MacOS
 - Toolchain: Zephyr SDK
 - Zephyr: latest (b6aed5c505e69f6200cdf9d3c1c549d47c126849) and v3.7 branch

gramsay0, 2024-10-08:
I gave this a quick go since I had an ESP32 and a W5500 handy.

There may be an issue in the ESP32 SPI driver when using GPIO chip select emulation, it appears to toggle chip select between each buffer.

Maybe something to do with `!ctx->num_cs_gpios` below?
```
/* keep cs line active until last transmission */
hal_trans->cs_keep_active =
	(!ctx->num_cs_gpios &&
	 (ctx->rx_count > 1 || ctx->tx_count > 1 || ctx->rx_len > transfer_len_frames ||
	  ctx->tx_len > transfer_len_frames));
```

@MoritzPoehlandt  as a quick workaround you could remove `cs-gpios = <&gpio0 5 GPIO_ACTIVE_LOW>;` from your overlay to enable automatic chip select (rather than GPIO emulation)

gramsay0, 2024-10-08:
Actually the issue might be that you're enabling both hardware and GPIO emulation chip select.

It works if I remove either from this overlay:
```dts
&spim3_default {
	group1 {
		pinmux = <SPIM3_MISO_GPIO19>,
				 <SPIM3_SCLK_GPIO18>,
				 <SPIM3_CSEL_GPIO5>; // <---------- Either remove this
	};
	group2 {
		pinmux = <SPIM3_MOSI_GPIO23>;
		output-low;
	};
};

&spi3 {
	status = "okay";
	cs-gpios = <&gpio0 5 GPIO_ACTIVE_LOW>; // <---------- or this
	spi_w5500: w5500@0 {
		compatible = "wiznet,w5500";
		reg = <0x0>;
		spi-max-frequency = <10000000>;
		int-gpios = <&gpio0 4 GPIO_ACTIVE_LOW>;
		reset-gpios = <&gpio0 15 GPIO_ACTIVE_LOW>;
	};
};
```

MoritzPoehlandt, 2024-10-08:
Thanks for your help. Just needed to add the  'local-mac-address' property as well and now works perfectly. Is this still a bug or more like a feature?

gramsay0, 2024-10-10:
> Thanks for your help. Just needed to add the 'local-mac-address' property as well and now works perfectly. Is this still a bug or more like a feature?

@MoritzPoehlandt I would say it is probably just a confusing and difficult to detect misconfiguration.

@sylvioalves what are your thoughts?

closed 2024-12-25 by github-actions[bot]