zephyrproject-rtos/zephyr #79512: Bluetooth: Host: Merge bsim test builds to reduce CI load

alwa-nordic, 2024-10-07:
For most tests we spend more time building them than actually running them. It would be better if they could all be built in a single build.

Issue spawned from: https://github.com/zephyrproject-rtos/zephyr/pull/79501#discussion_r1790165096

Some work has been drafted on how to merge builds that originally required differing configurations: https://github.com/zephyrproject-rtos/zephyr/pull/77122

PavelVPV, 2024-10-07:
Also, look if some code can be shared with [babblekit](https://github.com/zephyrproject-rtos/zephyr/tree/main/tests/bsim/babblekit) or reused.

aescolar, 2024-10-08:
> For most tests we spend more time building them than actually running them. It would be better if they could all be built in a single build.
> 
> Issue spawned from: [#79501 (comment)](https://github.com/zephyrproject-rtos/zephyr/pull/79501#discussion_r1790165096)
> 
> Some work has been drafted on how to merge builds that originally required differing configurations: #77122

Thanks @alwa-nordic . Note that at least tests of the same area which share configuration (like the one that triggered that comment) should be easy to build in one image (as we do for many other tests).