zephyrproject-rtos/zephyr #79519: Trying to run stm32f0_disco with blinky_pwm

adri1mart1, 2024-10-07:
Hello zephyr community,

**Describe the bug**

I am using zephyr 3.7.0.
I have tried to run the blinky_pwm example on the [stm32f0_disco board](https://docs.zephyrproject.org/latest/boards/st/stm32f0_disco/doc/index.html). The overlay file does not exist (even on master branch) yet so i have created this one along with other overlay files:

**To Reproduce**

Add an `stm32f0_disco.overlay` in the `zephyr\samples\basic\blinky_pwm\boards\` directory with the following content.

```
/ {
    pwmleds: pwmleds {
        compatible = "pwm-leds";
        status = "okay";
        green_pwm_led: green_pwm_led {
            pwms = <&pwm 1 PWM_MSEC(20) PWM_POLARITY_NORMAL>;
        };
    };

    aliases {
        pwm-led0 = &green_pwm_led;
    };
};

&timers1 {
    status = "okay";
    pwm: pwm {
        compatible = "st,stm32-pwm";
        status = "okay";
        #pwm-cells = < 0x3 >;
        pinctrl-0 = <&tim1_ch1_pa8>;
        pinctrl-names = "default";
    };
};
```

Note: green_pwm_led label is only a copy paste from another board, not very relevant probably but I did not check the schematic in details of the stm32f0_disco board. My goal here was just to have PA8 pin running a PWM signal.

In the blinky_pwm main.c, I've just modified the following two lines to be compatible with my hardware and my needs:

```
#define MIN_PERIOD PWM_SEC(1U) / 16384U
#define MAX_PERIOD PWM_SEC(1U) / 1024U
```

Compiling and flashing:

```
west build -b stm32f0_disco .\zephyr\samples\basic\blinky_pwm\
west flash
```

```
*** Booting Zephyr OS build v3.7.0 ***
PWM-based blinky
Calibrating for channel 1...
Done calibrating; maximum/minimum periods 976562/61035 nsec
Using period 976562
[00:00:00.014,000] <err> os: ***** HARD FAULT *****
```

If I set the `CONFIG_LOG_MODE_IMMEDIATE=n` in the prj.conf, now things are working:

**Expected behavior**

```
*** Booting Zephyr OS build v3.7.0 ***
PWM-based blinky
Calibrating for channel 1...
Done calibrating; maximum/minimum periods 976562/61035 nsec
Using period 976562
Using period 488281
Using period 244140
[...]
```

And I have a valid PWM output on PA8. 

Any idea on what's wrong with my try or is there something hidden ?

**Environment**

 - OS: Windows
 - Zephyr SDK 0.16.5

FRASTM, 2024-10-09:
looking for a  stm32f0_disco available but  maybe one of my stm32f072b_disco can be similar.
Looking at the board DTS, the tim1_ch1_pa8 is correctly set as output pwm pin of the timer1 channel1  but the green_pwm_led is on pin PC9. Maybe do period setiing. Investigating

FRASTM, 2024-12-10:
@adri1mart1 can you confirm you still see the bug ?

adri1mart1, 2024-12-15:
Hello,

Yes I confirm, I just did the test on the latest hash of main ( b0baeee03e5 ) with the exact same result.

Switching `CONFIG_LOG_MODE_IMMEDIATE` from `Yes` to `No` in the `prj.conf` removes the crash problem.

Regards,
Adrien M.

erwango, 2024-12-17:
> Switching CONFIG_LOG_MODE_IMMEDIATE from Yes to No in the prj.conf removes the crash problem.

Thanks. Closing

adri1mart1, 2024-12-20:
@erwango Well no, this is not sloved. 

If you look at the blinky_pwm sample [prj.conf](https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/basic/blinky_pwm/prj.conf) file, this option is enabled by default.
That mean the program crashes if you do nothing.

I've tried different things for a long time to finally find out the problem disapear if you change the CONFIG_LOG_MODE_IMMEDIATE from Yes to No, however, I don't know why !

If there's anything about this board that is incompatible with this option, it should be turned off.

Leaving this as it is is very confusing I believe..

FRASTM, 2024-12-20:
Re-opening.
I understand you still have the pb, that we cannot reproduce on the stm32072b_disco (with the same overlay for pwm on timers1 channel 1 pin PA8
The stm32f0_disco is based on the stm32f051X8

adri1mart1, 2024-12-26:
Not at the moment (i'm on vacation) but i'll check early January if i can find that at work and will let you know

closed 2025-03-12 by github-actions[bot]